kantlivelong/OctoPrint-PSUControl #89: [Request] Turn on printer when starting a print

drewcassidy, 2018-01-06:
My printer (Printrbot simple) has no way to power the logic board externally (at least not without removing the voltage regulator from the board, etc), so theres no way to send it Gcode commands until after the PSU has been turned on. This isnt a huge issue thanks to the button on the UI, but it would be nice to have the plugin automatically turn on the PSU when attempting to start a print, if possible.

kantlivelong, 2018-01-06:
I imagine if the board is not powered then the serial connection isn't live. If that's the case then a print job can't even be started.

drewcassidy, 2018-01-06:
Yeah thats why I wasnt sure if it was possible, you would have to override the process for starting a print to first power it on and then connect, and I dont know if thats possible in the API

kantlivelong, 2018-01-07:
Not really feasible without hacking up the UI as far as I can tell. 

However it appears that PrintrBoard supports power supplied by USB which would allow you to have it setup in an always-on fashion. See http://reprap.org/wiki/Printrboard

drewcassidy, 2018-01-09:
The Printrboard USB power isnt safe to use while also powering over the 12V, unfortunately. the jumper is only there to allow flashing the firmware without setting up a full power supply

Tailslide, 2018-04-03:
I'm having the same issue with the Rambo board... can't connect to it until it gets 12v.

Tailslide, 2018-04-10:
Since I print using a script from Simplify 3D in windows I was able to work around this by leaving the PSU off by default and running the following batch file print3d.bat which checks to see if the PSU is on, and if not uses the API to turn it on and connect octoprint to the printer and loop until it's connected. Then the script finally exits and the second line in my slicer script runs to send the gcoode to octopi.  I have the plugin configured to automatically turn off after 30 minutes of activity.

[printeron.bat.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1895083/printeron.bat.txt)
[print3d.bat.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1895103/print3d.bat.txt)

kantlivelong, 2018-04-10:
Thanks for the info @Tailslide.

I'll be tackling some of these issues in the very near future. Still not sure exactly how I will handle it but the more info the better.

kantlivelong, 2021-04-24:
So this won't be doable in the OctoPrint UI without some hacks that I'm not really fond of. However, I have added this functionality for when printing via the API (Sending from a slicer). Will be part of 1.0.5

50da9c1052acfad9213ded64b7cc3c008a6ebae5

closed 2021-04-24 by kantlivelong