kantlivelong/OctoPrint-PSUControl #90: [Question] Wiring up only RPi

ido1990, 2018-01-09:
Hi!

I wonder if there's any way to control the PSU only with the RPi.
I've server's ATX psu with 12VSB connected to USB voltage regulator that powers the RPi with 5V.
What I actually trying to do is just configuring the RPi to connect the GPIO with the PS_ON line to ground to start the PSU.
I don't see why I need to control it with the board and Gcode, just use a button on Octoprint to power on and that's it.

I already did it with ESP8266 but I think getting it all inside the Octoprint is much convenient.
Is that possible?

Thanks!

kantlivelong, 2018-01-09:
I assume you mean 5VSB? You can do it however you want.

[This](https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Wiring-Diagrams) may be helpful.

ido1990, 2018-01-09:
Thanks @kantlivelong 
No, I've 12VSB it's Dell server's PSU.
I've buck converter to get 5V on USB.
I saw the this page on the wiki but I don't unhderstand why I need to use transistor.
Why don't grounding the PS_ON pin so it start the PSU?
Because the RPI using the same ground (because it's connected to the PSU standby).
I hope I explained myself... :)

kantlivelong, 2018-01-09:
You can do that if you want. This really isn't the place for electronics support so I'm afraid you will need to look for assistance elsewhere.

ido1990, 2018-01-09:
I know, just wanted to know if the plugin support it.

kantlivelong, 2018-01-09:
It can control a power supply by using GPIO to turn a pin HIGH or LOW,  system commands to run a script you wrote, or send G-Code to the printer.

ido1990, 2018-01-09:
Ok, great!
Many thanks!

closed 2018-01-09 by ido1990