kantlivelong/OctoPrint-PSUControl #92: Error turning off heaters

fflosi, 2018-01-21:
Hi, i´m using Marlin-bugfix-1.1.x and octoprint 1.3.6.
Shutting off PSU is not working, and i got the debug error bellow.

` 2018-01-21 14:36:47,111 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 1 minute(s). Turning heaters off prior to shutting off PSU.
Exception in thread Thread-49:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 46, in run
    self.function(*self.args, **self.kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 411, in _idle_poweroff
    if self._wait_for_heaters():
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 422, in _wait_for_heaters
    if float(heaters.get(heater)["target"]) != 0:
TypeError: float() argument must be a string or a number
`

dont know if is something in my configuration, but i correct it by removing "float" from line 422.
`if  float(heaters.get(heater)["target"]) != 0:` to ` if (heaters.get(heater)["target"]) != 0:`

kantlivelong, 2018-01-21:
Hi @fflosi,

Id be interested in knowing what the value for heaters.get(heater)["target"] is. Would you mind reverting the change and place this above it:
`self._logger.debug("Heater %s target= %s" % (heater, heaters.get(heater)["target"])`

Shoot the logs over when the issue occurs. Be sure to enable DEBUG logging for psucontrol.

Thanks :)

kantlivelong, 2018-01-21:
Made a correction in the above comment.

fflosi, 2018-01-21:
Here is the result

2018-01-21 15:59:45,635 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 1 minute(s). Turning heaters off prior to shutting off PSU.
2018-01-21 15:59:45,636 - octoprint.plugins.psucontrol - DEBUG - Heater bed target= None
Exception in thread Thread-42:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 46, in run
    self.function(*self.args, **self.kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 411, in _idle_poweroff
    if self._wait_for_heaters():
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 424, in _wait_for_heaters
    if float(heaters.get(heater)["target"]) != 0:
TypeError: float() argument must be a string or a number

kantlivelong, 2018-01-21:
Ah. In _OctoPrint Settings -> Printer Profiles -> Print bed and volume_ do you have heated bed turned on?

fflosi, 2018-01-21:
No. it´s off.
Tried with bed turned on on octoprint, but the result was the same. 

2018-01-21 16:28:21,454 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 1 minute(s). Turning heaters off prior to shutting off PSU.
2018-01-21 16:28:21,455 - octoprint.plugins.psucontrol - DEBUG - Heater bed target= None

But if I enable the bed and set any value, then it starts to get target as 0.0.
It will work even if I disable the bed again on setup, but as soon as I restart OctoPrint ir goes back to "none" on last check. 

2018-01-21 16:41:13,565 - octoprint.plugins.psucontrol - INFO - Idle timeout reached after 1 minute(s). Turning heaters off prior to shutting off PSU.
2018-01-21 16:41:13,566 - octoprint.plugins.psucontrol - DEBUG - Heater bed target= 0.0
2018-01-21 16:41:13,567 - octoprint.plugins.psucontrol - DEBUG - Heater bed already off.

it might be the way OctoPrint work related to the bed without a sensor, or Marlin might be sending some other value, as it normally check "BED_MINTEMP". I've looked at Marlin code but could not identify a relation.

kantlivelong, 2018-01-21:
Thanks for the info. Should be fixed with 8039be5b1434e512d8987ef3f67634588554eded

Give it a shot and let me know.

fflosi, 2018-01-21:
Working now. tks.

closed 2018-01-21 by fflosi

kantlivelong, 2019-04-09:
#126