kantlivelong/OctoPrint-PSUControl #96: Request: Turn on PSU when pressing "connect"

jbkuma, 2018-02-18:
Along the lines "connect after turning on PSU" and "turn on when starting print" perhaps a feature that automatically turns on the PSU when the Connect button is pressed, with a settable delay to attempt the connection.

kantlivelong, 2018-02-20:
I get the request in general but could you elaborate the exact use case here?

jbkuma, 2018-02-20:
It's just a non-critical step saver.  Rather than having to hit the toggle then hit "connect to printer", one could click "connect to printer" and it would automatically toggle on the printer the attempt a connect after a brief delay. (or the power toggle could attempt the same)

kantlivelong, 2018-02-20:
Yes but what is the use case here? Let me clarify, how do you have everything wired and set up? What steps does everything go through? Is your printer using USB serial? etc...

jbkuma, 2018-02-20:
MP Select Mini which is USB Serial, I have the Pi a relay housed inside
cutting the DC to the printer.  Currently setup as NO, but I'm debating
switching it to it to NC so if/when I reboot the pi or restart octoprint it
won't cut power to the printer.  The power switch is wired ahead of the
relay and can cut power if necessary.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Gadroc, 2018-06-02:
This would be incredibly useful for me as well.  My setup is a TAZ6 which runs a RAMBo board.  I have a relay setup controlling power to the TAZ6.  Unfortunately auto connect via PortLister does not work in this scenario.  The chip running serial to USB on the Rambo board is powered via USB so the serial port is preset always while that USB cable is plugged in even though power is cut to the RAMBo board.

blaskkaffe, 2018-08-30:
Would be very useful for me too!
I have the Pi connected to +5VSB and the controller card connected via USB, but it doesn't power from usb, so it is powered from 12v. It doesnt have the option to power both from usb and DC, only selectable via jumper (Anycubic Trigorilla).

So clicking the connect button to turn on psu and connect would definitely save a step!

nionio6915, 2018-08-31:
blaskkaffe - have you looked at the Porlister plugin?

saschaludwig, 2018-10-01:
+1 for the "connect after turning on PSU". I use octopi (always on) with an anycubic i3 mega on USB and it always require a manual "connect" click after switching on the PSU of the printer. Would be great to save that step, otherwise the "Automatically turn PSU ON" is rather useless because one always has to click connect to start the print

nionio6915, 2018-10-01:
you need to look at portlister plugin. that is exactly what it does. it will connect to the printer when it comes online. you have to give a moment for portlister to find it. 
if you can not be patient for the watchdog cycle to complete, then you can connect it manually.

Martyfree123, 2018-10-01:
Just download and use the Pull. That's what I've been doing and the auto connect printer option works perfect. If the creator of the plug in is unwilling to merger the Pull into the master then we'll just have to download and use the Pull.

numanair, 2018-10-17:
Pull #101 seems to work perfectly for my use case, which sounds identical to @jbkuma's.