kantlivelong/OctoPrint-PSUControl #97: IOT Power Relay configuration help needed

4RnrRick, 2018-02-26:
I'm using a IOT Power Relay ( https://dlidirect.com/products/iot-power-relay) to power my Printer (CR-10s). I have OctoPrint installed on a Raspberry Pi 2 B. I'm using GPIO Pin 40 and Ground as the Signal wire to the IOT Power Relay.

The IOT Power Relay is Version 2 which has (1) Always-ON, (1) Normally-ON & (2) Normally-OFF outlets.

What I would like to do is to plug the RPi into the Always-ON and the Printer into Normally-ON outlets. Then when I flip the power switch on the IOT Power Relay: 
1.) The Printer and RPi will become powered.
2.) The Pi will boot with OctoPrint.
3.) The Printer will boot up as normally.
4.) Once loaded up, OctoPrint should auto connect to the Printer.
4.) If I could get the printer or RPi to beep, that would be awesome to let me know I can remote in to a local IP to start a Print, but not neccessary.

I would also like the green lighting bolt to be lit on the menu bar of OctoPrint whenever the printer is powered. If I toggle the lighting bolt, the printer should turn off with a warning window. Once a print job is done, I would also like the Printer to turn off with a small delay (say 5 mins).

The best I've been able to achieve is to plug the printer into Normally-OFF outlet as PSU Control appears to send a signal on GPIO Pin 40 and toggle the relay in the IOT Power Relay. But even using invert, I have to manually use the lightning bolt button to power up the Printer and about 60% of the time it won't let me connect as it flags a power failure. I also have to manually connect.

**Any and all help is appreciated** as this is driving me nuts and unfortunately this is only the second time I've ever messed with a RPi, so my knowledge is very limited. **I just don't understand the settings i need to use.** seems like a simple concept.

### Current setup:
RPi plugged into Always-ON
Printer plugged into Normally-Off
GPIO Pin 40 and Ground wired to the IOT Power Relay signal port.
PSU Control Settings:
--General-- 
checkmark next to Warnings
GPIO Mode: Board
--Switching--
Switching Method: GPIO Pin
On/Off: 40, no checkmark on invert
checkmark on enable G-code commands
On G-Code: M80
Off G-Code: M81
--Sensing--
Method: Internal
--Power On Options--
No Checkmark for Auto On
Delay is 0
--Power Off Options--
 No Checkmark for Auto Off
CheckMark on Disconnect on Power Off

Then under the GCODE scripts I have added:
---After print job completes---
;Wait 5 mins
G4 S300
;Turn of PowerRelay
M81

kantlivelong, 2018-02-26:
If I'm understanding this right you want:

1. RPi/Printer to beep on boot
2. Status indicator to be on because you default to on.
3. OctoPrint to auto connect to printer

Based on the above:
1. The RPi doesn't have an internal speaker and most printers do not either. But if your printer does happen to have one then **M300** would do the trick.
2. You can get this working by setting up [Sensing](https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Settings#sensing). While I'd recommend using GPIO based sensing you could also use command based sensing to call a script that utilizes the [gpio](http://wiringpi.com/the-gpio-utility/) command to read the pin status.
3. The [PortLister](https://plugins.octoprint.org/plugins/portlister/) plugin should do this for you.

4RnrRick, 2018-02-26:
I appreciate the quick response. 
1.) yes my printer has M300. so I can use that.
2.) Can you give me more set by step detail? I've tried so many things that I'm honestly confused as I don't fulling understand the Settings Wiki Page in basic terms for a scrub like myself.
3.) I'll look that up.
When I plug the printer into the Normally-ON outlet, Once Octoprint loads, it sends a signal to GPIO and turns the printer off. If is use Invert, to stop that action, then lighting bolt button and Gcode commands work in opposite.

kantlivelong, 2018-02-26:
1. Copy [this](https://gist.github.com/kantlivelong/d14427d9b5dbb84be54dd13d97f6549b) script onto the Pi at /home/pi/sense.sh
2. Set as executable `chmod 755 /home/pi/sense.sh`
3. Set Sensing to "System Command"
4. Set the command to `/home/pi/sense.sh`

You may need to alter the script slightly to match your configuration but it should work.

4RnrRick, 2018-02-27:
I hope I did this right.

I used Putty to SSH into RPi and used 
"sudo nano /home/pi/sense.sh" to copy and paste that script into it.

I then used "sudo chmod 755 /home/pi/sense.sh".

I installed Portlister via the plugin manager.

I then restarted OctoPrint.

I logged into OctoPrint locally from a win10 machine.

Under PSU Control, I changed "switching" to 'GPIO Pin #40'. No checkmark on Invert, Enable switching with G-code commands, M80 & M81. Under 'Sensing' i set it to 'System Command' and typed  "/home/pi/sense.sh". Power On Options is un checkmarked. Power Off Options, checkmark only next to 'disconnect on power off'.

I confirmed that "Auto-connect to printer on sever start" was checkmarked.

Under "GCODE Scripts", 'After Connection to Printer is established" I added:
G4 S2
M300 S300 P100

I then Shut down everything and tested it from scratch. RPi pluged into Always-ON, Printer plugged into Normally-ON.

I flipped the main switch on the IOT Power Relay.

OctoPrint Booted while the Printer Booted.
The Printer Beeped after about 1 mins 15 secs.
I then locally logged into OctoPrint and it was connected to the Printer. If I go to 'terminal' in Octoprint and type M81, The printer turns off correctly.
WHOO HOO.....

Now the bad.

The lighting bolt is not lit up. If I click the button, the printer instantly turns off without a warning dialog. If I click the button again, it give mes a dialog that im going to turn the printer off, but actually Im trying to turn it on. If I toggle the invert checkmark under PSU Control, then the printer turns off on boot up.

kantlivelong, 2018-02-27:
Run `gpio -1 read 40` with the supply on and then off. What does it return for each?

4RnrRick, 2018-02-27:
When Printer is on:
gpio -1 read 40
Returns "0"

When Printer is off:
gpio -1 read 40
Returns "1"

kantlivelong, 2018-02-27:
Enable invert on switching pin

4RnrRick, 2018-02-27:
If i do that, on boot, the printer will turn off mid mid stream. Like the RPi is sending a signal on GPIO 
activating the relay in the IOT Power Relay.

4RnrRick, 2018-02-27:
yeap - it did. just tested it again to confirm.

kantlivelong, 2018-02-27:
Although not required you generally want HIGH(1) to be ON and LOW(0) to be OFF. In your case you can edit the script on line 7 and 11. Swap the 0 and 1 on each.

4RnrRick, 2018-02-27:
So I swapped the 1 and the 0 . And then I toggled the "invert" back off (un-checkmarked) because the first test didn't work. I now tested a second time.

It booted good, printer beeped, Lightning bolt is green, but when I click the button, it does show the warning but it doesn't toggle the GPIO pin state now, the Printer stays powered .... but it does disconnect. once disconnected the light is still green. and now its giving me constant errors on trying to re-connect.

I hope this makes sense to you...

4RnrRick, 2018-02-27:
I tried to use Invert function and use the Normally-Off outlets on the IOT Power Relay, but I was having issues with the lighting bolt button no working that way also. its all really fustrating for me as i thought this was going to be a really nice device to use.

4RnrRick, 2018-02-27:
I tore apart the CR-10 Control box this morning thinking there might be a sleep function to the power supply like a ATX PSU, but I don't see anything. Now I'm thinking I might be better off to just developed some scripts using the System Command Editor Plugin.  I also got two in there for turning the RPi Camera on and off. but I don't know now to make the printer at the end of the printer run that script in that utility.

4RnrRick, 2018-02-27:
I decided to stop here as it currently beyond my knowledge to get OctoPrint to talk to the IOT Power Relay. I've just ordered a TP-Link Smart plug to use with the https://github.com/jneilliii/OctoPrint-TPLinkSmartplug plugin. Thanks for your quick response and help, much appreciated. Maybe someone else can gain knowledge from these attempts.

closed 2018-02-27 by kantlivelong