kantlivelong/OctoPrint-PSUControl #98: Request: Turn off printer when temperature is above limit

loidor, 2018-03-08:
As a safety feature, should a mosfet fail and the hotend or worse, the bed, run with full power and possibly burn stuff up, it would be awesome to have the server cut the power.

kantlivelong, 2018-03-09:
Hi @loidor,

I like the idea but I'm not entirely decided on if it should be in this plugin. Marking as a request for now.

loidor, 2018-03-09:
Sure :) I'll probably fork this and hack it in (since I don't think I have a use for waiting until cooled, I'll just swap out some operators), but I thought it might be a useful feature for more people than me.

GofX67, 2018-07-17:
Thanks for your work, but ...
that would really be a great extension!

Your plugin "knows" the temperatures of hotend and heatbed and in cause of overtemperature (as configured in the menu), simply shut off the printer.

ej0rge, 2019-03-05:
Bumping this because i think it would be great if there was a way to power off the printer automatically when the printer reports a thermal runaway error.

stevenshizzle, 2020-04-23:
Temperature Failsafe plugin

ej0rge, 2020-04-25:
> Temperature Failsafe plugin

That could indeed be used to power off the printer when a temperature is exceeded. 

Personally i was hoping for more flexibility in what octoprint does when the printer reports Error: or !!. Since i have given octoprint the ability to cut the power from the printer.

kantlivelong, 2020-04-25:
I've decided that this is not a feature I would implement into this plugin especially since the Temperature Failsafe plugin exists. #43 would pair nicely with the Temperatue Failsafe plugin. I'll likely take a look into it again.

I've also looked into providing internal API access to this plugin from other plugins but haven't worked what I want to do with that.

Going to close this specific feature.

closed 2020-04-25 by kantlivelong