kantlivelong/OctoPrint-PSUControl #100: PSU Health Pin (Feature Request)

mrwizerd, 2018-03-13:
Hello,

I would like to have the ability to plugin the PSU Healthy (Grey Wire on ATX Connector) to a pin on the Pi (through a 5v-3.3v converter like I do now with the 5v for PSU Status) that when it is low (Power Supply voltages NOT stable) it will trigger an emergency stop as an option.  Not sure if everyone knows but the Grey wire will go logic high when the power supply's 3.3/5/12v lines have stabilized and that is a signal to the Computer that it is OK to boot up.  If the power ever goes out of spec or destabilizes the grey wire will switch to logic low.  So in the event that you power supply starts fluctuating its voltages wildly or for some reason start to surge it would give us the ability to detect it and shut down the system giving a warning that PSU voltages were out of spec. 

I am going to ask a friend to help me create the functionality but I don't know if we will be able to do it and thus is why I am asking.  IF we do manage to get it working I will put in a pull request.

mrwizerd, 2018-03-20:
This is true, and if your Pi has a small battery backup and the printer is printing from octopi it could pause the print and come back afterwards theoretically...  I don't know if it will work like the power panic does, and it would require a backup power option for the pi but there are many of those out there.  I plan to start work on trying to figure this out soon but I just got the MK3 clone built and now I am sick so... I am down for a while

kantlivelong, 2018-03-20:
Funny enough you mention a battery. Someone in the IRC channel brought up using a UPS and it made me realize a simple desktop UPS can be used that sends the pause and turn supply off commands when low enough. Such a setup would allow seamless resuming provided the power isnt out too long. Could even have plenty of time to save the state.