kantlivelong/OctoPrint-PSUControl #102: Hardware button to activate the power supply.

peed6687, 2018-04-04:
Hi, first great work! I have a small problem. I want to use clipper on my new delta and therefore I bought a 7 "touch screen to not have to rely on the PC to operate the printer.The Raspberry is running on the 5V 2A line of the PC power supply, I wanted the voltage for the touch screen but to the normal 5V rail put, so this turns off with the disconnection of the power supply.The problem now is that I no longer get it without a PC would it be possible on the gpio pins a button to install the Raspberry then says switch the power supply and thus the display? I have been some time in the search for a solution, the use of a hardware button to start the power supply via Raspberry there would be in my opinion the absolute best thing would be possible.

greeting
Peter

kantlivelong, 2018-04-04:
If the Pi/printer pcb is on the 5VSB you can wire the screen to the 5V rail and the motors,hotend, etc.. to the 12v. Should work but not sure how the Pi will handle the poweroff of the display.

kantlivelong, 2018-04-04:
> Yes the Raspberry yes .. but not the touchscreen. I can not power the
> touch screen over the Raspberry because it gets too much for the 2A. Also,
> I would like to do that when the printer is off, which is also the display
> off. And if it is off, no TouchUI, so just using PC.

What model touchscreen?

peed6687, 2018-04-04:
its a Waveshare 7" v2.1 LCD Touchscreen.

nionio6915, 2018-04-05:
You might want to double check the power requirements- it should be around 200 milliAmps with the backlight. That is 0.2A...

You can easily add a power button to the raspi- here are a couple of ways that I have been playing around with- 
https://www.raspberrypi.org/magpi/issues/57/  in the ~ middle of the issue, free download
https://howchoo.com/g/mwnlytk3zmm/how-to-add-a-power-button-to-your-raspberry-pi

Patschke, 2018-06-19:
Have you read pull request #28 ?

kantlivelong, 2018-06-20:
@Patschke I have but right now I'm contemplating phasing out RPi.GPIO for a generic SysFS GPIO module which may make the changes incompatible. Currently in the process of moving and settling in so the next version will be a little while from now.