kantlivelong/OctoPrint-PSUControl #103: PSUControl powering ON/OFF issue.

Karjala, 2018-04-17:
My PSUControl is controlling printer ON and OFF through the GPIO pin and is sensing printer ON/OFF state through the other GPIO pin.
 Everything works fine if I turn power ON and OFF from the Octopi only. If printer is powered ON manually from printer side, then it can not be turned OFF from Octopi with the PSUControl. Opposite happens if printer is turned ON from the Octopi and powered OFF manually from printer side. 
 Looks like PSUControl does not poll actively what power state printer is. If printer is powered on manually from printer side, PSUControl does not poll that the printer went ON and that's why it does not let power been turned OFF from Octopi side because it's state tells it is already OFF. And opposite happens when power is switched ON from Octopi and manually OFF.
 Can be nice if this issue is corrected by actively polling printer power state from the GPIO pin.

kantlivelong, 2018-04-17:
This sounds like a logical issue in your circuit. What are you connecting the sense line to?

SysGh-st, 2020-09-02:
Sounds like gpio pins being left in a floating state. Care to share a wiring diagram with all the pulldown and pullup resistors?

kantlivelong, 2021-04-16:
Stale. Closing

closed 2021-04-16 by kantlivelong