kantlivelong/OctoPrint-PSUControl #105: PSUControl shuts down e

g0rill4z, 2018-05-28:
#### What were you doing?
I turned on my printer and let the bed heat up to 80 °C. After 15 minutes (my idle time) my printer just shut down even though the minimum temperature was not reached. 

1. Turn on your printer, connect it
2. Set your heat bed to 80 °C and wait for 15 minutes.

#### What did you expect to happen?
Nothing
#### What happened instead?
it shut down
#### Version of OctoPrint-PSUControl
0.1.7

#### Operating System running OctoPrint
OctoPrint 1.3.8 auf OctoPi 0.14.0 (up to date) 

#### Printer model & used firmware incl. version
Anet A8 Marlin

kantlivelong, 2018-05-28:
Logs?

g0rill4z, 2018-05-28:
2018-05-27 16:02:24,463 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2018-05-27 16:02:24,464 - octoprint.server - INFO - OctoPrint 1.3.8
2018-05-27 16:02:24,469 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Autoselect Plugin (0.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autoselect
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  DisplayProgress (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayprogress
|  Firmware Updater (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Navbar Temperature Plugin (0.9) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
|  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/octopi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Print History Plugin (1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  PSU Control (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol
|  RequestSpinner (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_requestspinner
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2018-05-27 16:02:24,497 - octoprint.environment - INFO - Detected environment is Python 2.7.9 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 900.0
|    ram: 901689344
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: 2B
|      revision: a01041
|      version: 0.14.0
|  python:
|    pip: 9.0.1
|    version: 2.7.9
|    virtualenv: /home/pi/oprint
2018-05-27 16:02:24,498 - octoprint.server - INFO - ------------------------------------------------------------------------------
2018-05-27 16:02:24,459 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 16:17:24,499 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 16:32:24,502 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 16:47:24,504 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 17:02:24,506 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 17:17:24,508 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 17:32:24,511 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 17:47:24,513 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 18:02:24,515 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 18:17:24,517 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 18:32:24,519 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 18:47:24,522 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 19:02:24,524 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 19:17:24,526 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 19:32:24,528 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 19:47:24,530 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 20:02:24,533 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 20:17:24,535 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 20:32:24,537 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 20:47:24,539 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 21:02:24,541 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 21:17:24,544 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 21:32:24,546 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 21:47:24,548 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 22:02:24,550 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 22:17:24,552 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 22:32:24,554 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 22:47:24,556 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 23:02:24,559 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 23:17:24,561 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 23:32:24,563 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-27 23:47:24,565 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... 71 lines cut ...]
2018-05-28 11:50:10,409 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
[... cut ...]
2018-05-28 13:02:24,728 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-28 13:17:24,730 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-28 13:32:24,732 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-05-28 13:44:30,754 - octoprint.util.comm - INFO - Finished in 4255.428 s.
2018-05-28 13:44:30,899 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Operational"
2018-05-28 13:47:24,736 - octoprint.server.heartbeat - INFO - Server heartbeat <3

kantlivelong, 2018-05-28:
Sorry I don't see anything. Please recreate the issue with serial logging and ensure PSUControl has DEBUG enabled.

g0rill4z, 2018-05-28:
Sorry, cannot contribute recreating anything. I just noticed what happened and told you guys.

closed 2020-04-14 by kantlivelong