kantlivelong/OctoPrint-PSUControl #106: Status bar on/off trigger doesn´t work

Tialf, 2018-07-02:
Hi,

it seems, that the statusbar trigger doesn´t work. If have connectet the signal cable to different GPIO-pins, but problem is still the same:

1. connecting signal to GPIO-pin
2. green led on relay is off -> no signal on the connected pin -> thats ok
3. changing in the configuration the GPIO-pin to the wired -> save -> trigger gets imediately a signal and turn the led into green, so this part of your software seems to work.
4. closing configuration
5. clicking on the statusbar icon for on/off -> it changes the symbol, but trigger is not working.

The GPIO-pin isn´t changing its status. If I manually plug in/out the cable it works, but not within your button :(
I´ve tested pin 3, 5, 7 and 11. Always the same game.

2018-07-01 21:58:36,848 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 918192128
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: 3B
|      revision: a02082
|      version: 0.15.1
|  python:
|    pip: 10.0.1
|    version: 2.7.13
|    virtualenv: /home/pi/oprint

PSU Control (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol
2018-07-02 20:31:00,761 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2018-07-02 20:31:00,762 - octoprint.plugins.psucontrol - INFO - Using internal tracking for PSU on/off state.
2018-07-02 20:31:00,763 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2018-07-02 20:31:00,764 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2018-07-02 20:31:00,765 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 3

How can I get a valid log for you? Sry, I am not into such coding things ;)

kantlivelong, 2018-07-03:
Hello,

You can get a fresh log by restarting OctoPrint then replicate the issue. Once done you can download the log from Settings->Logs

daexys, 2018-07-09:
Hello,
I have similar problem. I am using jqc-3ff-s-z single relay. When I setup On/Off GPIO pin (40 in my case) printer tuns on and I cannot turn it off from status bar. To turn it off I must change my 40 pin to some other.

vbz89, 2018-07-28:
having the same problem as described by OP

kantlivelong, 2018-07-28:
@daexys @vbz89 Need logs.

vbz89, 2018-07-28:
2018-07-28 20:07:56,706 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2018-07-28 20:07:56,707 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2018-07-28 20:07:56,708 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 7
2018-07-28 20:08:00,742 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2018-07-28 20:08:04,247 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2018-07-28 20:08:13,564 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2018-07-28 20:08:19,962 - octoprint.plugins.psucontrol - INFO - Switching PSU Off

Whenever I set the pin in the settings, the relay lights up. Toggling the power indicator does nothing. When I select a pin which I have not connected to the relay, the relay led goes off. 

Basically, the switching doesn't seem to be working with the indicator toggle.

I am also using a single relay module.

kantlivelong, 2018-07-28:
It sounds like you have something wired incorrectly.

vbz89, 2018-07-28:
Yup ok, I ran "gpio readall" in the terminal and the toggle indicator is working. There is something wrong with my wiring then. I'll let you know if I figure out what fixes it. Thanks

kantlivelong, 2018-08-04:
@Spyder19 open a new ticket and provide the required info.

kantlivelong, 2018-08-04:
Closing this ticket,

closed 2018-08-04 by kantlivelong