kantlivelong/OctoPrint-PSUControl #107: Automatic switching off doesn't work

rbk1rbk, 2018-07-10:
#### What were you doing?
1. printer off
2. starting a print, printer turns on
3. print finishes, printer keeps running

#### What did you expect to happen?
switch off PSU

#### What happened instead?
PSU still running. PSU was automatically shut down only once, when I restarted OctoPrint service. Manual switch ON and OFF works.

#### Version of OctoPrint-PSUControl
Octoprint 1.3.8, PSU Control probably the latest

#### Operating System running OctoPrint
pi@octopi:~ $ cat /etc/octopi_version
0.15.0

#### Wiring diagram
I use GPIO pin 23 for on/off and pin 24 for sensing, although currently sensing is not configured. Timeout is set to 1 minute.

[octoprint.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/2182068/octoprint.zip)

kantlivelong, 2018-07-10:
Hi @rbk1rbk,

What time-frame should I be looking at? Looking at the end of the log indicates that a print job is still occurring.

kantlivelong, 2018-07-10:
Nevermind I see it.

It looks as if your hotend never goes below the value you specified for idleTimeoutWaitTemp(30). I'd suggest increasing that to something more reasonable like 50C and try again.

rbk1rbk, 2018-07-10:
Hi @kantlivelong 
I changed it to 50 degrees and timeout to 10 minutes, which in turn reset  my printing but nevermind :-) It's been cca 20 minutes since then, please see the logs.
[octoprint.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/2182162/octoprint.zip)

kantlivelong, 2018-07-10:
It shuts off all heaters(hotends/bed) then waits for the hotends to cool to the temp specified. 

It looks like you changed the settings then reset the printer mid print? Can you make sure it's done on a clean/fresh start?

rbk1rbk, 2018-07-11:
Hi @kantlivelong 
I'm wondering, how does it shut all heaters off, I see no command in serial log around this time... But if you believe it is not related, I don't need to know.
I'm sending today's log after octoprint restart. I did a few prints, so logs are slightly bigger.
[serial.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/2186052/serial.zip)

rbk1rbk, 2018-07-11:
Maybe I can run a testcase like in issue #82, if that helps.

kantlivelong, 2018-07-11:
It's possible and likely the case that your idleTimeout is just too short. Try increasing it to a value that exceeds the maximum time it takes to heat your bed/hotend or whichever is the longest operation.

rbk1rbk, 2018-07-12:
Timeout is set to 10 minutes, I have no heated bed, hotend is hot in cca 3 minutes. Why are suggesting maximum time to _heat_? Shouldn't it be to cool down?
Should PSUControl switch off printer after 10 minutes when I turn it on and do nothing?

kantlivelong, 2018-07-12:
The plugin determines timeout by outbound activity. During certIn commands
there is no outbound(also sometimes inbound) comm activity. If your hotend
takes 3 mins to heatup then you have already timed out with a 1 minute
timeout defined.

kantlivelong, 2018-07-12:
... It's also just a suggestion at this point as I don't see a clear reason why it's occuring other than the two options mentioned. If it's still an issue after that then hardware/wiring would be the next thing I'd look at.

rbk1rbk, 2018-07-18:
I think the wiring is OK. I can switch PSU on and off with the bolt icon. Also I noticed, that when I restart octoprint or reboot RPI, the plugin switches PSU off -> I believe this means that:
sensing of PSU status is OK
wiring is OK
printer idle detection is OK

rbk1rbk, 2018-09-05:
I've found the pinpoint. The issue is in the following part of the code:
    def _wait_for_heaters(self):
...
        for heater in heaters.keys():
            if float(heaters.get(heater)["target"]) != 0:
It is because the first heater Octoprint returns is the bed heater, but my printer has no bed heater, and this call never returns. If I put the following there
            if heater == 'bed':
                continue
the automatic shutdown works well.
So I suggest you to check if the bed heater is really there.

nionio6915, 2018-09-06:
I can confirm that this is the case with my unheated bed printer as well, PSUControl will not turn off even though the hotend is cool enough and the timeout has been exceeded.

kantlivelong, 2018-09-06:
@rbk1rbk interesting. OctoPrint doesn't show a temp(even 0/null) for the non-existent bed, correct?

kantlivelong, 2018-09-06:
@rbk1rbk tagging you as well. ^

nionio6915, 2018-09-07:
@rbk1rbk I am trying to understand your suggestion. 

1. With the bed heater DISABLED in both Marlin and Octoprint, there is no reporting of bed temps in the terminal. 

Octoprint settings =>Printer Profiles => Print Bed & Volume => uncheck "Heated Bed"
i get
Recv: T:27.66 /0.00 @:0

2. Looking at the code, I see the syntax that you are suggesting at ~ line 439. 
Where are you suggesting that it get inserted to ?

rbk1rbk, 2018-09-10:
I ruined my temp sensor, so I'm not able to test this for some time...
This is my edit around line 422:
        for heater in heaters.keys():
            if heater == 'bed':
                continue
            self._logger.info("Getting temperature of %s heater " % (heater))
            temp = float(heaters.get(heater)["actual"])
            self._logger.debug("Heater %s = %sC" % (heater,temp))

without the _if_ the code never returns from heaters.get.
"Heated bed" checkbox is unchecked, I'm getting:
Recv: T:25.59 /0 @:0
Maybe it is the Octopi's issue that the call tnever returns?

kantlivelong, 2018-09-28:
well this is a little embarrassing. I did a fix for this a bit ago but apparently didn't merge it.

#92 

Though the fix was intended for setups where heated bed was enabled when the firmware/printer didn't really support it. It appears that the bed temp values will be *None* regardless if heated bed is enabled/disabled. Not sure if this is an OctoPrint, Plugin, or Firmware bug. Will check with @foosel.

foosel, 2018-09-28:
OctoPrint always returns what it gets back from the printer. If the printer includes bed temperatures those will be populated, regardless of whether the printer profile states it supports a bed or not. The profile setting will only be used to 

   * adjust the UI
   * filter out blocking heatups for the bed from comms

You'll still get a default "bed" entry in the temperature reply (as in, actual and target both None) since that makes it a bit easier to process.

kantlivelong, 2019-04-10:
This should be fixed by #126 in **0.1.8**.

closed 2020-04-14 by kantlivelong