kantlivelong/OctoPrint-PSUControl #110: missing schematic in wiki

nionio6915, 2018-07-31:
https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Wiring-Diagrams

The image of the "All There" wiring diagram is missing. 

Can it be recovered?

[image]

mad2xlc, 2018-11-01:
Anything new on this?

nionio6915, 2019-03-05:
I dug this up from my notes. I offer no guarantee on accuracy. 

[image]

svg in the zip file b/c git no likey the extension. 

[file.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/2929834/file.zip)

SysGh-st, 2020-09-02:
> I dug this up from my notes. I offer no guarantee on accuracy.
> 
> [image]
> 
> svg in the zip file b/c git no likey the extension.
> 
> [file.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/2929834/file.zip)

Neat. But keep in mind that most ATX supplies have a rather weak 5VSB (5V, StandBy) supply around a few hundred milliamps to about an ampere at best. Far from enough for a Raspberry Pi. Might work at first, but will burn out the 5VSB supplying IC/regulator prematurely.

kantlivelong, 2021-04-18:
Decided to remove schematics for now.

closed 2021-04-18 by kantlivelong