kantlivelong/OctoPrint-PSUControl #112: Strange issue when i want to turn it on

michalss, 2018-08-29:
Dear support,

I have a strange issue with PSU Control. I have 5VDC rele with 2 leds green and red. Also got Rb PI 3b. Connected as like this :

RELE:
VCC --- > pin 2, 5V rpb
GND ---> pin 6 GND rpb
IN ---> using GPIO pin 8 as signal

My problem is this, when i set gio pin 8 in PSU control config, rele is automatically switch on when i hit save button, without any other interaction. After that it cannot be control with button in menu at all. To turn it off i need to go into PSUControl config and actually switch the gpio pin from 8 to anything else and then rele would switch off. This is very strange. How can this be fixed pls ?

kantlivelong, 2018-08-29:
Hello @michalss,

This tracker is intended for bugs, feature requests, and pull requests. 

You can try the [OctoPrint Forum](https://discourse.octoprint.org) for support.

closed 2018-08-29 by kantlivelong

michalss, 2018-08-29:
But this is a bug with PSU Control..

kantlivelong, 2018-08-30:
Test it with the `gpio` command. Make sure OctoPrint is stopped first.