kantlivelong/OctoPrint-PSUControl #113: Request: USB Power down

MichiTheSailor, 2018-09-26:
Ciao,

First of all thanks for you great job!

Im useing Octoprint 1.3.9 (Octopi 0.15.1) with Psu_control 0.1.7 to control my Creality CR-10s printer.

To power down the printer I implemented an relays board which is connected to Gpio21 on my Paspberry 3.
Switching via PSU control works fine, but as long as usb is still connected (between raspberry and printer), the raspberry still power up the printer as well (not enougth for printing but enougth to let still the processor and display continue to work on the printer.

Consequences:
If I shut down my printer via PUS, it stops printing but printer-software continues...

The only possibility I found was to manually shut down the Raspberry (via Octoprint menu)

It would be really great if you can include:

- switching down USB at the same time as the Gpio port switch to power down, that this will not continues powering up the printer      OR

- defining an additional timeout after printing/inactivity when Raspberry should shut completly down (not nice but better than it keeps partly running)

Thanks and kind regards, 
Michi

kantlivelong, 2018-09-27:
Hi @MichiTheSailor ,

I'm not really sure if this is something that I'd be wanting to implement at this time. However, you can do this fairly easily yourself by using System Commands instead of GPIO. Call a script that toggles the GPIO pin as well as shutting down the port.

tobecwb, 2018-11-17:
this is my case too..

Until now, I'm using System Commands like @kantlivelong commented.
you will need `python` and `uhubctl`.

first, install `python-pip` and `RPi.GPIO`:
`sudo apt-get install python-pip`

then, download and compile `uhubctl`:
```
sudo apt-get install libusb-1.0-0-dev
git clone https://github.com/mvp/uhubctl.git
cd uhubctl
make
```
next, give permission to normal user access the usb
```
sudo nano /etc/udev/rules.d/52-usb.rules
SUBSYSTEM=="usb", ATTRS{idVendor}=="0424", MODE="0666"
sudo udevadm trigger --attr-match=subsystem=usb
```
(**0424** - vendor id of usb hub of raspberry pi 3 b+)

now, we will create two script files.. one for turn on the printer on relay and usb, and another to turn off relay and usb hub.

create a file named: `turn-on.py`
```
import RPi.GPIO as gpio
import os

gpio.setwarnings(False)
gpio.setmode(gpio.BOARD)
gpio.setup(8, gpio.OUT) # change the port number for your case
gpio.output(8, gpio.HIGH) # change the port number for your case

os.system('/home/pi/uhubctl/uhubctl -a on') # full path to the uhubctl
```

create another file named: `turn-off.py`
```
import RPi.GPIO as gpio
import os

gpio.setwarnings(False)
gpio.setmode(gpio.BOARD)
gpio.setup(8, gpio.OUT) # change the port number for your case
gpio.output(8, gpio.LOW) # change the port number for your case

os.system('/home/pi/uhubctl/uhubctl -a off') # full path to the uhubctl
```

now, on plugin options, on `Switching Mode`, change to `System Command`
on the `On System Command` option, use:
`python /home/pi/uhubctl/turn-on.py`

on the `Off System COmmand` option, use:
`python /home/pi/uhubctl/turn-off.py`

JLIspace, 2019-03-11:
I had the same 'problem' with my Ender 3 Pro. I just cut the VCC USb wire so the printer's mainboard won't be backpowered from the pi.

downace, 2020-05-26:
I'm using `gpio` command from `wiringpi` package and [`uhubctl`](https://github.com/mvp/uhubctl) for this purpose. It works nice with my Pi 4B.

*On System Command*: `gpio write 0 1 && uhubctl -l 1-1 -a on`
*Off System Command*: `gpio write 0 0 && uhubctl -l 1-1 -a off`

But I think it would be convenient to allow user to combine two *Switching Method*s. Or add some kind of on/off hooks allowing user to perform some custom actions when switching power

kantlivelong, 2020-05-26:
Whoops I should have closed this by now. Aside from RPi.GPIO (Want to get away from that too) I won't be implementing any platform specific features. As previously mentioned you can write a script and call that.

closed 2020-05-26 by kantlivelong

downace, 2020-05-26:
I'm **not** proposing usage of platform specific features.

Currently plugin allows to select **only one** switching method, and the idea was to allow combining them, e.g. toggle GPIO pin *and* execute system command. It will allow users to just use plugin's GPIO control feature instead of writing Python scripts. Much more convenient.

Anyway, it's not what this issue was initially about, workarounds described above are fine too.