kantlivelong/OctoPrint-PSUControl #114: Setting up GPIO for “switching” and “sensing”

empakoso, 2018-10-28:
#### What were you doing?
1. Installed plugin - entered OctoPrint Setup GUI
2. Set PSU switching and sensing to GPIO
3. Selected pins 21 and 12 respectively
4. Saved the OctoPrint Setup
5. Shutdown and restarted
6. M80 is working normally 
7. Reviewed the PSU switching and sending
8. Found the default settings had returned to “G-code” and “internally”
9. All the rest of the settings had been retained and been assumed to work normally because of point 6 above.

I assume it is just a graphical interface - recall of current settings - issue.

#### What did you expect to happen?
All the settings are left as they were after saving them.

#### What happened instead?
Switching and sensing option reverted from the saved GPIO selection back to original gcode-command and internally selections.

#### Version of OctoPrint-PSUControl
Vrs. 0.1.7

#### Operating System running OctoPrint
OctoPrint 1.3.9 running on OctoPi 0.15.1

#### Printer model & used firmware incl. version
Prusa clone using ZRIB board.

ej0rge, 2019-03-05:
I have seen this behavior as well

thesourcerer, 2019-05-03:
I have the same problem with 0.1.8

Miranio, 2019-06-20:
Confirm the behavior and confirm that it don't affect the saved setting. It is really a display bug.

leoheck, 2019-08-16:
Oh, I think I reported the same here https://github.com/kantlivelong/OctoPrint-PSUControl/issues/138

kantlivelong, 2019-08-16:
Thanks for confirming everyone. I've been a bit occupied with other projects but will eventually get back to this. Will definitely fix any major issues that should arise though :smile:

kantlivelong, 2021-04-05:
This should no longer be an issue as of 27e66a2 in devel

kantlivelong, 2021-04-16:
v1.0.0 released.

closed 2021-04-16 by kantlivelong