kantlivelong/OctoPrint-PSUControl #119: Add another Power ON/OFF button

3Dciq, 2019-01-06:
What were you doing?
I'm running this plugin over GPIO and a relay to turn PSU ON/OFF and it's working great.
I was wondering if there is a way to add a similar button for light controls the same way or is it only possible to control one thing with this Plug-In and I have to install this Plug-In one more time like it's described in Issue #116 116?

Version of OctoPrint-PSUControl
0.1.7

#### Operating System running OctoPrint
OctoPrint 1.3.10 , OctoPi 0.15.1
OctoPi (Raspberry 3B+)

Printer model & used firmware incl. version
Anycubic i3 Mega 
ANYCUBIC Mega_ULTRABASE_V1.1.2

kantlivelong, 2019-01-06:
[OctoPrint-LightControl](https://github.com/kantlivelong/OctoPrint-LightControl) should do what you want.

penright, 2019-03-08:
> [OctoPrint-LightControl](https://github.com/kantlivelong/OctoPrint-LightControl) should do what you want.
Thanks for the code.
I have never done a pull request, don't know how to PM you, so this is the only place I know how to communicate. I am open to learning how. :-)
Light Control has an issue if you invert the pin. In the check_light_state you were not compensating for the invert flag.  I just added a brance and reverse what you were testing on r.
Since the light was a flag it kept being set back to the same state.

Here is the code:

```
   def check_light_state(self):
        old_isLightOn = self.isLightOn

        self._logger.debug("Polling Light state...")
        r = 0
        try:
            r = GPIO.input(self._gpio_get_pin(self.onoffGPIOPin))
        except (RuntimeError, ValueError) as e:
            self._logger.error(e)
        self._logger.debug("Result: %s  invertonoffGPIOPin: %s" % (r, self.invertonoffGPIOPin))

        if self.invertonoffGPIOPin:
            if r==0:
                self.isLightOn = True
            elif r==1:
                self.isLightOn = False
        else:
            if r==1:
                self.isLightOn = True
            elif r==0:
                self.isLightOn = False
        
        
        self._logger.debug("isLightOn: %s " % self.isLightOn)

        self._plugin_manager.send_plugin_message(self._identifier, dict(isLightOn=self.isLightOn))

```

kantlivelong, 2019-03-08:
Good catch. I'll accept a PR there. I'm usually in #octoprint on Freenode if you have a question.

kantlivelong, 2019-03-08:
Closing this.

closed 2019-03-08 by kantlivelong