kantlivelong/OctoPrint-PSUControl #121: [Request] Re-connect after power on

Esteban16899, 2019-02-26:
I would like the plugin to "re-connect" (diconnect then connect if its necessary) to the printer after is turned on. It would be useful to have a configurable delay of this action. This is because sometimes octoprint sends a command that makes a high amperage request such as preheat the nozzles or move the motors, in that cases the pi cant provide the current for the controllers to act and the voltage drops causing the reset on the printer controller, therefore the pi still tries to send commands and disconnects by saying "printer not working properly). I experimented with this on an ender 3. PSU is controlled by a relay in series with one of the 220v wires.

ej0rge, 2019-03-05:
I would appreciate this functionality as well. 

In my scenario, my printer and the raspberry pi are powered by a pair of old fileserver power supplies, modified and wired in series. 5A of 5v standby power is always available and powering the Pi. 

I used to also use that 5v rail to power the logic on my controller and use gcode to turn the 24v on and off, but when i upgraded to trinamic drivers that was no longer possible because trinamic requires VMOT to be powered before VCC. 

I now use a DPDT relay to turn the power supplies "on" and "off", so the control board loses power on power-off and thus disconnects. And i have to manually connect again after power-on.

kantlivelong, 2019-04-10:
@Esteban16899 

Implementing this feature sounds more like a workaround hack rather than fixing the actual problem which is voltage drop. I'd argue that having an always-on power supply for the controller like like the one used for the Pi would be a far better solution.

Esteban16899, 2019-04-10:
I undestand your sight on this, but thik of this case: 
Raspi Always ON via Switching PSU
PSU control - Commanding a relay that shorts 3D printer PSU
RaspiPSU isn't the same as 3DPrinter

So, Raspi sends a command via serial that tells Steppers or Heaters to run. 
This causes the voltage to drop because PSU control activated the relay after the command  (gcode) was sent, so the 3D Printer board resets and looses conection before gaining Power. This happens in microseconds, but the relay actuation is slower than the serial comunication. 

After writing all this, that could be useful, i dont know what would be the reseting and reconnecting behavior. I'll wait for your comment. I'm a lottle bit lost now :)

kantlivelong, 2019-04-10:
So if I understand right you are powering the controller via the Pi USB power and then when the PSU is switched on the controller also gets power from that PSU. 

Assuming the controller has protection and can receive power from two sources I'd be curious if you would be able to modify the USB cable as follows:

```
                                    /---> New Power Supply Ground
Pi USB Ground   <-----------------------> Printer Controller Ground
Pi USB Data +   <-----------------------> Printer Controller Data +
Pi USB Data -   <-----------------------> Printer Controller Data -
N/C             <-----------------------> Printer Controller 5V
                                    \---> New Power Supply 5V
```

Esteban16899, 2019-04-10:
If I do so and correct me if I'm wrong, is the same as using the same supply I have for the pi, but instead of running through the pi it would be attached directly. I see no difference. I really don't know if the 5V regulation on the pi limits the current but I think not.
Now I'm thinking I don't understand why turining on a heater or moving a stepper would make that signigificant voltage drop, I guess the interfaces aren't as simple as I think they are (by interfaces I mean stepper controllers and heater regulator circuits). Next week's I'll be playing whith the circuitry and I'll tell you about the results.

mneumark, 2020-09-28:
I'd also like this feature. I am using a smart plug to control power to the PSU. I wrote a script that can turn it on/off and check status. But when it turns on, it still doesn't connect back up to the printer. Its the last piece of the puzzle that is missing.

kantlivelong, 2021-04-16:
Connect on power on has been in since 0.1.10. Closing

closed 2021-04-16 by kantlivelong