kantlivelong/OctoPrint-PSUControl #124: Can't use this from Cura

Tailslide, 2019-03-11:
When using this from simplify 3d I have simplify call a batch file to talk to the PSU Control API and power on the printer before printing.  Using Cura with the octoprint plugin I can't duplicate this functionality since there is no way to run a batch file and the octoprint plugin for cura doesn't have support for PSU Control.

I am unable to use the auto on feature of PSU Control since I have the rambo board which does not function until the power supply is powered on (it doesn't run off USB power).

I can think of a couple solutions to this:

1) Have a feature to automatically power on and start a print when a new print is queued.
2) Have a PSU Control plugin for Cura.

  Thanks for the plugin I use it a lot.

chriswal, 2019-03-14:
Hi,
you can use System Command to call a batch file for power on/off as seen in the attachment.
i use curl for sending http requests

[image]

kantlivelong, 2019-04-10:
Hey @Tailslide,

> Have a feature to automatically power on and start a print when a new print is queued.

When a job is sent to OctoPrint via the API whilst the connection to the controller is the disconnected state OctoPrint will just store the file instead of queuing a job for later printing.  I'm sure something could be made to handle this but it would require an assumption that _any_ uploaded file should be printed when not connected which is not always correct.

> Have a PSU Control plugin for Cura.

I don't use Cura but if it can be done I'd be happy to take the PR if it fits.

Tailslide, 2019-04-10:
For me any file uploaded should be printed right away because that's my workflow. I don't have any steps I need to do in between sending it to print from the slicer and it actually printing.. I just want it to print.  So, the option would be nice.

I don't use Cura either mostly because of this issue but the new version has some very cool features like ironing the top layer of a print to smooth it, and it has GUI support for setting acceleration, etc separately for infill and middle or outer perimeters IIRC

If I get ambitious I might try a plugin for Cura at some point but it'll probably be it's own thing that calls your API.

Anyways, thanks for your comments !

kantlivelong, 2020-04-13:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/pull/124 should do what your looking for.

closed 2020-04-13 by kantlivelong