kantlivelong/OctoPrint-PSUControl #127: [Request] Mosfet Failure (High Bed/End Tempurature limits) shut-off the PSU. Would it be possible to set Bed and Hotend High Temp Limits to prevent Mosfet failure temperature run-away? Temp protection in Marlin seems to only protect with Termistor failure. Recently I had a Mosfet fail, and it caused my hotend to go over 300 Degrees, thermistor read the heat, but the voltage could not be turned off due to the Mosfet failure.

Sevcav, 2019-04-13:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Version of OctoPrint-PSUControl

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

#### Wiring diagram

kantlivelong, 2019-04-13:
Seriously? Fill out the ticket proper.

nionio6915, 2019-04-13:
Sevcav- why do you think this is PSU control's issue? You need to check your Marlin configs. 
I had same thing happen- I don't trust Marlin anymore. You might want to google more and find  something like this- 
https://plugins.octoprint.org/plugins/temperaturefailesafe/

Sevcav, 2019-04-13:
Not a PSU control issue, sorry never meant to state that. However as a request, turning off the PSU based on a MOSFET failure, voltage will always be applied to bed or hotend with the failure. PSU control turning off the Power would be a nice protection feature for this type of printer failure.

Marlin is always checking the Thermistor for failure, but when a MOSFET fails, the heaters will constantly be applied voltage regardless, nothing beyond removing the power would as I understand it protect from thermal run-away.

Thank you for the link for the other plugin, but unless securing the PSU power I don’t see how that protects anything, and is the same as what is in Marlin 1.9 currently.

Again, this is not an issue with PSU control, it works great, and did not cause my MOSFET failure. In fact post the failure I installed the Ardunio relay and PSU control plugin in hopes to protect against this dangerous situation when a MOSFET fails.

Thank you for your time and consideration,
Sevcav

https://plugins.octoprint.org/plugins/temperaturefailesafe/

nionio6915, 2019-04-15:
Sevcav- 

look at that plugin again, it will trigger a script if the temp threshold is exceeded. 

by using a one or two line script, calling on something like pigpio or wiringpi to write to invert the gpio pin you are using to control the psu, you ARE controlling the PSU state. 

http://wiringpi.com/the-gpio-utility/