kantlivelong/OctoPrint-PSUControl #129: Installiation Issue

sinaneryigit, 2019-05-20:
Hi, 

Plugin is not being installed. 
I get this error attached.

What can be the issue?
[image]

kantlivelong, 2019-05-20:
Fill out the issue template.

sinaneryigit, 2019-05-28:
This is the terminal error I get:
pi@roboko:~ $ sudo pip install "https://github.com/kantlivelong/OctoPrint-PSUControl/archive/master.zip"
[sudo] password for pi: 
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/kantlivelong/OctoPrint-PSUControl/archive/master.zip
  Downloading https://github.com/kantlivelong/OctoPrint-PSUControl/archive/master.zip (122kB)
     |████████████████████████████████| 122kB 231kB/s 
Collecting OctoPrint (from OctoPrint-PSUControl==0.1.8)
  ERROR: Could not find a version that satisfies the requirement OctoPrint (from OctoPrint-PSUControl==0.1.8) (from versions: none)
ERROR: No matching distribution found for OctoPrint (from OctoPrint-PSUControl==0.1.8)

Other info you requested:
**What I was doing?**
1. I tried instaling it from Web UI, didn't work. 
2. I tried Mac Terminal, no luck.
3. I tried removing and re-installing pip, didn't work.

**Version:** OctpPrint 1.3.11 running on OctoPi 0.16.0ZA<>za<>

Printer: Tevo Tarantula, Marlin 1.1.0 RC7

**LOG FILES:**
https://sinaneryigit.com/share/plugin_pluginmanager_console.log
https://sinaneryigit.com/share/octoprint.log

Thanks in advance.

sinaneryigit, 2019-05-28:
By the way, I tried a minor hack. Deleted RPi.GPIO 0.6.3 requirement from requirements.txt file and installed the plugin locally, successfully. But, I don't know if it'll work because I didn't tried it yet. And "Sense GPIO" setting is not there as you show on settings screenshot. :) 

So, waiting for your help. Thanks.

sinaneryigit, 2019-06-15:
Hey! Are you going to look into this? What info do you want more that I didn't supply yet?

kantlivelong, 2020-04-14:
I'll assume this was resolved as it seems like an environmental issue. If not please check back.

closed 2020-04-14 by kantlivelong