kantlivelong/OctoPrint-PSUControl #131: Feature request: Make the psu power on at boot up

wbarber69, 2019-05-23:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Version of OctoPrint-PSUControl

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

#### Wiring diagram

chriswal, 2019-08-23:
?
?
?

stevenshizzle, 2020-04-23:
Connect your PSU to the Normally Open node of the relay.

electronicm, 2021-08-31:
Add to /boot/config.txt:
'gpio=XX=ip,dh'

electronicm, 2021-09-01:
Same as my #222

safetyABJ, 2022-08-20:
I have a working solution for this that's somewhat inelegant but effective. It involves creating a bash script containing an API command for PSUControl to turn on the PSU and then using OctoPrint Events Manager to run that script on Startup.

1. Get your Global API Key from OctoPrint: **Settings > API**
2. SSH into OctoPrint: **ssh pi@octopi.local**  *or whatever your OctoPrint address is
3. Create the bash script: **sudo nano /usr/local/bin/PSUon.sh**
4. Enter this into the script. Be sure to replace YOUR_API_KEY with your Global API key:
```
#!/bin/bash
until $(curl --fail -s -H "X-Api-Key: YOUR_API_KEY" -X GET http://octopi.local/api/plugin/psucontrol | grep true -q); do
    sleep 1
    curl -s -H "Content-Type: application/json" -H "X-Api-Key: YOUR_API_KEY" -X POST -d '{ "command":"turnPSUOn" }' http://octopi.local/api/plugin/psucontrol
done
```
5. Save script and exit: **Control-O, Control-X**
6. Make the script executable: **sudo chmod +x /usr/local/bin/PSUon.sh**
7. Go into OctoPrint: **Settings > Event Manager and click +Add**
8. Use these settings: Event **Startup**, Command **PSUon.sh**, Type **system**
9. Click **Close** then **Save**. 
10. Reboot the system.

The inelegant part is where the script keeps executing the "turnPSUOn" API command until the plugin responds that it is on. I found that if the command were simply issued once it wouldn't take and that I had to put a sleep duration in before it, but the required length of sleep wasn't consistent.