kantlivelong/OctoPrint-PSUControl #133: [Request] : Hardware button

Miranio, 2019-06-14:
First I really love the plugin and this is why I'm trying to implement such a feature ( but I'm horrible at coding :( especially with object oriented coding ... I couldn't save my life ).

I have looked up pull request #28 and it is not what I have in mind for multiple reason. Mainly that it check for momentary switch.

I wish to use a led ring latching switch to control the printer as well as using it in the web interface. I could use the enclosure plugin and it would work but I like the confirmation window and the timer of this pluging.

So the improvement would be as follow

The icon in the title bar display the current state of the PSU. This state could be copied to a gpio pin for connecting the led ring.

The latching switch would be connected to another set of gpio where if the switch would work as follow.

1- Starting with the PSU OFF
2- Switch from open to close : put the psu on and update the web interface button as well and the led ring light up ( or down to be discussed )

1- Starting with the PSU OFF
2- Psu triggered by web interface : put the psu on and LED ring light up ( or down )

1- Starting with the PSU ON 
2- Switch from open to close : nothing happen since the psu is already on and closed mean on

1- Starting with the PSU ON
2- Switch from close to open : Check if confirmation selected to power off
    2.1 - If confirmation checked : flash LED ring for 5 sec to give time to put switch back to closed position and if switch is back to closed, do nothing as the psu is supposed to be on
    2.2 - If no confirmation : power off PSU and update web interface as well as LED Ring

What do you think ?

Marianne

Miranio, 2019-06-16:
Somehow I've managed to learn how to code and do something that ... as of now is working.

As I said, I'm really new to programming, and a pull request look like I want to push my code in your repository ( feel free to educate me ).

You could find what I did here : https://github.com/Miranio/OctoPrint-PSUControl

Hope it don't make you pull your hairs too much :/

electronicm, 2021-09-02:
@Miranina 
There is now a plugin for Button: [PhysicalButton](https://plugins.octoprint.org/plugins/physicalbutton/)
(use System: `~/oprint/bin/octoprint plugins psucontrol:toggle`)

The LED ring could be controlled via  [LightControls](https://plugins.octoprint.org/plugins/LightControls/).
(use Connect / Disconnect)