kantlivelong/OctoPrint-PSUControl #135: [Request] PSU always ON while start

modmt, 2019-07-21:
Hello all,

I am using the RPI3B in combination with Ramps 1.4 board and Klipper FW.

For different reasons I want that the PSU is alwas ON while start (e.g. the TMC2208 with UART connection needs power all the time while start to read/write registers, otherwise I will get the standard settings for TMC... :-().

The currect behaviour with my combination is that the PSU starts for 30 seconds or so an then the pin 3 (the pin on rpi which i selected) changes it status to shutdown the PUS.  
So when octoprint is fully loaded the PSU is OFF.

I need it always ON as first state after plugin the power.

In Attachment you can find the setting, but I tried also different settings with no success.
The whole rest of the functions are working great.

Is there a possibility to implement that?

Thank you for your help.
Matthias

[image]

chriswal, 2019-08-23:
try
https://github.com/chriswal/OctoPrint-PSUControl/archive/initial_state_config.zip
and set the initial state power on option
couldnt test it with gpio ..so tell me if it works for you

[image]

wick3r, 2021-04-18:
I second this request :) .

kantlivelong, 2021-04-24:
So if I were to implement this I think I would have an dropdown with the following options:

State On Startup:
- No Change
- Always On
- Always Off

Raspberry Pi users would also want to ensure they are using the right GPIO pin to match. See https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf pg102

electronicm, 2021-09-01:
Same problem as my #222

ChrisButt, 2022-01-14:
I have the same problem.
My printer is connected to the NC of the relais, so the printer will run without Octoprint.
The relais is connected to GPIO 17, so when you pull it high, the printer turns off.
With this configuration you need to connect the relais to GPIO 9-27, because these are LOW on powerup.
After the RPI booted, PSU Control turns off the printer. Everything except of this works fine, i just want to not automatically switch the printer off after booting.

sidamos, 2023-01-19:
I am using this plugin together with the OpenHAB sub-plugin and IMHO it is pretty bad that OctoPrint always switches the PSU to OFF when starting. It should just leave it alone. I had a print running for 10 hours, the PI was not connected to the printer and I started the PI and then it switched off the printer (via OpenHAB and smart power plug) and I lost my print.