kantlivelong/OctoPrint-PSUControl #136: PSUControl enforces pin change after RP reboot

juliandroid, 2019-07-21:
#### What were you doing?
```
                                 GPIO27 (default state on boot is 0v)
                                      |
                                    (0)
Printer   ->  NC Relay   => Printer is ON
```
By using NC relay as default and also GPIO27 on RP 3B+ with OFF state by default on boot I expect the with RP3 ON or OFF / connected/disconnected to never interrupt power cable by default. That works so far. No matter if I power cycle the RP!

With PSUControl set as:
GPIO Mode: BOARD
On/Off GPIO pin: 13 (that is GPIO27)   and Inverted IS checked ON.
No other checkbox is turned on (i.e. auto PSU ON/OFF is turned OFF)

#### What did you expect to happen?
The icon is ON (i.e. Printer is still ON, the GPIO27 is still OFF and the Icon is ON due to Invert flag).
If I turn the RP off (or power cycle it), the Printer continue to be ON (which is good), once the Octoprint and PSUControl is loaded after reboot, I expect the PSUControl to keep the printer ON, i.e. not touch the GPIO27 (i.e. leave it OFF).

#### What happened instead?
Once the Octoprint and PSUControl is loaded after reboot, the PSUControl turns the printer OFF, i.e. sets the GPIO27 ON, which effectively turns the Printer OFF.

If I set the Invert flag to OFF (i.e. unchecked) then the PSU Control works correctly, i.e. does NOT turn the Printer off (when started the Icon is always grey no matter of the GPIO27 state), however now the Turn On/Off logic of the printer is wrong!

What is happening is that the PSUControl switches the GPIO27 ON (when the Inverted is checked/ON) although I haven't selected checkbox to automatically power On on boot.

Also GPIO27 is OFF by default on boot and if I leave the Invert flag On/checked then I expect that the boot state OFF is now equivalent to ON (green icon) and I don't expect the PSUControl to touch it, i.e. I expect the PSUControl NOT to touch the current state of the PIN if I haven't instructed the plugin to do anything after reboot.

#### Version of OctoPrint-PSUControl
0.1.8

#### Operating System running OctoPrint
OctoPrint 1.3.11 running on OctoPi 0.16.0

#### Printer model & used firmware incl. version
CR-10S Pro, TinyMachine B7

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG
https://pastebin.com/guVDLaNQ

#### Wiring diagram

```
                                 GPIO27 (default state on boot is 0v)
                                      |
                                    (0)
Printer   ->  NC Relay   => Printer is ON
```

chriswal, 2019-08-23:
what is your sensing configuration?

juliandroid, 2019-08-23:
Sensing is internal. Is that wrong in my use case?

chriswal, 2019-08-23:
as the internal state is off at the beginning (off is low in the normal setting and high at the inverted setting) so even if the power supply is on the state is off.

chriswal, 2019-08-23:
you can try my quick fix
install the plugin with this url
https://github.com/chriswal/OctoPrint-PSUControl/archive/initial_state_config.zip

you have to configure the initial power on after restart and your gpio pin as inverted.
i cant test it because i have no gpio controlled power supply

juliandroid, 2019-08-23:
> you can try my quick fix
> install the plugin with this url
> https://github.com/chriswal/OctoPrint-PSUControl/archive/initial_state_config.zip
> 
> you have to configure the initial power on after restart
But if I printer was left "off" and I restart the RP then the PSUControl will turn the printer on?

Ideally the PSUControl shouldn't try to switch the printer ON or OFF upon restart unless asked to.
For example if I print from SDCard and I start/restart the RP - I don't want the PSUControl to kill the print. Also If the last state of the printer was OFF and I restart the RP - I don't want PSUControl to start the printer.

juliandroid, 2019-08-23:
[image]
With this configuration I acheave everything that I want, **except** for the Icon/state. Right now when the Icon is "off" the printer is On. When the Icon is On, the printer is Off and I cannot find a way to "invert" this and keep the exact same behaviour.

chriswal, 2019-08-23:
psucontrol has a state at the beginning .. normal this is off
so when we talk about gpio:  normal is    low is power off     high is power on
inverted  it is    low  power on     high power off
so if the state is off ...pin will be initalised to high  ...to represent power off
this option doesnt turn on the power ..it only changes the internals state to power on
and reverses the initialisation of the gpio pins

so restart  state is power on  your gpio is inverted power on is low  so its initalised to low

if you use a pin on the rasperry pi which is low on boot, and your logic is low is power on ..the yes reboot will turn the printer on

chriswal, 2019-08-23:
[image]

juliandroid, 2019-08-23:
> psucontrol has a state at the beginning .. normal this is off

That GPIO27 is low by default (on boot), so if the "default" value relates to the "inverted" flag then my problem will be solved? I.e. the "psucontrol state at the beginning" to be on or off depending of the inverted flag?

chriswal, 2019-08-23:
so if gpio 27 is low on boot you need psucontrol to be low at start
because your power is on, on low ,you need to switch the starting internal state to on
because your power is switched on with low you have to invert the gpio pin

that means if you reboot the rasperry pi your power is on and the psustate is on

kantlivelong, 2019-08-23:
Internal sensing should really just be called None. This is why I don't touch initial state as its completely dependant on the hardware configuration used. Use GPIO or Command based sensing and wire you're On/Off pin appropriately for the result you want.

chriswal, 2019-08-23:
but that doesnt work with the internal sensing

kantlivelong, 2019-08-23:
Then don't use it. Period

kantlivelong, 2019-08-23:
Basically if you want it to **NOT** be reliable use Internal.

chriswal, 2019-08-23:
yes you are right
but how you are wiring it (and configure) if you want the power to be on at boot?

kantlivelong, 2019-08-23:
That can be done by using command sensing and switching. Make a simple
script to send a pulse over the GPIO and on the other end of the pin set up
a latching relay. On boot  you can have a script to check the power state
and send a pulse to turn on if neccesary.

chriswal, 2019-08-23:
yes i know how i can do that with a relay and scripts ..the question was how to do it with psucontrol
but this is offtopic ..so i think we should stop here..
by the way in the next release of the octoprint cura plugin there is an option to turn on the printer with psucontrol

kantlivelong, 2019-08-23:
Something that supports that could be implemented as it can be hardware
independant.

- Option for GPIO based command to send a pulse.

Not sure how I’d feel about implementing the on on start though.