kantlivelong/OctoPrint-PSUControl #138: [Minor Bug]

leoheck, 2019-08-16:
[image]

I set the "Switching method" to "GPIO pin", it is working fine. But when I go back to the PSU Control config page, it shows the default "G-Code Command". In other words, it is not saving the state of this field.

#### What did you expect to happen?
To see the "Switching method" to "GPIO pin" as it was set.

#### What happened instead?
It is displaying the same "G-Code Command" string. The behavior is OK, the issue is only the string that is not being stored.

#### Version of OctoPrint-PSUControl
0.1.8

#### Operating System running OctoPrint
RaspiberryPi 3
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.9 (stretch)
Release:	9.9
Codename:	stretch

#### Printer model & used firmware incl. version
Ender 3

chriswal, 2019-08-22:
when i try to set gpio it works on my system. octoprint 1.3.11 and psucontrol 0.1.8.
i use a system command for switching --so i didnt start from gcode.
try to set a system command before and look if its saved. then the try the gpio again.

TazerReloaded, 2020-06-17:
Still having this bug in 0.8.9. Steps to reproduce:
- set to "GPIO pin"
- save settings
- reload page
- open settings, it now says "System command" again, sometimes just one option, sometimes both

leoheck, 2020-06-17:
I agree, there is something strange with this page it does not load the right settings when coming back to the page.

ccanevit, 2020-09-16:
I'm experiencing this issue as well, PSU Control (0.1.9), OctoPrint Version 1.4.2. I set this to GPIO pin, and it keeps reverting to G-CODE. Although it works when I manually press the PSU control from OctoPrint, it won't turn the PSU on on startup of the server.

This causes an issue, since my printer (Ender 3 v2) will not accept any G-Code commands unless I power it via usb or have the PSU on.

kantlivelong, 2020-12-16:
This should be fixed in 0.1.10

closed 2021-03-09 by kantlivelong