kantlivelong/OctoPrint-PSUControl #139: [Request] Connect Serial

leoheck, 2019-08-16:
Using version 0.1.8

It would be really nice, to (re)connect the serial automatically, when PSU is turning on.

chriswal, 2019-08-21:
you can use my fork
https://github.com/chriswal/OctoPrint-PSUControl
URL to install: https://github.com/chriswal/OctoPrint-PSUControl/archive/master.zip
be aware that with switching off and on it can happen that the serial port in octoprint changes.
so use the serial by-id ..example /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
thats my printers serial id. you have to look up yours.

nionio6915, 2019-08-22:
You mean like the PortLister plugin does?  Or with serial:action via Action Trigger ?

leoheck, 2019-08-23:
I mean, automatically connecting the Octoprint to the serial when clicking the flash icon.
My printers (Ender 3 and Ender 5) remain disconnected after turning the motors on.

[image]

chriswal, 2019-08-23:
[image]

kantlivelong, 2019-08-23:
@chriswal wouldnt this feature be better suited as a separate plugin or an
enhancement to PortLister?

chriswal, 2019-08-23:
we talked about this when i made the pull request .. i switch on the printer with psucontrol..and i didnt want to be dependent of a second plugin for the connection. as you already used the disconnect from octoprint i only used the connect function to connect after power on ...its the same as when you press the button connect on octoprint ..so why another plugin?

kantlivelong, 2019-08-23:
I vaguely remember that. Primarily so people dont need to use PSUControl
for automatic [re-]connection of serial printers.

chriswal, 2019-08-23:
> Primarily so people dont need to use PSUControl
> for automatic [re-]connection of serial printers.

i dont understand what you are saying
they dont need to use it for automatic  (re-)connection they can use it if they power on with psu-control

chriswal, 2019-08-23:
this is the connect
self._printer.connect(self.autoConnectPort,self.autoConnectBaudrate,self.autoConnectPrinterProfile)
and for this i should make a new plugin ?

kantlivelong, 2019-08-23:
Yes or contrib to PortLister. Instead of that block of code being run when
a user clicks the power button the plugin could poll the port and determine
when it is ready for connection, or alternative just repeatedly try to
connect. Solves the auto reconnect problem for users with and without
PSUControl.

chriswal, 2019-08-23:
i dont understand your point ...but i accept it ...
i see it different ...

wish you a nice weekend and thanks for your great plugin

kantlivelong, 2019-08-23:
Well thats the beauty of open source :)

Maybe I’ll work on such a plugin at some point.

leoheck, 2019-08-25:
Hi, I was not able to get the conclusions here. 

However, I really appreciate @chriswal's new settings [here](https://github.com/kantlivelong/OctoPrint-PSUControl/issues/139#issuecomment-524411866) since all we need is a single button to turn the printer on and to connect the serial port.

chriswal, 2019-08-26:
> wouldn't this feature be better suited as a separate plugin or an
> enhancement to PortLister?
for the conclusion i translate:
this feature is better suited as a separate plugin or an enhancement to Portlister.

i do not agree to this that is why i have the fork with the changes.

chriswal, 2019-08-27:
@nionio6915 if you have started the print you have a lot time for the outside world. ;-)
yes you are right but it depends on the workflow.. i wanted to turn my printer on when i send the data to the printer. so i implemented this in the cura plugin. i slice the part, then when i send it to octoprint, the plugin turns on the printer , and send it to octoprint and start printing.

when i turn on the printer with the octoprint webinterface .. i can also press the button to connect it (then i need no automatic connection) but if i want an automated workflow i need it
after pressing the "send to octoprint" button you have your 30 seconds to enjoy the world ;-)

chriswal, 2019-08-27:
@leoheck its the Octoprint Plugin i dont know in the moment the name in the Marketplace
its this one
https://github.com/fieldOfView/Cura-OctoPrintPlugin/
but you have to wait for the next version..its a basic implementation that turns on the printer and waits 20 seconds and then sends the data.

if you are adventurous you can try to install it from source ;-) (the latest 3.5 branch)

oxcxsxixbxaxn, 2019-09-18:
So I just switched from an RAMPS/mega board to an SKR1.3.  The RAMPS had power from USB so even with PSU in standby, the RAMPS board was always ON, but the 12v was OFF.  PSU Control worked well... Press Print and the PSU 12v jumped on and the the bed was heating....
SKR1.3 receives power from USB (config mode) **OR** from the 12v (print mode) but not both.  So I had to move the PS_ON control to PI's GPIO which is fine... in that switching works great.
But because the SKR is coming from powered off state, firstly it takes a little while to start (ok, timeout option is there), but octoprint is not connected to the serial and does not autoconnect (unless it is restarted itself).

Now I see the exact use case described here, and @chriswal has provided what looks like  perfect enhancement to PSU-Control plugin.  Why is this not accepted in the main branch?  It would be configurable right? so wouldn't affect those with "more conventional or older hardware"...

kantlivelong, 2020-05-12:
Closing this as it's now been requested on FeatHub.

https://feathub.com/kantlivelong/OctoPrint-PSUControl/+2

closed 2020-05-12 by kantlivelong