kantlivelong/OctoPrint-PSUControl #140: [Request/Question] How to control Tasmota plugs

NovaViper, 2019-09-21:
Hi! I'm wondering if there's a way to control Tasmota plugs, just like how the TPLink scripts allows for control over those plugs. If there isn't currently, then I requesting support for Tasmota plugs.

tht, 2020-05-17:
If your Tasmota plugs are connected to MQTT, this will work.

First prepare a configuration file with your MQTT parameters like this. I called mine `.mqtt`

```bash
MQTT_HOST=192.168.1.1
MQTT_USER=tasmota
MQTT_PASS=supersecretpassword
```

Now add this small bash script in the same directory as the configuration file above. I called mine `psu_getstate`

```bash
#!/bin/bash

source $(dirname $0)/.mqtt
MQTT_BASE=tasmota/bas_sm2

(sleep 1; mosquitto_pub -h "$MQTT_HOST" -u "$MQTT_USER" --pw "$MQTT_PASS" -t "${MQTT_BASE}/cmnd/POWER" -m '')&
mosquitto_sub -h "$MQTT_HOST" -u "$MQTT_USER" --pw "$MQTT_PASS" -t "${MQTT_BASE}/stat/POWER" -R -W 3 -C 1 | grep -q ON
```

And a file `psu_switch` (also in the same directory)

```bash
#!/bin/bash

source $(dirname $0)/.mqtt
MQTT_BASE=tasmota/bas_sm2

case "$1" in
    on)
        mosquitto_pub -h "$MQTT_HOST" -u "$MQTT_USER" --pw "$MQTT_PASS"" -t '${MQTT_BASE}/cmnd/POWER' -m 'ON'
        ;;
    off)
        mosquitto_pub -h "$MQTT_HOST" -u "$MQTT_USER" --pw "$MQTT_PASS"" -t '${MQTT_BASE}/cmnd/POWER' -m 'OFF'
        ;;
    *)
        echo "Usage $0 on|off"
        exit 1
esac
```

Don't forget to change `MQTT_BASE` in both scripts to match your configuration. Also make sure the two scripts are executable:

    chmod +x psu_getstate psu_switch

Now add these scripts as *system command* in PSU Control.

kantlivelong, 2020-05-17:
@tht Nice :)

lsaranto, 2021-01-05:
See [this OctoPrint forum post](https://community.octoprint.org/t/how-do-i-control-my-tasmota-sonoff-with-psu-control-plugin/15068/13).

jneilliii, 2021-01-05:
or use my Tasmota plugin...no MQTT server required.

r15ch13, 2021-01-24:
Cool. I don't use Tasmota or MQTT so I adapted @tht's script to use the Home Assistant REST API. :smile:

https://gist.github.com/r15ch13/1ab042d8453818c604c530aae17ac456

kantlivelong, 2021-04-05:
Going to close this now that sub-plugin support has been added in devel via 27e66a21e91cfb855bf1b78006c16cbd0cedd632. Not released yet but will be coming.

closed 2021-04-05 by kantlivelong