kantlivelong/OctoPrint-PSUControl #145: Swithing On RPi4 working ?

arut16, 2020-05-07:
#### What did you expect to happen?

Relay Openning/Closing by control button in octoprint Bar

#### What happened instead?

Nothing ! Reley remain open everytime.

#### Version of OctoPrint-PSUControl

[image]

#### Operating System running OctoPrint

OctoPi Version 1.4.0
Octoprint, Version 0.17.0, running on Raspberry Pi 4 Model B Rev 1.1

#### Printer model & used firmware incl. version

Ender 3 Pro with SKR 1.3 and Marlin 2.0.5.3

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://gist.github.com/arut16/d2f934afa5e332d664a8552556a7faf9

#### Wiring diagram
[image]

kantlivelong, 2020-05-07:
I don't have one to test unfortunately. Doesn't look like debugging was enabled for psucontrol in your logs either.

Is the mode set correctly?
Did you test the GPIO pin with a multimeter?

arut16, 2020-05-07:
Solved : I tried with 3.3V on PiN 1 and it starts to work! Very strange. Normally my Tongling JQC-3FF-S-Z (very common relay) is 5V logical...

[image]

closed 2020-05-07 by arut16