kantlivelong/OctoPrint-PSUControl #146: [Request] PSU off after printer error

jiberan1, 2020-05-23:
I would like use PSU off automatically after printer error. Like safety after thermal runaway,... to disconnect all power from printer.

Till now no option like this.

LongLiveCHIEF, 2020-05-23:
I'm working on a PR for this now.  I came to open a tracking issue for things I need to cover, so if you don't mind, let's close this issue in favor of that issue #147

kantlivelong, 2021-04-26:
Added via e88ee188d94bc17f46a61f5b1a09360236d84a73

Will be in 1.0.5

cosmicc, 2021-04-27:
I tested the dev branch and its working good.  Is it possible to make separate checkboxes for comm error and firmware error to enable?

Also timing is a bit tricky if you use portlister, cause if it tries too quick and comms error, it shuts the power supply off.  But, just a little adjustment in the portlister timing fixed this.

kantlivelong, 2021-04-27:
> Is it possible to make separate checkboxes for comm error and firmware error to enable?

Unfortunately no, error detection is handled by OctoPrint and both are treated as an Error event.

> Also timing is a bit tricky if you use portlister, cause if it tries too quick and comms error, it shuts the power supply off.

Not sure what I can do for that on my end. I use PortLister w/ a 5 sec autoconnect delay and don't recall ever had comm errors from it connecting. What value were you using before?

cosmicc, 2021-04-27:
Nothing to be done.  I think my portlister was set to 0 sec.  Once I changed it to 3 or more seconds it worked fine.  Maybe just a note for people that enable the "connection error power off" option to make sure portlister has enough time to not fail the first connect and shut off the PSU.

kantlivelong, 2021-04-28:
Will add it to the troubleshooting list should it become a problem.

1.0.5 released

closed 2021-04-28 by kantlivelong