kantlivelong/OctoPrint-PSUControl #147: WIP - PowerOff on Printer Communication Error

LongLiveCHIEF, 2020-05-23:
I'm working on a PR that will trigger psu poweroff whenever printer communication fails.  This is a bit tricky depending on the users setup, so I'm opening this issue to help me keep track of the conditions and scenarios this logic needs to meet in order to be effective:

Conditionals to account for:

- The primary means of triggering will be by detecting a `Events.ERROR`. However, this event may not always be triggered if it is disabled by config.yaml, or [handled by another plugin][printer errors]
- (manual user input required) the psu off might be handled by the printer mainboard, and in such case, a serial communication error would mean that you wouldn't be able to send the gcode required to shut down psu. However, there is a `SUICIDE_PIN` feature in Marlin that the user could enable, and this plugin could connect to that for emergency events instead
- The logs will also contain a `SerialException` string, and thus could be polled or checked if a triggering condition is met
- on initialization check if ERROR event is enabled -> use log parsing only if not
-  enable/merge parsing of log for SerialException errors to trigger `turn_psu_off()`
- need to test if `Events.DISCONNECTED` and `PrinterStateChanged` can be used in combination as a fallback/assurance to ensure reliability of catching errors
- need to consider if it would be better to have an `Emergency Shutdown` section with a few configurable options based on all of the above

This is a WIP, and work can be tracked here: https://github.com/kantlivelong/OctoPrint-PSUControl/compare/master...LongLiveCHIEF:auto-turnoff-on-serialexception

*Note: I'm not sure why it shows that I've changed the entire `psucontrol_settings.jinja2` file`

[printer errors]: https://docs.octoprint.org/en/master/events/index.html#printer-communication

LongLiveCHIEF, 2020-05-23:
Just to make sure the implications of the issue are _completely obvious_, Due to the nature of the event handling, the nature/reliability of log parsing as a backup, and the nature of communication with the motherboard and the motherboard possibly handling PSU contro", there are a lot of "no guarantees" in this setup.

It's totally possible (and honestly very likely) that I will find that the results would be irresponsible to implement, because they could give a false sense of safety to users without don't understand the ways in which our monitoring could fail to catch a condition they believe would be caught.

@kantlivelong it might be worth considering that angle before we go too far down this road.

Thoughts?

kantlivelong, 2020-05-25:
Seems like a good idea but as you said it will really depend on how reliable it is.

BroncoNS, 2021-03-05:
This is a needed (and still currently missing) feature within octo plugins library. Hoping it sees maturity.

cosmicc, 2021-04-17:
I think this totally necessary.  In an error situation the user should already notified and it still should be regarded as an emergency to check immediately.   But, this would at least buy some time by trying to shut off all power until you can get there.  If I ran to the store during a print, and an error occurred, either way im running back quick to make sure everything is okay.  But it sure would be nice to at least have the plugin make an attempt for me to kill all power if it can, quicker then I could, even if I was sitting next to it.

kantlivelong, 2021-04-26:
Added via e88ee188d94bc17f46a61f5b1a09360236d84a73

Will be in 1.0.5

closed 2021-04-26 by kantlivelong