kantlivelong/OctoPrint-PSUControl #149: [Request] M27 should be added to default list of ignored commands

tophattwaffle, 2020-07-09:
When an SD card file is selected in Octoprint, it will constantly send M27 asking the printer the current SD status. This means that after a print is completed from SD, the idle timeout is never met as M27 is sent every X seconds, depending on the user's Octoprint config.

Adding M27 to the ignored list resolves this issue and lets the printer idle out so it can auto shut off. Since many people use Octoprint and still print from SD every now and then I believe this should be added to the default ignored commands.

kantlivelong, 2020-07-17:
Just curious, how is your environment set up with PSUControl? I only ask because idle power off doesn't really work if printing from SD and may actually shut off mid print. 

I honestly make no effort to test w/ SD as many hooks simply won't work.

tophattwaffle, 2020-07-17:
I'm using a TPLink outlet using a python script (switching method: System
Command) that PSU control uses to turn off and on the target.

I've been running multiple prints both through SD and streaming over serial
after adding the M27 ignore. I have not had the printer shut off mid print.

When I start a print over SD octoprint sees an SD print is running - even
when starting the print from the printer display. I am assuming PSUcontrol
knows about the printing state and does not idle power off.

kantlivelong, 2020-07-17:
Ah right it is checking if printing (DOH!).

Makes sense to add it then. Surprised you're the first to request it.