kantlivelong/OctoPrint-PSUControl #159: Switching method "GPIO" not available

zud222, 2020-12-27:
#### What were you doing?
For some time I'm only able to chose "System Command" as switching method because there is no option "GPIO" anymore. It's like the plugin/octoprint does not see my GPIO pins anymore. However, they work as expected when directly addressed via my workaround bash script.

#### What did you expect to happen?
Being able to set up GPIO pins for switching and sensing.

#### What happened instead?
The option "GPIO" is not available.

#### Version of OctoPrint-PSUControl
0.1.10

#### Operating System running OctoPrint
Raspbian Buster
OctoPrint 1.5.2

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG
<script src="https://gist.github.com/zud222/55b376bdd9311389bb9f63b9279c9c72.js"></script>

kantlivelong, 2020-12-28:
Looks like RPi.GPIO isn't installed.

SSH into the pi and as the pi user run:
```
source ~/oprint/bin/activate
~/oprint/bin/python -m pip install RPi.GPIO
```

then restart OctoPrint.

zud222, 2020-12-28:
This doesn't solve the problem:

`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in ./OctoPrint/lib/python2.7/site-packages (0.7.0)`

RPi.GPIO seems to be already installed. Restarting OctoPrint didn't change anything.

kantlivelong, 2020-12-28:
Interesting. 

- When did this start happening?
- Have you tried a different browser or private view?

zud222, 2020-12-28:
It started a while ago after I hadn't used the printer/OctoPrint for a couple of months (and therefore most likely after an update). It doesn't make a difference if I use Firefox or Chromium

spawnreaper, 2021-01-02:
Same problem here. But I just installed psu control. Ive been trying for days try to get it to work. 

GPIO python test script shows a strange response from the gpio ports. Setting the port status to true or false just turns the relay on. The relay can only be turned off with input command.  (test script used https://github.com/kgbplus/gpiotest/blob/master/gpiotest.py) maybe helpful in trying to resolve the issue.

kantlivelong, 2021-04-05:
This should no longer be an issue as of 27e66a21e91cfb855bf1b78006c16cbd0cedd632 in devel

kantlivelong, 2021-04-16:
v1.0.0 released.

closed 2021-04-16 by kantlivelong