kantlivelong/OctoPrint-PSUControl #162: [Bug] Impossible to finish a setup wizard when PSUControl is installed

Renaud11232, 2021-01-20:
Hello,

I'm working on another plugin and have had mutliple issues opened regarding a problem that makes it impossible to finish the setup wizard of my plugin. (Renaud11232/OctoPrint-Marlin-Flasher#64, Renaud11232/OctoPrint-Marlin-Flasher#63 and Renaud11232/OctoPrint-Marlin-Flasher#32)

After a bit of digging around, I think that the issue might be linked to this plugin.

I tried to reproduce the issue and the same thing happens with any plugin which has a setup wizard.

#### What were you doing?

To reproduce the issue

1. Start up an OctoPrint instance with PSU Control installed
2. Install any plugin with a setup wizard (I tried with Marlin-Flasher, Kronos Data Collector and Octoprint-Anywhere)
3. Restart OctoPrint
4. When logging into OP again, there is a Wizard page for the newly installed plugin
5. Click "Finish" (With or without filling in the wizard fields)

#### What did you expect to happen?

The settings to be saved and the wizard to close.

#### What happened instead?

The wizard is still open and the request to `/api/settings` failed with HTTP 500

This shows up in the logs
```
2021-01-20 16:59:19,840 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1539, in decorated_view
    return func(*args, **kwargs)
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/octoprint/server/api/settings.py", line 388, in setSettings
    response = _saveSettings(data)
  File "/home/renaud/OctoPrint/venv/lib/python3.8/site-packages/octoprint/server/api/settings.py", line 936, in _saveSettings
    "gcode", name, script.replace("\r\n", "\n").replace("\r", "\n")
AttributeError: 'NoneType' object has no attribute 'replace'
2021-01-20 16:59:19,844 - tornado.access - ERROR - 500 POST /api/settings (192.168.1.150) 7.56ms
```

This is the body of the `POST` request made when trying to finish the wizard.

[image]

#### Version of OctoPrint-PSUControl

0.1.10 (Installed using the Plugin Manager)

#### Operating System running OctoPrint

I tested on Ubuntu Server 20.04, Windows 10 20H2.
People who reported the issue used OctoPi (Raspbian)

#### Printer model & used firmware incl. version

No printer connected

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/5844095/octoprint.log)

#### Wiring diagram

N/A

Thanks

kantlivelong, 2021-01-20:
Interesting. Could you also provide js console logs? The gcode scripts are pre-processed on the js side and should always be submitted whether empty or not. I suspect something is breaking there.

Renaud11232, 2021-01-21:
Here are the logs from Chrome
[192.168.1.24-1611212966557.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/5847461/192.168.1.24-1611212966557.log)

Bostjan35, 2021-01-25:
Hello. I hawe one issue. 
I whant to create octoprint 0.17.0 for secound printer on <ip:5001>. I make all settings like is in one wideo crisBasement for multiple printers.
But when I have all sett on Wizard and click on Finish button the windows dont whant to close - nothing hapend. What is solution for this?
Please for help

kantlivelong, 2021-01-27:
@Renaud11232 can you try with this? https://github.com/kantlivelong/OctoPrint-PSUControl/archive/fix_162.zip

Renaud11232, 2021-01-27:
Hello,

I just tried, the same thing happens with the exact same error in the logs.

kantlivelong, 2021-01-27:
Hmm.. Will have to look further.

cp2004, 2021-02-01:
This issue was PSU Control all along? It only happened with 1.5.0+, and mitigation was added for 1.6.0 to make sure it is not none, since I wasn't able to find the issue. We had multiple reports on the forums, other plugins etc. and I never spotted if this was the common factor...

QuinnDamerell, 2021-02-01:
This is a great find! I'm the developer of OctoEverywhere and I also had users having trouble with this. I would be happy to help out however I can, just let me know. :)

kantlivelong, 2021-02-01:
Fixed in 0.1.11.

Thanks for the help.

PS: This also affects [OctoPrint-SmartPreheat](https://github.com/kantlivelong/OctoPrint-SmartPreheat) which will be getting a fix shortly.

closed 2021-02-01 by kantlivelong

mikekscholz, 2021-02-02:
Omg, I've been working on making RGB-Status plugin Python 3.7 compatible and getting the wizard to finish was the last part of the puzzle that I couldn't get, I never would have suspected it was another plugin.

kantlivelong, 2021-02-02:
> Omg, I've been working on making RGB-Status plugin Python 3.7 compatible and getting the wizard to finish was the last part of the puzzle that I couldn't get, I never would have suspected it was another plugin.

Welcome! :smile:

commit de4fd3e4 (2021-02-01): Don't update gcode scripts if gcode script obvservables are still undefined. This occurs when settings hasn't been in opened such as the wizard. (Fixes #162)