kantlivelong/OctoPrint-PSUControl #164: Fresh install fails on Python 3.8

Raptoaaah, 2021-01-24:
#### What were you doing?

1. Fresh install on a 3.8 venv
2. Install the plugin

#### What did you expect to happen?

Clean install.

#### What happened instead?

Failed.

#### Version of OctoPrint-PSUControl

0.1.10

#### Operating System running OctoPrint

Ubuntu 20.10

#### Printer model & used firmware incl. version
Irrelevant

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://gist.github.com/Raptoaaah/af791aef0c606dd80c7fcd476eaac73d

#### Wiring diagram
Irrelevant

ping-localhost, 2021-02-23:
I'm having the same issue using OctoPi for Raspberry Pi 4 8GB, 64bits. 

```
2021-02-23 19:49:04,503 - octoprint.plugins.pluginmanager - ERROR - Could not install plugin from https://github.com/kantlivelong/OctoPrint-PSUControl/archive/master.zip
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 738, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1414, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 209, in execute
    return self.call(command, **kwargs)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 511, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 530, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 31: ordinal not in range(256)
2021-02-23 19:49:04,504 - octoprint.plugins.pluginmanager - ERROR - Reason: UnicodeEncodeError('latin-1', '  source/c_gpio.c: In function ‘setup’:\n', 31, 32, 'ordinal not in range(256)')
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 738, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1414, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 209, in execute
    return self.call(command, **kwargs)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 511, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/pip.py", line 530, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/oprint/lib/python3.8/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 31: ordinal not in range(256)
```

kantlivelong, 2021-02-23:
Which version of RPi.GPIO is installed in the venv?

ping-localhost, 2021-02-23:
@kantlivelong Sorry for the late reply, was waiting for my print to finish. Not sure what you are looking for. If I'm reading these logs correctly, installing RPi.GPIO itself failed? Here is the full gist if that helps: https://gist.github.com/ping-localhost/aad12f9619b13deaf5e2d6810407a0f5

cp2004, 2021-02-23:
> I'm having the same issue using OctoPi for Raspberry Pi 4 8GB, 64bits.

Is this the ubuntu server based version? Some GPIO issues are being looked at at the moment, since it didn't include some of the right info.

ping-localhost, 2021-02-23:
> Is this the ubuntu server based version? Some GPIO issues are being looked at at the moment, since it didn't include some of the right stuff.

Yes, it's based on Ubuntu according to [the issue in OctoPi](https://github.com/guysoft/OctoPi/issues/711). Maybe I should report it there as well? 🤔 
 
> _Edit_: Also note that the 64 bit builds are not known for their stability...

Of course, but that shouldn't stop me from reporting that the issue exists, right? 😜

Raptoaaah, 2021-02-23:
Would also be nice to know if this is even related to the plugin or if the underlying parser is messed up - UnicodeEncodeError could also be a problem of the pluginmanager failing at handling \u2018 and \u2019 characters while using unicode. Maybe even changing the encoding may be necessary at some point if this is escalating.

kantlivelong, 2021-02-23:
Do other plugins which depend on `RPI.GPIO` work? Try https://github.com/catgiggle/OctoPrint-GpioControl

Raptoaaah, 2021-02-23:
Fails as well.

kantlivelong, 2021-02-23:
For now going to mark this as invalid since it appears to be an upstream issue. Are you able to test on 32bit?

ping-localhost, 2021-02-24:
@kantlivelong PSU control worked fine on the 32bit nightly build I used before.

kantlivelong, 2021-02-24:
> @kantlivelong PSU control worked fine on the 32bit nightly build I used before.

Yeah so it's def a 64bit issue. Also just took a look at the issue tracker for RPi.GPIO and these came up:
- https://sourceforge.net/p/raspberry-gpio-python/tickets/194/
- https://sourceforge.net/p/raspberry-gpio-python/tickets/191/

The in dev https://github.com/kantlivelong/OctoPrint-PSUControl/tree/periphery_gpio branch adds GPIO support through Periphery but that won't remove the dependency for RPi.GPIO for a while. I've been using it for a short while without issue. If you want to use it you would need to strip out rpi.gpio from the requirements.txt. Do be aware that the Pi pull up/down resistor support does not work yet as kernel support is still being worked on.

kantlivelong, 2021-04-05:
27e66a21e91cfb855bf1b78006c16cbd0cedd632 in devel brings in Periphery support and yanks out RPi.GPIO. Not released yet but expect it to come.

kantlivelong, 2021-04-16:
v1.0.0 released.

closed 2021-04-16 by kantlivelong