kantlivelong/OctoPrint-PSUControl #168: Octopi should be connected if sending detect psu is On

DisasteR, 2021-02-05:
#### What were you doing?

I am using psu-control with system scripts to sense and manage m'y Xiaomi Smart plug.
If i power on m'y plug manually from outside octopi sensing append correctly and psu-control detect printer is on.
But octopi did mot connect automatically.

#### What did you expect to happen ?

Psu control should reconnect when sensing Psu ON.

#### What happened instead?

Octopi stay disconnected.

#### Version of OctoPrint-PSUControl

0.1.11

#### Operating System running OctoPrint

Octopi 0.17.0

#### Printer model & used firmware incl. version

Ender 3, skr mini e3, marlin 2.0.7.2

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

Markeno76, 2021-02-06:
I also expected that behavior, and have found it does not automatically connect to the printer if it Senses power has been enabled by reading GPIO set in the Sense setting.  It does detect power is to be present though.

kantlivelong, 2021-03-06:
Logs?

kantlivelong, 2021-04-17:
Closing due to inactivity.

closed 2021-04-17 by kantlivelong

bcat, 2021-09-24:
For what it's worth, I just set up PSU Control with a smart plug (Tasmota) and also noticed that although the "connect automatically" option in PSU Control causes OctoPrint to connect after turning on the plug through OctoPrint UI, no automatic connection occurs if I manually turn on the plug from its physical button. Here's what I see in the logs:

```
2021-09-23 23:21:14,978 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.psucontrol level to 10
2021-09-23 23:21:14,978 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.psucontrol_tasmota level to 10
2021-09-23 23:21:16,656 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-09-23 23:21:16,756 - octoprint.plugins.psucontrol_tasmota - DEBUG - cmd=Status 0, status_code=200, text={"Status":{"Module":41,"DeviceName":"Rufus-Plug","FriendlyName":["Rufus-Plug"],"Topic":"tasmota_D9B4A6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T05:23:24","StartupUTC":"2021-09-23T22:57:52","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-09-23T22:03:53","SaveCount":51,"SaveAddress":"F9000"},"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"375/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CatsOfDarkness-Legacy",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100068000005A0A000000000000","00000080","00006000","00000000"]},"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"plug-rufus","IPAddress":"192.168.1.56","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:D9:B4:A6","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusTIM":{"UTC":"2021-09-24T04:21:16","Local":"2021-09-24T05:21:16","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:40","Sunset":"18:43"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2021-09-24T05:21:16","ENERGY":{"TotalStartTime":"2021-09-23T22:03:53","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}},"StatusSTS":{"Time":"2021-09-24T05:21:16","Uptime":"0T05:23:24","UptimeSec":19404,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"CatsOfDarkness-Legacy","BSSId":"D0:17:C2:59:9B:68","Channel":1,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}
2021-09-23 23:21:16,756 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2021-09-23 23:21:21,763 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-09-23 23:21:21,893 - octoprint.plugins.psucontrol_tasmota - DEBUG - cmd=Status 0, status_code=200, text={"Status":{"Module":41,"DeviceName":"Rufus-Plug","FriendlyName":["Rufus-Plug"],"Topic":"tasmota_D9B4A6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T05:23:29","StartupUTC":"2021-09-23T22:57:52","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-09-23T22:03:53","SaveCount":51,"SaveAddress":"F9000"},"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"375/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CatsOfDarkness-Legacy",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100068000005A0A000000000000","00000080","00006000","00000000"]},"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"plug-rufus","IPAddress":"192.168.1.56","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:D9:B4:A6","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusTIM":{"UTC":"2021-09-24T04:21:21","Local":"2021-09-24T05:21:21","Start [... cut ...]
2021-09-23 23:21:47,768 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
```

So it does appear that PSU Control is detecting that the plug is turned on (and I see the icon turn green in the UI), and it's firing a `psu_state_changed` event, but not attempting to start a serial connection.

If this just isn't implemented, I can file a feature request, but I wasn't sure if this was expected to work and I'd just come across a bug.

Either way, let me know if there's any more info I can provide. Thanks!

bcat, 2021-09-24:
(Or maybe this is working as intended and the right way would be to write a simple subplugin to listen to `psu_state_changed`? That's fine too of course, I just wasn't sure what was the intended behavior.