kantlivelong/OctoPrint-PSUControl #172: PSU idle timer ignores non-Marlin (Klipper) g-codes and shuts down when not idle!!

CharlieTemplar, 2021-03-06:
#### What were you doing?

DELTA_CALIBRATE METHOD=manual
after going thru 6 of the 7 points, manually adjusting the z-height with TESTZ command and then moving on to next point with ACCEPT command

#### What did you expect to happen?
complete the calibration steps and SAVE_CONFIG

#### What happened instead?
the PSU Plugin decided what I was doing was not worthy of staying powered on for and abruptly shut-off, losing all of the operations and data captured up to that point.

#### Version of OctoPrint-PSUControl
PSU Control (0.1.11)

#### Operating System running OctoPrint

 Raspbian GNU/Linux 8.0 (jessie)

#### Printer model & used firmware incl. version

Delta / SKR1.3 / raspi2b / ATX PSU 

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG
NA

#### Wiring diagram

NA

kantlivelong, 2021-03-06:
logs?

CharlieTemplar, 2021-03-06:
haven't enabled DEBUG for this as wasn't expecting these commands to be ignored.  
I have since turned on DEBUG on the plugin and it doesn't say anything... except isPSUOn : true
```
2021-03-06 13:50:36,021 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-03-06 14:00:08,135 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.psucontrol level to 10
2021-03-06 14:00:11,600 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:16,604 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:21,612 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:26,619 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:31,626 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:36,632 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:41,639 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:46,646 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:51,654 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:00:56,663 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:01,672 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:06,678 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:11,694 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:16,705 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:21,713 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:26,718 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:31,726 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:36,734 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:41,741 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:46,747 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:51,756 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:01:56,767 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:01,779 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:06,786 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:11,793 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:16,799 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:21,620 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-03-06 14:02:21,806 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:26,814 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:31,820 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:36,827 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:41,835 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:46,846 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:51,852 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:02:56,858 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:01,866 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:06,872 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:11,882 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:16,889 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:21,899 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:26,906 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:31,914 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:36,920 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:41,926 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:46,932 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:51,941 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:03:56,947 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:01,955 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:06,964 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:11,972 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:16,979 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:21,986 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:26,994 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:32,001 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:37,008 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:42,014 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:47,021 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:52,028 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:04:57,035 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:05:02,043 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:05:07,048 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:05:12,057 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:05:17,066 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:05:22,076 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
[... cut ...]
2021-03-06 14:07:02,230 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:07:07,237 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 14:07:12,244 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True

...

2021-03-06 14:17:21,622 - octoprint.server.heartbeat - INFO - Server heartbeat <3

...

```
I was running though the klipper style commands during this.  The idle timer is set for 25 mins and probably G28 will not be classed as idle so not sure how long is left before the idle time will kick in

kantlivelong, 2021-03-06:
Going to need full logs. May as well include serial logs.

CharlieTemplar, 2021-03-06:
Serial logs not enabled nor is general DEBUG, just DEBUG on plugin and the full log is the same as above except I chopped of the repeated isPSUon and some of the heartbeats every 15 mins, otherwise, that was all she wrote...
I might turn the idle time down to a couple of minutes and redo a partial delta_calibrate with all logging set to DEBUG

CharlieTemplar, 2021-03-06:
octoprint.log with octoprint DEBUG and plugin DEBUG and idle timer set to 2 mins
```
2021-03-06 15:08:40,587 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2021-03-06 15:08:45,071 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-03-06 15:08:45,074 - octoprint.plugins.psucontrol - DEBUG - Switching PSU On Using GPIO: 4
2021-03-06 15:08:45,223 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 15:08:50,233 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-03-06 15:08:52,308 - octoprint.events.fire - DEBUG - Firing event: Connecting (Payload: None)
2021-03-06 15:08:52,310 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69a4b030>>
2021-03-06 15:08:52,315 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-03-06 15:08:52,317 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-03-06 15:08:52,322 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-03-06 15:08:52,327 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-03-06 15:08:52,331 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-03-06 15:08:52,334 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-03-06 15:08:52,338 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-03-06 15:08:52,411 - octoprint.plugin - DEBUG - Calling on_event on tracking
2021-03-06 15:08:52,415 - octoprint.plugin - DEBUG - Calling on_event on cancelobject
2021-03-06 15:08:52,417 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2021-03-06 15:08:52,418 - octoprint.plugin - DEBUG - Calling on_event on klipper
2021-03-06 15:08:52,422 - octoprint.plugin - DEBUG - Calling on_event on psucontrol
2021-03-06 15:08:52,511 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2021-03-06 15:08:52,519 - octoprint.util.comm - INFO - Connecting to port /tmp/printer, baudrate 250000
2021-03-06 15:08:52,528 - octoprint.events.fire - DEBUG - Firing event: PrinterStateChanged (Payload: {u'state_id': 'OPEN_SERIAL', u'state_string': u'Opening serial connection'})
2021-03-06 15:08:52,554 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69a4b030>>
2021-03-06 15:08:52,559 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-03-06 15:08:52,564 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-03-06 15:08:52,565 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-03-06 15:08:52,568 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-03-06 15:08:52,573 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-03-06 15:08:52,574 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-03-06 15:08:52,576 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2021-03-06 15:08:52,578 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-03-06 15:08:52,587 - octoprint.plugin - DEBUG - Calling on_event on tracking
2021-03-06 15:08:52,590 - octoprint.plugin - DEBUG - Calling on_event on cancelobject
2021-03-06 15:08:52,591 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2021-03-06 15:08:52,592 - octoprint.plugin - DEBUG - Calling on_event on klipper
2021-03-06 15:08:52,593 - octoprint.plugin - DEBUG - Calling on_event on psucontrol
2021-03-06 15:08:52,595 - octoprint.events.fire - DEBUG - Firing event: PrinterStateChanged (Payload: {u'state_id': 'CONNECTING', u'state_string': u'Connecting'})
2021-03-06 15:08:52,596 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69a4b030>>
2021-03-06 15:08:52,603 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-03-06 15:08:52,610 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-03-06 15:08:52,611 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-03-06 15:08:52,616 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 15:08:52,621 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-03-06 15:08:52,624 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-03-06 15:08:52,629 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-03-06 15:08:52,634 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 15:08:52,642 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-03-06 15:08:52,651 - octoprint.plugin - DEBUG - Calling on_event on tracking
2021-03-06 15:08:52,658 - octoprint.plugin - DEBUG - Calling on_event on cancelobject
2021-03-06 15:08:52,655 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2021-03-06 15:08:52,662 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2021-03-06 15:08:52,671 - octoprint.plugin - DEBUG - Calling on_event on klipper
2021-03-06 15:08:52,684 - octoprint.plugin - DEBUG - Calling on_event on psucontrol
[... cut ...]
2021-03-06 15:12:22,832 - Send: M105
2021-03-06 15:12:22,850 - Recv: ok B:20.6 /0.0 T0:19.7 /0.0
2021-03-06 15:12:25,258 - Send: TESTZ z=-.1
2021-03-06 15:12:25,306 - Recv: // Z position: ?????? --> 0.698 <-- 0.798
2021-03-06 15:12:25,315 - Recv: ok
2021-03-06 15:12:27,835 - Send: M105
2021-03-06 15:12:27,855 - Recv: ok B:20.6 /0.0 T0:19.7 /0.0
2021-03-06 15:12:28,244 - Connection closed, closing down monitor
2021-03-06 15:12:28,254 - Changing monitoring state from "Operational" to "Offline"
```

CharlieTemplar, 2021-03-06:
@kantlivelong What **Information** are you still **Awaiting**?
these are the FULL logs.

kantlivelong, 2021-03-06:
The octoprint.log is not complete. It lacks startup info or even a rollover notice thus it's a snippet This ticket so far consisted of a painful attempt at getting logs when none were provided to begin with. Enable serial logs. Wipe the logs. Restart OctoPrint. Re-create issue. UPLOAD logs.

CharlieTemplar, 2021-03-06:
sorry that only including relevant logs caused you such pain :D 
[serial.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6095776/serial.log)
[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6095777/octoprint.log)

kantlivelong, 2021-03-06:
Not all that familiar with Klipper but I'd imagine the macros are mostly used outside of normal printing (with exception of tool changes maybe?). #166 once merged/released should fix that. For now you should be able to workaround by using a longer idle timeout as 2 minutes is pretty short (For instance Z probing can take 2+ minutes)

CharlieTemplar, 2021-03-06:
Lol, I set the timeout to 2 mins just to reproduce the issue quicker.  Was 25 mins when I first encountered this! 
I'll give it a try...  is it just those 2 lines.

CharlieTemplar, 2021-03-09:
What PSU control option causes the plugin to send the postPowerOn command(s) when printing?
When I press print, even though the PSU is on, PSU control still sends the postOnGcodeScript command: 
2021-03-09 21:15:05,669 - Changing monitoring state from "Operational" to "Starting"
2021-03-09 21:15:05,757 - Send: N0 M110 N0*125
2021-03-09 21:15:05,761 - Recv: ok
2021-03-09 21:15:05,928 - PSUControl: ok
2021-03-09 21:15:05,969 - Send: N1 M140 S60*82
2021-03-09 21:15:05,996 - Recv: ok
2021-03-09 21:15:06,037 - Send: FIRMWARE_RESTART
2021-03-09 21:15:07,090 - Recv: ok
2021-03-09 21:15:07,098 - Recv: // Klipper state: Disconnect
2021-03-09 21:15:07,104 - Changing monitoring state from "Starting" to "Printing"
2021-03-09 21:15:07,147 - Send: N2 M105*37
2021-03-09 21:15:10,690 - Recv: ok T:0
2021-03-09 21:15:10,723 - Send: N3 M105*36
2021-03-09 21:15:10,730 - Recv: ok T:0
2021-03-09 21:15:10,739 - Send: N4 M190 S60*90
2021-03-09 21:15:10,746 - Recv: // Printer is not ready
2021-03-09 21:15:10,758 - Recv: // The klippy host software is attempting to connect.  Please
2021-03-09 21:15:10,760 - Recv: // retry in a few moments.
2021-03-09 21:15:10,764 - Recv: !! Printer is not ready
2021-03-09 21:15:10,808 - Changing monitoring state from "Printing" to "Cancelling"
2021-03-09 21:15:11,399 - Recv: ok
2021-03-09 21:15:11,416 - Send: N5 G28*22
2021-03-09 21:15:11,426 - Send: N6 M84*25
2021-03-09 21:15:11,434 - Recv: // Printer is not ready
2021-03-09 21:15:11,519 - Recv: // The klippy host software is attempting to connect.  Please
2021-03-09 21:15:11,534 - Recv: // retry in a few moments.
2021-03-09 21:15:11,545 - Recv: !! Printer is not ready

I implemented the above mentioned fix to recognise klipper commands, don't know if this is related

kantlivelong, 2021-03-09:
Is this a separate issue? If so, open a new ticket with logs.

kantlivelong, 2021-04-05:
This should be fixed with 23a88bba1c9cd16c8cdd50ec49acded852a5ccb5 in devel.

kantlivelong, 2021-04-16:
v1.0.0 released.

closed 2021-04-16 by kantlivelong