kantlivelong/OctoPrint-PSUControl #173: PSU Control with per-group ACL permissions

netfriend85, 2021-03-07:
Hi, 
Big Thanks for creating this extremely useful plugin. It is a great addition that I use it constantly. 

I am seeing that 'PSE control' can be switched on and off by all authenticated users in the Admins, Operators, Read-Only Access groups.

- Can the PSU controls be restricted (configurable?) based on the user's ACL group membership?
- Can the PSU control permissions be added to the list of permissions listed under the groups permissions tab?

For example, by default, only users in the Admin, and Operators groups should be able to switch the 'PSU Control'. 
Users in the Read-Only group should only be able to see the PSU status (on or off) without ability to switch it. 
If non-default config is needed, the 'PSU Control' switching permissions should be configurable in the Groups permission tab.

Cheers,

kantlivelong, 2021-03-09:
Never noticed that read-only users had that kind of access with legacy access control. Makes sense to implement then.

netfriend85, 2021-03-11:
Thank you for looking into and considering it.

The credit for detecting this goes to my 8-year old son as he was the one pointing it to me. 
In OctoPrint, I have read-only account for him where he loves to watch live the magic of 3D-printing of his next toy character. :-)

Cheers,

kantlivelong, 2021-04-05:
Added to devel via f820e0ab9014b060890fcc6f27393859dc59355c

kantlivelong, 2021-04-16:
v1.0.0 released.

closed 2021-04-16 by kantlivelong