kantlivelong/OctoPrint-PSUControl #175: PSU Control sends postPowerOn commands after print starts.

CharlieTemplar, 2021-03-10:
#### What were you doing?

Printing via Octoprint/Klipper

#### What did you expect to happen?

Pre-Heat and Print

#### What happened instead?

PSU control had already switched on, was reporting isPSUOn:True and everything else was good and reporting ready.  PressedPrint in Octoprint,  Heater ON gcode sent and PSU Control sent the postPowerOn code (FIRMWARE_RESTART) which caused the printer board to report Not Ready and octoprint to cancel the print. heat-on, Firmware-restart, heat-off then home.  log contains 2 instances of this and tried many other times with various options.

#### Version of OctoPrint-PSUControl

OctoPrint 1.5.3
PSU Control (0.1.11) (+ #166)

#### Operating System running OctoPrint

Raspbian Jessie

#### Printer model & used firmware incl. version

SKR1.3, Pi2b, Klipper (v0.9.1-278-g1c1c517), Delta

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

[serial.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6112171/serial.log)
[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6112174/octoprint.log)

#### Wiring diagram

NA

kantlivelong, 2021-03-10:
Are you sending M80 anywhere? (gco, octoprint scripts, etc...?)

CharlieTemplar, 2021-03-10:
No, only in the PSU plugin afaik...
[image]

[D_calibrate_size.gcode](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6112396/D_calibrate_size.gcode.txt)

kantlivelong, 2021-03-10:
If you could. Since you know how to patch. Can you modify and recapture logs:

```
    def hook_gcode_queuing(self, comm_instance, phase, cmd, cmd_type, gcode, *args, **kwargs):
        skipQueuing = False
```

to

```
    def hook_gcode_queuing(self, comm_instance, phase, cmd, cmd_type, gcode, *args, **kwargs):
        self._logger.debug(cmd)
        skipQueuing = False
```

kantlivelong, 2021-04-17:
Closed due to inactivity.

closed 2021-04-17 by kantlivelong