kantlivelong/OctoPrint-PSUControl #176: Gpio Pin saving problem

Ruissilva, 2021-03-21:
Hello Kantlivelong,

I have to change the gpio number to a wrong number, then save. Then go back no psu menu, chose the correct gpio number and only then the toggle button works. 

I unistalled and cleaned data and then installed the "https://github.com/kantlivelong/OctoPrint-PSUControl/archive/fix_settings.zip" with the same results. 

I also tried a different browser also with the same results. 

What can i try next?

Thanks in advance,
Rui Silva

Drzet, 2021-03-27:
Seems like this issue has been reported countless times and there is no fix. Waiting

Ruissilva, 2021-03-27:
I found that the problem was conflict between plugins. The problem is between the smart filament sensor plugin and gpio affecting plungins, like enclosure, psu control and octorelay. I  already reported in smart filament sensor github but had no response, if you guys could contact them about their problem that whould be great. The only thing I find buggy in psu contol is that there are options that don't work like automatic shutdown. But right now I have smart filament sensor installed so I cannot have any gpio affecting plugin, so makes no difference.

Sorry I didn't right in here earlier, but you can close this issue.

kantlivelong, 2021-03-27:
@Ruissilva makes sense.

PSUControl actually tries to be flexible with other plugins by converting the pin config to whatever GPIO mode is currently in use. So if you specify BCM mode but another plugin has already setup in BOARD then PSUControl will convert the pin to BOARD. Most other plugins do not do this which causes the wrong pin to be configged. Not much I can do here other than recommend requestinf other GPIO plugins implement similar functionality. However that is only a short term solution anyway as RPi.GPIO has been deprecated and will be replaced soon anyway.

Ruissilva, 2021-03-27:
Ok, nice tip, mantain all plugins with the same config, thanks :)

Drzet, 2021-03-28:
> I found that the problem was conflict between plugins. The problem is between the smart filament sensor plugin and gpio affecting plungins, ... The only thing I find buggy in psu contol is that there are options that don't work like automatic shutdown. 
Possible, I also have SmartSensor plugin. Strangely, automatic shutdown works jut fine for me.

kantlivelong, 2021-03-28:
There is no automatic shutdown option. If you're referring to power off fhen make sure manually shutting off works in the UI. If that doesn't work you likely have a wiring issue.

Drzet, 2021-03-28:
> There is no automatic shutdown option. If you're referring to power off fhen make sure manually shutting off works in the UI. If that doesn't work you likely have a wiring issue.

I am not sure we are talking about the same thing. **There is** automatic shutdown option in PSU control under " Power Off options"

kantlivelong, 2021-03-28:
Right, that will switch otf the GPIO pin. Not to be confused with a system shutdown. If it's not working you likely have a config or wiring problem.

griehsler, 2021-04-01:
I also got this issue ever since installing Smart Filament Sensor plugin next to PSU Control. Turns out that in a [related bug on their github](https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/issues/31#issuecomment-761606310) a solution is discussed and that one fixed it for me, too.

The issue is that Smart Filament Sensor runs `GPIO.cleanup()` in its initialization, thus reverts the GPIO setup done by all other plugins that happened to setup before (facepalm). Commenting that line out fixes PSU Control.

kantlivelong, 2021-04-05:
Closing since it's not due to this plugin.

closed 2021-04-05 by kantlivelong