kantlivelong/OctoPrint-PSUControl #180: Button at the top of OctoPi interface stopped working

Arakon, 2021-04-16:
I'm using a Mendel 90 with this wiring setup (minus the lights): https://hydraraptor.blogspot.com/2014/06/lights-camera-action.html
I'm using a Raspberry 3b, with the power on wire connected to physical pin 22/GPIO25.

After updating to PSUControl 1.0.0, the plugin stopped working for me.
After experimenting a bit, I found that GPIO25 will switch the printer on once I set it to 25 and Save, and off when I select "Invert" and save.
However, the lightning bolt button at the top of the OctoPi interface has no function at all anymore for me, only way to turn the printer on or off is to go into settings and toggle the "Invert" setting on or off.

[image]

deadSoldier9, 2021-04-16:
Same for me, thank your for opening a ticket. was on bcm14 (TX UART) and thought this was the problem. Switched to BCM4(GPIO) and tried with many settings. Switching with invert button does work but not on the mentioned button on top of Octoprint GUI

krawcu183, 2021-04-16:
I have the same issue. Relay is connected to pin 16 on RPi3b. Before update in the settings I had selected GPIO Pin 16, after update I had to change it to 23.
Lake other mentioned, the only way to switch printer is to check/uncheck invert.
Button in upper part of the main screen as well as option in the app in the phone doesn't work.
Printer do not switch off with Idle Timeout.

oerkel47, 2021-04-16:
Got pretty much the same problem.

Log says this after clicking button:
`
octoprint        | 2021-04-16 11:10:16,915 - octoprint.plugins.psucontrol - INFO - Switching PSU On
octoprint        | 2021-04-16 11:10:16,915 - octoprint.plugins.psucontrol - ERROR - Invalid value type, should be bool.
`

DerDingsCS, 2021-04-16:
I think I managed to solve this issue. 
Setup PSU Control to use PSU Control RPi.GPIO
[image]
then setup PSU Control RPi.GPIO
[image]

I also switched from pin 12 to GPIO 18.

DerDingsCS, 2021-04-16:
Addition information to post 

> 
> 
> I think I managed to solve this issue.
> Setup PSU Control to use PSU Control RPi.GPIO (Switching Method)
> [image]
> then setup PSU Control RPi.GPIO
> [image]
> 
> I also switched from pin 12 to GPIO 18.

Too bad it is not working. New issue is
2021-04-16 13:40:08,090 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 13:40:08,093 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-16 13:40:10,438 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-04-16 13:40:10,440 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT

Arakon, 2021-04-16:
Sadly, this method just allows the button to turn green, but it doesn't actually affect my printer in any way.

kantlivelong, 2021-04-16:
I can't reproduce. Please provide the following info:

#### Version of OctoPrint-PSUControl

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

#### Browser console logs ( See https://www.wickedlysmart.com/hfjsconsole/ )

DerDingsCS, 2021-04-16:
I send all needed data as soon as possible, but it will take a while.
Do you run any other plugin using gpio.
I'm using Enclosure Plugin (4.13.1) and when I setup the same GPIO as output there, the toggle button works. But it seems that both plugins try to set the state. It goes on and off then.

kantlivelong, 2021-04-16:
I do use other plugins that use RPi.GPIO including my own. Plugin works with the newer builtin GPIO and a sub plugin that maintains RPi.GPIO.

Once I get some logs I should be able to see something or have something to work from.

franknepomuceno, 2021-04-16:
This latest version doesn't work for me either. The previous version worked like a charm.  I wished this was tested more thoroughly.

kantlivelong, 2021-04-16:
> This latest version doesn't work for me either. The previous version worked like a charm. I wished this was tested more thoroughly.

Got any logs to provide or would you rather it's never fixed?

z00mantwo, 2021-04-16:
It's happening to me too. When I press the button logs show:

2021-04-16 13:54:38,960 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 13:54:38,961 - octoprint.plugins.psucontrol - ERROR - Invalid value type, should be bool.

I also get the following when trying the plugin

2021-04-16 13:57:16,727 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-16 13:57:16,727 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT

My power is on pin 18 (GPIO24) with a pi 4b.

kantlivelong, 2021-04-16:
> It's happening to me too. When I press the button logs show:
>
> 2021-04-16 13:54:38,960 - octoprint.plugins.psucontrol - INFO - Switching PSU On
> 2021-04-16 13:54:38,961 - octoprint.plugins.psucontrol [...]

Please provide full logs.

z00mantwo, 2021-04-16:
If I "invert" the pin I get

2021-04-16 14:02:38,731 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-04-16 14:02:38,731 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT

z00mantwo, 2021-04-16:
by playing with it a lot and not using plug. The PSU is now on and button is green.

When I press the button to turn off I get:

2021-04-16 14:04:40,740 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-04-16 14:04:40,740 - octoprint.plugins.psucontrol - ERROR - Invalid value type, should be bool.

and the PSU stays on.

franknepomuceno, 2021-04-16:
> > This latest version doesn't work for me either. The previous version worked like a charm. I wished this was tested more thoroughly.
> 
> Got any logs to provide or would you rather it's never fixed?

Sorry @kantlivelong, my comment was uncalled for.  I'll post my logs soon.

Arakon, 2021-04-16:
> #### Version of OctoPrint-PSUControl

1.10 

> #### Operating System running OctoPrint

OctoPi on Raspberry Pi 3b
[image]

> #### Printer model & used firmware incl. version

Shouldn't matter as the printer is not even connected or powered at this point, but: Mendel 90 running Marlin 2.0.7

> #### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6325329/octoprint.log)

> #### Browser console logs ( See https://www.wickedlysmart.com/hfjsconsole/ )

[Browserlog.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6325331/Browserlog.txt)

z00mantwo, 2021-04-16:
I was able to fix things by editing /boot/config.txt and adding the following at the end:

gpio=24=op,dh

This forces the GPIO24 to be an output. PSU control button now works.

z00mantwo, 2021-04-16:
Note above fix was with using the plugin. Does not work without plugin and change to /boot/config.txt

z00mantwo, 2021-04-16:
Also a full pi reboot was required for /boot/config.txt change to take affect.

kantlivelong, 2021-04-16:
@Arakon THANK YOU :1st_place_medal: 

Give this build a try: https://github.com/kantlivelong/OctoPrint-PSUControl/archive/refs/heads/issue_180_fix.zip

kantlivelong, 2021-04-16:
> I was able to fix things by editing /boot/config.txt and adding the following at the end:
> 
> gpio=24=op,dh
> 
> This forces the GPIO24 to be an output. PSU control button now works.

You shouldn't need to do this. Revert the change and give the build linked above a try.

Arakon, 2021-04-16:
Seems to not have worked with the new version.
Updated log (also tried using the GPIO Plugin in this log):
[octoprint(1).log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6325494/octoprint.1.log)

Also, on a side note, when installing the GPIO Plugin, Octoprint wants to refresh, but for the plugin to show up properly and work, it actually needs to restart octoprint completely.

kantlivelong, 2021-04-16:
> Seems to not have worked with the new version.

My bad, goofed on the push. Retry if you can :)

Zw1d, 2021-04-16:
Stopped working after update. Changed to System Command instead of GPIO (pin#3 in my case):
gpio -g write 3 1
gpio -g write 3 0

Arakon, 2021-04-16:
> 
> 
> > Seems to not have worked with the new version.
> 
> My bad, goofed on the push. Retry if you can :)

That did the trick. However, the printer now turns on automatically as soon as Octoprint starts up, while the button assumes it is turned off. So you have to press the button once to make it believe it's on, and then again to turn it off.
After that, it seems to be working as it should. Attaching updated log.

[octoprint(2).log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6325602/octoprint.2.log)

kantlivelong, 2021-04-16:
>  However, the printer now turns on automatically as soon as Octoprint starts up, while the button assumes it is turned off.

This should fix that. Hopefully that's all but :shrug:

https://github.com/kantlivelong/OctoPrint-PSUControl/archive/refs/heads/issue_180_fix.zip

Arakon, 2021-04-16:
Yup, seems to work as intended now.

kantlivelong, 2021-04-16:
Once I get a few other confirmations I'll push this out as 1.0.1.

oerkel47, 2021-04-16:
Can confirm the button works now. "Invert sensing" option appears fixed too, since it doesn't change the output gpio state anymore.

Sensing on same Pin doesn't work for me, but I am sure thats not related to this issue here.

kantlivelong, 2021-04-16:
> Sensing on same Pin doesn't work for me, but I am sure thats not related to this issue here.

Kind of expected since the pin shouldn't be setup as both an output and an input. Really better off using Internal mode. If anything that can be enhanced.

oerkel47, 2021-04-16:
It worked with the old method though. If you read an output, it should return the last written value. Not sure how thats implemented in Python-Periphery. If thats a different command to the one used for sensing, could you implement it?

kantlivelong, 2021-04-16:
> It worked with the old method though. If you read an output, it should return the last written value. Not sure how thats implemented in Python-Periphery. If thats a different command to the one used for sensing, could you implement it?

Opened #182 . For now use Internal for sensing.

z00mantwo, 2021-04-16:
Okay. I took out my /boot/config.txt fix. Rebooted the pi then installed your 180 fix and everything is working now.

Thank you very much!

oerkel47, 2021-04-16:
> Opened #182 . For now use Internal for sensing.

Will do. Thanks for opening and thank you for the quick fix of this.

deadSoldier9, 2021-04-16:
The fix did the trick, switched to internal sensing and it works now.

Prizmagnetic, 2021-04-16:
I'm on PSU control 1.0.0
using pin BCM 23

I was able to get the pin to toggle when I disabled the Filament Sensors Revolutions plugin (only other thing using the GPIO)
PSU button still doesn't seem to work.

I guess I test the patch next...

bjhiltbrand, 2021-04-16:
I can confirm also that the fix in the zip file above works as expected.

kantlivelong, 2021-04-17:
Version [1.0.1](https://github.com/kantlivelong/OctoPrint-PSUControl/releases/tag/1.0.1) has been released which addresses this.

closed 2021-04-17 by kantlivelong