kantlivelong/OctoPrint-PSUControl #181: After latest upgrade can't get rid of setup wizard

szabocs, 2021-04-16:
PSU Control just prompted me for upgrade, so I did it.
After reboot I am getting a setup wizard with only a finish button on it.
If I click this button the screen stays and won't go away, basically rendering my
octoprint installation unusable and I cannot print at the moment.
I will try to manually remove the plugin for now because I need to print.

[image]

moarkaboom, 2021-04-16:
same issue for me. I deactivated all other display-plugins (themeify, etc.) and it still wont close.

valiobg, 2021-04-16:
I have the same problem.
[image]

valiobg, 2021-04-16:
~/oprint/bin/pip uninstall OctoPrint-PSUControl
This is a temporary fix.

kantlivelong, 2021-04-16:
I can't reproduce. Please provide the following info:

#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Version of OctoPrint-PSUControl

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

#### Browser console logs ( See https://www.wickedlysmart.com/hfjsconsole/ )

#### Wiring diagram

kantlivelong, 2021-04-16:
As a workaround you can modify the ~/.octoprint/config.yaml and set the following:

```
server:
  seenWizards:
    psucontrol: 1
```

Be sure to stop octoprint before editing.

d3relict, 2021-04-16:
console log:
ConfiguredLoggers has not changed. Not saving.
packed_core.js?22357642:431 Uncaught TypeError: Cannot read property 'done' of undefined
    at WizardViewModel.self.finishWizard (packed_core.js?22357642:431)
    at Object.onFinish (packed_core.js?22357642:430)
    at HTMLButtonElement.finish (packed_libs.js?ab739628:1209)
    at HTMLButtonElement.dispatch (packed_libs.js?ab739628:1)
    at HTMLButtonElement.v.handle (packed_libs.js?ab739628:1)

OctoPrint 1.5.3
Python 3.7.3
OctoPi 0.18.0

kantlivelong, 2021-04-16:
@d3relict thanks. It's not entirely clear where the issue lies as I cannot reproduce and the error is a bit cryptic due to the way OctoPrint compresses/packs JS.

I've made a quick change to test if you could try it: https://github.com/kantlivelong/OctoPrint-PSUControl/archive/d2ae20efef8f4d665f84475dd06198e78bd314d2.zip

d3relict, 2021-04-16:
i installed over my current version (without removal), and disabled the workaround you suggested above. i get the same error. do you need me to completely remove the plugin and install from the zip, or was this sufficient?

d3relict, 2021-04-16:
also, as far as i see, the js file is just serialized, after using "beautification" in chrome devtools, the source is completely readable.

```
        self.finishWizard = function() {
            var deferred = $.Deferred();
            self.finishing = true;
            self.settingsViewModel.saveData().done(function() {
                OctoPrint.wizard.finish(self.wizards).done(function() {
                    deferred.resolve(arguments);
                }).fail(function() {
                    deferred.reject(arguments);
                }).always(function() {
                    self.finishing = false;
                });
            }).fail(function() {
                deferred.reject(arguments);
            });
            return deferred;
        }
```
the error is thrown on line `self.settingsViewModel.saveData().done(function() {`, as `saveData()` returns undefined.

The-EG, 2021-04-16:
Getting this after trying the dev branch. Attached:
 - [op.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6326565/op.log): octoprint log with `--debug`
 - [config.yaml.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6326580/config.yaml.txt) (apparently github doesn't like yamls...)

Installed plugins:
```terminal
2021-04-16 12:15:21,046 - octoprint.plugin.core - INFO - 32 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Camera Settings (0.1.0rc1) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_CameraSettings
|  Consolidated Tabs (1.0.1) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_consolidatedtabs
|  Core Wizard (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Custom Background (1.0.0) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_custombackground
|  Discovery (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.26.0) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  File Check (2021.2.23) (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.2.4) (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_firmware_check
|  Fullscreen Plugin (0.0.6) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_fullscreen
|  GCode Viewer (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
| !GPIO Control (1.0.8) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_gpiocontrol
|  Hardware Buttons (1.0.0) = /home/taylor/.octoprint/plugins/octoprint_hwbuttons.py
|  Logging (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/logging
|  M73 Progress Plugin (0.2.1) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_m73progress
|  Octolapse (0.4.1) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (2021.3.26.post2) (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintTimeGenius Plugin (2.2.7) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  PrusaSlicer Thumbnails (0.1.4) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  PSU Control (1.0.0) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_psucontrol
|  Software Update (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  SpoolManager Plugin (1.3.3) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_SpoolManager
|  Themeify (1.2.2) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_themeify
|  UBL Mesh Editor (0.3.0) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint_ublmeshedit
|  Virtual Printer (bundled) = /home/taylor/OctoPrint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible

Note: 'Hardware Buttons' is a plugin that I wrote for running something on button press via GPIO. It uses gpiozero (and by depenency, RPi.GPIO).
```

Egocentrical, 2021-04-16:
I uininstalled SpoolManager and was able to click on the finish button if that helps anyone

d3relict, 2021-04-16:
i can confirm, disabling SpoolManager makes the modal work again.

kantlivelong, 2021-04-16:
Interesting. I had to step away from my dev env but if you have some time to spare I'm curious what happens if you disable psucontrol, enable spool manager, restart OctoPrint , install https://github.com/kantlivelong/OctoPrint-SSHInterface, then restart as usual after installs. That plugin also has a simple text wizard but uses builtin settings. If it happens with that too I'd lean towards SpoolManager being the culprit.

Once I return I will test it out myself as well.

The-EG, 2021-04-16:
I get the same issue with the wizard after installing SSHInterface with Spool Manager installed & enabled.

kantlivelong, 2021-04-16:
So I've just confirmed that this is something with Spool Manager. Simply re-enabling `firstRun` in `config.yaml` is enough to cause the issue. Will open with the dev.

valiobg, 2021-04-17:
As a temporary fix you can use this plugin to power on/off your printers - https://github.com/catgiggle/OctoPrint-GpioControl. I hope PSU Control will be fixed soon !

kantlivelong, 2021-04-17:
> As a temporary fix you can use this plugin to power on/off your printers - https://github.com/catgiggle/OctoPrint-GpioControl. I hope PSU Control will be fixed soon !

It's not a PSU Control issue. See above posts.

moarkaboom, 2021-04-17:
Updating Spoolmanager resolved for the issue for me. Thanks for the quick response!

reipet, 2021-04-17:
Hi, can you explain how to update or uninstall the SpoolManager to solve this Issue please. I can SSH into Octopi but then I dont know how to continue. THX!

kantlivelong, 2021-04-17:
Use Safe Mode then uninstall or upgrade.

https://docs.octoprint.org/en/master/features/safemode.html

moarkaboom, 2021-04-17:
(for me the easier way was):
1. when SSHd into your Octoprint, uninstall PSU Control:
` /home/pi/oprint/bin/pip uninstall --yes Octoprint-PSUControl
`
2. then restart the Service:
` sudo service octoprint restart
`
3. reload the WebUI, and run "Check for Update now" in Software Update in the Settings
4. first update Spool Manager
5. then update PSU Control
6. after that I had to restart my RPi to get everything working again properly.
7. done

reipet, 2021-04-17:
I had to play around a bit but finally got it working again. THX!

kantlivelong, 2021-04-18:
Going to close this now that Spool Manager has been updated. Will still keep pinned for a bit.

closed 2021-04-18 by kantlivelong