kantlivelong/OctoPrint-PSUControl #182: Sensing doesn't work if both Switching and Sensing use the same GPIO pin.

kantlivelong, 2021-04-16:
After 1.0.0 some users using switching and sensing on the same pin can no longer do so. Expected with Python-Periphery because pins are configured as objects.

Short term workaround is to use Internal sensing. Will need to investigate long term solution.

kantlivelong, 2021-04-18:
So this actually seems to have been a long running bug due to the way GPIO was configured. In previous versions with RPi.GPIO the sensing pin was setup then the switching was setup afterwards. This essentially overwrote the sensing pins direction from input to output but RPi.GPIO would still return a value.

Markeno76, 2021-04-18:
Would feeding Switching Pin back to another GPIO Pin for Sensing be a suggested way to still use GPIO Sensing?  Internal will make it functional for now for me, but I very often trigger a manual power up at the printer with the Enclosure Plugin.  It seems if PSUControl doesn't think the printer is powered on that it will automatically power it down.

kantlivelong, 2021-04-18:
> Would feeding Switching Pin back to another GPIO Pin for Sensing be a suggested way to still use GPIO Sensing? Internal will make it functional for now for me, but I very often trigger a manual [...]

You can bridge an adjacent GPIO pin and use that sure.

kantlivelong, 2021-04-19:
PSUControl-RPi.GPIO 1.0.2 has been released which brings this "bug" old behavior back.