kantlivelong/OctoPrint-PSUControl #183: Add downgrade instructions to a working version

stablestud, 2021-04-16:
Hi,

unfortunately the new update is broken.
In another issue you where suggesting to do a downgrade via the release tab.
However its missing instructions how to do so? Me and probably others don't know what to do with that zip nor with its contents.

Can you please add them?
Thank you.

cp2004, 2021-04-16:
It depends, you can either install the fixed versions in the relevant issues, or work around the wizard as set out in the relevant tickets: #180 and #181, or downgrade. Or, even better, provide more details so that if you are having an issue it can be fixed.

These URLs for fixes or to downgrade can be entered into the plugin manager -> get more > from URL.

kumters, 2021-04-16:
I was finally able to get it working with the subplugin, but subplugin installation seems broken = I had to restart the whole RPI to get that subplugin available to configure....

kantlivelong, 2021-04-16:
That's already been figured out in https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO/issues/3

Just give me some time to whip in a good solution.

kantlivelong, 2021-04-17:
Version [1.0.1](https://github.com/kantlivelong/OctoPrint-PSUControl/releases/tag/1.0.1) has been released which addresses built-in GPIO bugs. If on a Pi you really shouldn't need **PSU Control - RPi.GPIO**.

If you have the wizard issue though it will remain as it's due to another plugin.

closed 2021-04-17 by kantlivelong