kantlivelong/OctoPrint-PSUControl #184: GPIO Error

HRjunior, 2021-04-17:
As i just updated to 1.0.1, i can't turn on my PSU anymore

Even if PSUControl point to pin n°3, it does trigger the N°5, and turn on the led everytime

araziforum, 2021-04-17:
You have to install https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO/archive/master.zip

kantlivelong, 2021-04-17:
Please provide the information required when opening the ticket.

kantlivelong, 2021-04-17:
> You have to install https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO/archive/master.zip

That's not a solution it's a fallback. GPIO should work without that plugin.

wick3r, 2021-04-17:
I have the same issue. Updated to 1.0.1 and now my psu doesn't come on automatically with power-on, and when i press the lightning bolt nothing happens. It was working last night with 1.0.0, and it seems to have stopped working with 1.0.1.

I don't have the Rpi-Gpio.

Markeno76, 2021-04-17:
How does one use the master.zip archive to go back to 1.0?  I have the same issues.  Installed the update, just before starting a print now I can't use the lightingbolt to power on the printer.  I am concerned if I manage to power on the printer it will mistakenly turn it off at some point now.

I am not happy to have used this great plugin all this time, and the updates to warns it has made "some change" in how it interacts with the gpio AFTER installing the update, without a way to easily go back.  What was the change referring to and where would someone read details about it?

MichiTheSailor, 2021-04-17:
Ciao,
I have the same issue :o(
It worked fine with 0.1.11 until yesterday - Yesterday update to 1.0.0 and id did not work anymore
I tried and installed the workaround with RPiGPIO but and also the update to 1.0.1 but no success.
If I look in the log I got a lot of strange messages:
2021-04-17 22:28:05,995 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:28:06,132 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:08,269 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:28:08,375 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:13,382 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:18,389 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:23,397 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:28,403 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:33,408 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:38,415 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:43,420 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 22:28:43,652 - octoprint.plugins.psucontrol_rpigpio - INFO - Configuring GPIO for pin 40
2021-04-17 22:28:43,652 - octoprint.plugins.psucontrol_rpigpio - INFO - Configuring GPIO for pin 40
2021-04-17 22:29:00,396 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:29:00,400 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:29:03,772 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:29:03,776 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:29:36,566 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:29:36,569 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:29:39,327 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:29:39,329 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:30:45,653 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:30:45,654 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:30:47,059 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:30:47,061 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:31:07,330 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:12,346 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:12,846 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:31:12,848 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:12,974 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:17,984 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:19,738 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:31:19,741 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:19,863 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:24,877 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:29,883 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:34,890 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
2021-04-17 22:31:40,028 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:31:43,496 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:31:43,497 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:31:43,649 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:31:46,972 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 22:31:46,974 - octoprint.plugins.psucontrol_rpigpio - ERROR - The GPIO channel has not been set up as an OUTPUT
2021-04-17 22:31:47,094 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:31:52,114 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:31:57,120 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:02,124 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:07,132 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:12,142 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:17,151 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:22,157 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:27,163 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:32,168 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-17 22:32:33,640 - octoprint.plugins.psucontrol_rpigpio - ERROR - 40 is not in list
[... cut ...]
2021-04-17 22:32:33,642 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 40
2021-04-17 22:32:46,830 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
[octoprint.log-2.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6330029/octoprint.log-2.txt)
[Systeminfo.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6330031/Systeminfo.txt)
[Setting.docx](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6330035/Setting.docx)
Please help to recover functionality.
Thanks for your support
Michi

kantlivelong, 2021-04-17:
@MichiTheSailor Open a new ticket. If you want to keep using PSUControl-RPi.GPIO open it at https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO/. Otherwise remove that and open a ticket here with all the required info.

Going to close this dumpster of a ticket. Above goes for everyone else.

closed 2021-04-17 by kantlivelong