kantlivelong/OctoPrint-PSUControl #185: no BCM Numbering option anymore in 1.0.1, not working after update

RetroCare, 2021-04-17:
I only have the option for /dev/gpiochip0 in GPIO Device.
I am using GPIO pin 3 (gpio2)
The button does show On and Off, it just doesnt switch my Relay anymore, also measured with a multimeter I could see that it does nothing, im not sure if its because BCM is not a option to choose (I have no options in the list of devices, it just says /dev/gpiochip0)

OctoPrint 1.5.3 Python 3.7.3 OctoPi 1.0.0

2021-04-17 13:51:14,243 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 13:51:14,244 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2021-04-17 13:51:24,247 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 13:51:24,247 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2021-04-17 13:51:26,139 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 13:51:26,140 - octoprint.plugins.psucontrol - DEBUG - Switching PSU On Using GPIO: 3
2021-04-17 13:51:34,251 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 13:51:34,252 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-04-17 13:51:34,252 - octoprint.plugins.psucontrol - DEBUG - PSU state changed, firing psu_state_changed event.
2021-04-17 13:51:44,262 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 13:51:44,263 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-04-17 13:51:44,298 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2021-04-17 13:51:44,299 - octoprint.plugins.psucontrol - DEBUG - Switching PSU Off Using GPIO: 3
2021-04-17 13:51:44,554 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2021-04-17 13:51:44,572 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-04-17 13:51:44,686 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 13:51:44,686 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2021-04-17 13:51:44,687 - octoprint.plugins.psucontrol - DEBUG - PSU state changed, firing psu_state_changed event.

kantlivelong, 2021-04-17:
Shouldn't you have a 2 there then instead of a 3?

RetroCare, 2021-04-17:
added additional "new" log
[PSU_Control_log.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6329193/PSU_Control_log.log)

RetroCare, 2021-04-17:
> Shouldn't you have a 2 there then instead of a 3?

its on hardware pin 3 , that pin name is GPIO2

changed to 2 but doesnt work, in case it could be my relay I will change it out and update here on the progress.
Is not having the option for BCM correct? should it work with these settings?

kantlivelong, 2021-04-17:
BOARD numbering waa dropped as it's Pi specific. So in the Pi world it's using BCM numbering now. It should have been automatically converted on upgrade.

RetroCare, 2021-04-17:
im closing this issue,

coincidence that my relay broke the day I updated PSU Control, changed out my relay and works now.

closed 2021-04-17 by RetroCare

kantlivelong, 2021-04-17:
Ouch!

RetroCare, 2021-04-17:
also about the GPIO pin I also had it wrong, I had to set it to the GPIO pin not the hardware pin number :), guess I changed that after fiddling around when it broke after the update