kantlivelong/OctoPrint-PSUControl #187: PSU won't turn on after 1.0.1 update

wick3r, 2021-04-17:
#### What were you doing?

1. I powered on the raspberry pi 
2. Then clicked the lightning bolt button

#### What did you expect to happen?

I expected the PSU to power on automatically with the pi when the pi powered on.

When that didn't happen, I expected the PSU to power on when I clicked the lightning bolt button.

#### What happened instead?

The PSU did not power on

#### Version of OctoPrint-PSUControl

Version 1.0.1

#### Operating System running OctoPrint

Octopi

#### Printer model & used firmware incl. version

Ender 3 pro, Marlin 2.0.7

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

https://pastebin.com/TW8kAGmp

#### Wiring diagram

PSU ---  Relay ---- Ender 3
    I                I
  R Pi       Rpi Gpio 3

kantlivelong, 2021-04-17:
Logging is not set to debug.

wick3r, 2021-04-17:
Sorry, I'm new to this. Did it work this time? https://pastebin.com/neQyZNu8

kantlivelong, 2021-04-17:
Thanks.

So to confirm. Based on [this diagram](https://roboticsbackend.com/wp-content/uploads/2019/05/raspberry-pi-3-pinout.jpg) You are using GPIO 3 which is Pin 5 on the board?

wick3r, 2021-04-17:
I was using the "bcm" , sorry for the confusion: so GPIO 2, pin 3 on the board.

kantlivelong, 2021-04-17:
All good then?

wick3r, 2021-04-17:
No, the hardware has been connected correctly to GPIO 2, pin 3 on the board. The PSU doesn't work.

kantlivelong, 2021-04-17:
So you changed the 3 to a 2 and it still doesn't work?

wick3r, 2021-04-17:
No, I made no changes. It has always been 2. (I just miswrote it as 3 previously, thinking in terms of board pins). It was working before the update, and stopped working after the update, with no changes to the hardware.

kantlivelong, 2021-04-17:
The log says it's using gpio 3.

wick3r, 2021-04-17:
You're brilliant hahah. I must have changed it by mistake. It works now! Thanks so much for your time.

closed 2021-04-17 by kantlivelong