kantlivelong/OctoPrint-PSUControl #188: GPIO sensing not working but python can see pin properly

whinis, 2021-04-17:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?
Using GPIO Sensing green lightning should work.

#### What happened instead?

PSU Control continues to think the PSU is off

#### Version of OctoPrint-PSUControl

1.0.1
#### Operating System running OctoPrint

Rapbian 10.8

#### Printer model & used firmware incl. version

Folgertech Ft-5 but the PSU and Sensing is entirely custom.

#### Wiring diagram

Gnd/Board Pin 34 -> SSR Relay Ground
Board Pin 26/GPIO16 -> SSR relay + Cntrl

PSU 24v->LD7805 -> LD1117LD33v ->Board Pin32/GPIO12

Python seems to have no issue seeing the state of the pin
`Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
>>> GPIO.setmode(GPIO.BCM)
>>> GPIO.setup(12, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
__main__:1: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
>>> GPIO.input(12)
1
>>> GPIO.input(12)
0
>>> GPIO.input(12)
1
>>> GPIO.input(12)
1
`

kantlivelong, 2021-04-17:
Link to octoprint.log with octoprint.plugins.psucontrol set to DEBUG

whinis, 2021-04-17:
https://cloud.whinis.com/index.php/s/XXQSdE5iTNHBRXM

kantlivelong, 2021-04-17:
You're going to want to restart OctoPrint to capture full logs. Nothing of use in there really.

Set debug
restart octoprint
reproduce
send logs

whinis, 2021-04-17:
Rebooting Octoprint seemed to fix the issue, however here is the log

```
2021-04-17 16:37:28,506 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-04-17 16:37:28,553 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-04-17 16:37:28,554 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-04-17 16:37:28,554 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-04-17 16:37:37,249 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-04-17 16:37:40,301 - octoprint.util.pip - INFO - Using "/home/octoprint/python_venvs/octoprint1/bin/python3 -m pip" as command to invoke pip
2021-04-17 16:37:43,138 - octoprint.util.pip - INFO - Version of pip is 20.0.2
2021-04-17 16:37:43,152 - octoprint.util.pip - INFO - pip installs to /home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-04-17 16:37:43,153 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-04-17 16:37:43,496 - octoprint.plugins.psucontrol - DEBUG - GPIODevice: /dev/gpiochip0
2021-04-17 16:37:43,497 - octoprint.plugins.psucontrol - DEBUG - switchingMethod: GPIO
2021-04-17 16:37:43,499 - octoprint.plugins.psucontrol - DEBUG - onoffGPIOPin: 16
2021-04-17 16:37:43,500 - octoprint.plugins.psucontrol - DEBUG - invertonoffGPIOPin: 0
2021-04-17 16:37:43,503 - octoprint.plugins.psucontrol - DEBUG - onGCodeCommand: M80
2021-04-17 16:37:43,506 - octoprint.plugins.psucontrol - DEBUG - offGCodeCommand: M81
2021-04-17 16:37:43,509 - octoprint.plugins.psucontrol - DEBUG - onSysCommand: 
2021-04-17 16:37:43,512 - octoprint.plugins.psucontrol - DEBUG - offSysCommand: 
2021-04-17 16:37:43,515 - octoprint.plugins.psucontrol - DEBUG - switchingPlugin: 
2021-04-17 16:37:43,517 - octoprint.plugins.psucontrol - DEBUG - enablePseudoOnOff: 1
2021-04-17 16:37:43,520 - octoprint.plugins.psucontrol - DEBUG - pseudoOnGCodeCommand: M80
2021-04-17 16:37:43,523 - octoprint.plugins.psucontrol - DEBUG - pseudoOffGCodeCommand: M81
2021-04-17 16:37:43,526 - octoprint.plugins.psucontrol - DEBUG - postOnDelay: 5.0
2021-04-17 16:37:43,529 - octoprint.plugins.psucontrol - DEBUG - connectOnPowerOn: 0
2021-04-17 16:37:43,532 - octoprint.plugins.psucontrol - DEBUG - disconnectOnPowerOff: 0
2021-04-17 16:37:43,535 - octoprint.plugins.psucontrol - DEBUG - sensingMethod: INTERNAL
2021-04-17 16:37:43,537 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPin: 12
2021-04-17 16:37:43,540 - octoprint.plugins.psucontrol - DEBUG - sensePollingInterval: 5
2021-04-17 16:37:43,543 - octoprint.plugins.psucontrol - DEBUG - invertsenseGPIOPin: 0
2021-04-17 16:37:43,546 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPinPUD: PULL_DOWN
2021-04-17 16:37:43,549 - octoprint.plugins.psucontrol - DEBUG - senseSystemCommand: 
2021-04-17 16:37:43,552 - octoprint.plugins.psucontrol - DEBUG - sensingPlugin: 
2021-04-17 16:37:43,556 - octoprint.plugins.psucontrol - DEBUG - autoOn: 1
2021-04-17 16:37:43,559 - octoprint.plugins.psucontrol - DEBUG - autoOnTriggerGCodeCommands: G0,G1,G2,G3,G10,G11,G28,G29,G32,M104,M106,M109,M140,M190
2021-04-17 16:37:43,562 - octoprint.plugins.psucontrol - DEBUG - enablePowerOffWarningDialog: 1
2021-04-17 16:37:43,565 - octoprint.plugins.psucontrol - DEBUG - powerOffWhenIdle: 1
2021-04-17 16:37:43,568 - octoprint.plugins.psucontrol - DEBUG - idleTimeout: 30
2021-04-17 16:37:43,571 - octoprint.plugins.psucontrol - DEBUG - idleIgnoreCommands: M105
2021-04-17 16:37:43,574 - octoprint.plugins.psucontrol - DEBUG - idleTimeoutWaitTemp: 50
2021-04-17 16:37:43,673 - octoprint.plugins.pi_support - WARNING - Fetching the current throttle state via "/usr/bin/vcgencmd get_throttled" doesn't work
2021-04-17 16:37:45,872 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2021-04-17 16:37:45,994 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2021-04-17 16:37:46,778 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._http._tcp.local.' for _http._tcp.local.
2021-04-17 16:37:47,615 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._octoprint._tcp.local.' for _octoprint._tcp.local.
2021-04-17 16:37:47,648 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on raspberrypi for SSDP
2021-04-17 16:37:47,776 - octoprint.plugins.pi_support - WARNING - Fetching the current throttle state via "/usr/bin/vcgencmd get_throttled" doesn't work
2021-04-17 16:37:48,212 - tornado.access - WARNING - 404 GET /intermediary.gif?_=1618691865431 (192.168.1.40) 222.88ms
2021-04-17 16:37:48,344 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2021-04-17 16:37:48,569 - tornado.access - WARNING - 404 GET /intermediary.gif?_=1618691867932 (192.168.1.40) 188.83ms
2021-04-17 16:37:48,904 - tornado.access - WARNING - 404 GET /cached.gif?_=1618691868017 (192.168.1.40) 276.65ms
2021-04-17 16:37:50,271 - tornado.access - WARNING - 404 GET /cached.gif?_=1618691869582 (192.168.1.40) 326.32ms
2021-04-17 16:37:51,429 - tornado.access - WARNING - 404 GET /cached.gif?_=1618691870924 (192.168.1.40) 153.92ms
[... cut ...]
2021-04-17 16:39:14,441 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 11.14ms
2021-04-17 16:39:14,459 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 9.26ms
2021-04-17 16:39:14,989 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 16:39:14,990 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2021-04-17 16:39:19,992 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 16:39:19,993 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
```

kantlivelong, 2021-04-17:
Config shows `sensingMethod: INTERNAL` and not GPIO

whinis, 2021-04-17:
Oh, I forgot to change it back 1 moment

whinis, 2021-04-17:
Updated Log, now not working obviously since im not using internal

`2021-04-17 16:49:46,997 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-04-17 16:49:47,043 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-04-17 16:49:47,044 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-04-17 16:49:47,044 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-04-17 16:49:54,075 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-04-17 16:49:56,849 - octoprint.util.pip - INFO - Using "/home/octoprint/python_venvs/octoprint1/bin/python3 -m pip" as command to invoke pip
2021-04-17 16:49:59,440 - octoprint.util.pip - INFO - Version of pip is 20.0.2
2021-04-17 16:49:59,447 - octoprint.util.pip - INFO - pip installs to /home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-04-17 16:49:59,447 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-04-17 16:49:59,704 - octoprint.plugins.psucontrol - DEBUG - GPIODevice: /dev/gpiochip0
2021-04-17 16:49:59,706 - octoprint.plugins.psucontrol - DEBUG - switchingMethod: GPIO
2021-04-17 16:49:59,707 - octoprint.plugins.psucontrol - DEBUG - onoffGPIOPin: 16
2021-04-17 16:49:59,708 - octoprint.plugins.psucontrol - DEBUG - invertonoffGPIOPin: 0
2021-04-17 16:49:59,709 - octoprint.plugins.psucontrol - DEBUG - onGCodeCommand: M80
2021-04-17 16:49:59,713 - octoprint.plugins.psucontrol - DEBUG - offGCodeCommand: M81
2021-04-17 16:49:59,717 - octoprint.plugins.psucontrol - DEBUG - onSysCommand: 
2021-04-17 16:49:59,719 - octoprint.plugins.psucontrol - DEBUG - offSysCommand: 
2021-04-17 16:49:59,720 - octoprint.plugins.psucontrol - DEBUG - switchingPlugin: 
2021-04-17 16:49:59,721 - octoprint.plugins.psucontrol - DEBUG - enablePseudoOnOff: 1
2021-04-17 16:49:59,723 - octoprint.plugins.psucontrol - DEBUG - pseudoOnGCodeCommand: M80
2021-04-17 16:49:59,726 - octoprint.plugins.psucontrol - DEBUG - pseudoOffGCodeCommand: M81
2021-04-17 16:49:59,730 - octoprint.plugins.psucontrol - DEBUG - postOnDelay: 5.0
2021-04-17 16:49:59,733 - octoprint.plugins.psucontrol - DEBUG - connectOnPowerOn: 0
2021-04-17 16:49:59,739 - octoprint.plugins.psucontrol - DEBUG - disconnectOnPowerOff: 0
2021-04-17 16:49:59,742 - octoprint.plugins.psucontrol - DEBUG - sensingMethod: GPIO
2021-04-17 16:49:59,746 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPin: 12
2021-04-17 16:49:59,749 - octoprint.plugins.psucontrol - DEBUG - sensePollingInterval: 5
2021-04-17 16:49:59,752 - octoprint.plugins.psucontrol - DEBUG - invertsenseGPIOPin: 0
2021-04-17 16:49:59,754 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPinPUD: PULL_DOWN
2021-04-17 16:49:59,758 - octoprint.plugins.psucontrol - DEBUG - senseSystemCommand: 
2021-04-17 16:49:59,762 - octoprint.plugins.psucontrol - DEBUG - sensingPlugin: 
2021-04-17 16:49:59,764 - octoprint.plugins.psucontrol - DEBUG - autoOn: 1
2021-04-17 16:49:59,765 - octoprint.plugins.psucontrol - DEBUG - autoOnTriggerGCodeCommands: G0,G1,G2,G3,G10,G11,G28,G29,G32,M104,M106,M109,M140,M190
2021-04-17 16:49:59,766 - octoprint.plugins.psucontrol - DEBUG - enablePowerOffWarningDialog: 1
2021-04-17 16:49:59,771 - octoprint.plugins.psucontrol - DEBUG - powerOffWhenIdle: 1
2021-04-17 16:49:59,774 - octoprint.plugins.psucontrol - DEBUG - idleTimeout: 30
2021-04-17 16:49:59,777 - octoprint.plugins.psucontrol - DEBUG - idleIgnoreCommands: M105
2021-04-17 16:49:59,780 - octoprint.plugins.psucontrol - DEBUG - idleTimeoutWaitTemp: 50
2021-04-17 16:49:59,910 - octoprint.plugins.pi_support - WARNING - Fetching the current throttle state via "/usr/bin/vcgencmd get_throttled" doesn't work
2021-04-17 16:50:03,002 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2021-04-17 16:50:03,132 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2021-04-17 16:50:04,150 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._http._tcp.local.' for _http._tcp.local.
2021-04-17 16:50:05,057 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._octoprint._tcp.local.' for _octoprint._tcp.local.
2021-04-17 16:50:05,084 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on raspberrypi for SSDP
2021-04-17 16:50:05,180 - octoprint.plugins.pi_support - WARNING - Fetching the current throttle state via "/usr/bin/vcgencmd get_throttled" doesn't work
2021-04-17 16:50:05,726 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2021-04-17 16:50:08,960 - octoprint.util.pip - INFO - Using "/home/octoprint/python_venvs/octoprint1/bin/python3 -m pip" as command to invoke pip
2021-04-17 16:50:08,989 - octoprint.util.pip - INFO - pip installs to /home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-04-17 16:50:08,989 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-04-17 16:50:09,103 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 462.0MB. That is considered sufficient for updating.
[... cut ...]
2021-04-17 16:51:24,188 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 16:51:24,189 - octoprint.plugins.psucontrol - DEBUG - Result: 0
2021-04-17 16:51:24,189 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: 0
2021-04-17 16:51:25,480 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 9.33ms
2021-04-17 16:51:25,498 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 9.92ms
2021-04-17 16:51:25,515 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 9.37ms
`

kantlivelong, 2021-04-17:
`2021-04-17 16:50:09,510 - octoprint.plugins.psucontrol - ERROR - [Errno 16] Opening input line handle: Device or resource busy`

What happens if you set sense to float instead of pull_down? (supply logs again)

whinis, 2021-04-17:
Same Issue, Log starting from change

```2021-04-17 17:04:19,955 - octoprint.plugins.psucontrol - DEBUG - GPIODevice: /dev/gpiochip0
2021-04-17 17:04:19,956 - octoprint.plugins.psucontrol - DEBUG - switchingMethod: GPIO
2021-04-17 17:04:19,957 - octoprint.plugins.psucontrol - DEBUG - onoffGPIOPin: 16
2021-04-17 17:04:19,958 - octoprint.plugins.psucontrol - DEBUG - invertonoffGPIOPin: 0
2021-04-17 17:04:19,959 - octoprint.plugins.psucontrol - DEBUG - onGCodeCommand: M80
2021-04-17 17:04:19,959 - octoprint.plugins.psucontrol - DEBUG - offGCodeCommand: M81
2021-04-17 17:04:19,960 - octoprint.plugins.psucontrol - DEBUG - onSysCommand: 
2021-04-17 17:04:19,967 - octoprint.plugins.psucontrol - DEBUG - offSysCommand: 
2021-04-17 17:04:19,971 - octoprint.plugins.psucontrol - DEBUG - switchingPlugin: 
2021-04-17 17:04:19,975 - octoprint.plugins.psucontrol - DEBUG - enablePseudoOnOff: 1
2021-04-17 17:04:19,977 - octoprint.plugins.psucontrol - DEBUG - pseudoOnGCodeCommand: M80
2021-04-17 17:04:19,978 - octoprint.plugins.psucontrol - DEBUG - pseudoOffGCodeCommand: M81
2021-04-17 17:04:19,980 - octoprint.plugins.psucontrol - DEBUG - postOnDelay: 5.0
2021-04-17 17:04:19,982 - octoprint.plugins.psucontrol - DEBUG - connectOnPowerOn: 0
2021-04-17 17:04:19,984 - octoprint.plugins.psucontrol - DEBUG - disconnectOnPowerOff: 0
2021-04-17 17:04:19,986 - octoprint.plugins.psucontrol - DEBUG - sensingMethod: GPIO
2021-04-17 17:04:19,988 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPin: 12
2021-04-17 17:04:19,990 - octoprint.plugins.psucontrol - DEBUG - sensePollingInterval: 5
2021-04-17 17:04:19,991 - octoprint.plugins.psucontrol - DEBUG - invertsenseGPIOPin: 0
2021-04-17 17:04:19,992 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPinPUD: 
2021-04-17 17:04:19,992 - octoprint.plugins.psucontrol - DEBUG - senseSystemCommand: 
2021-04-17 17:04:19,993 - octoprint.plugins.psucontrol - DEBUG - sensingPlugin: 
2021-04-17 17:04:19,994 - octoprint.plugins.psucontrol - DEBUG - autoOn: 1
2021-04-17 17:04:19,996 - octoprint.plugins.psucontrol - DEBUG - autoOnTriggerGCodeCommands: G0,G1,G2,G3,G10,G11,G28,G29,G32,M104,M106,M109,M140,M190
2021-04-17 17:04:19,997 - octoprint.plugins.psucontrol - DEBUG - enablePowerOffWarningDialog: 1
2021-04-17 17:04:19,999 - octoprint.plugins.psucontrol - DEBUG - powerOffWhenIdle: 1
2021-04-17 17:04:20,004 - octoprint.plugins.psucontrol - DEBUG - idleTimeout: 30
2021-04-17 17:04:20,006 - octoprint.plugins.psucontrol - DEBUG - idleIgnoreCommands: M105
2021-04-17 17:04:20,007 - octoprint.plugins.psucontrol - DEBUG - idleTimeoutWaitTemp: 50
2021-04-17 17:04:20,009 - octoprint.plugins.psucontrol - DEBUG - Cleaning up pin GPIO 16 (name="", label="periphery", device=/dev/gpiochip0, line_fd=23, chip_fd=21, direction=out, edge=none, bias=default, drive=default, inverted=False, chip_name="gpiochip0", chip_label="pinctrl-bcm2835", type=cdev)
2021-04-17 17:04:20,012 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2021-04-17 17:04:20,013 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 16
2021-04-17 17:04:20,015 - octoprint.plugins.psucontrol - INFO - Using GPIO sensing to determine PSU on/off state.
2021-04-17 17:04:20,016 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 12
2021-04-17 17:04:20,019 - octoprint.plugins.psucontrol - ERROR - [Errno 16] Opening input line handle: Device or resource busy
2021-04-17 17:04:24,716 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 17:04:24,717 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 17:04:24,717 - octoprint.plugins.psucontrol - DEBUG - Result: 0
2021-04-17 17:04:24,717 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: 0
2021-04-17 17:04:25,508 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 9.54ms
2021-04-17 17:04:25,528 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 11.03ms
2021-04-17 17:04:25,546 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 9.91ms
2021-04-17 17:04:27,717 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2021-04-17 17:04:27,719 - octoprint.plugins.psucontrol - DEBUG - Switching PSU On Using GPIO: 16
2021-04-17 17:04:29,720 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 17:04:29,720 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 17:04:29,720 - octoprint.plugins.psucontrol - DEBUG - Result: 0
2021-04-17 17:04:29,720 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: 0
2021-04-17 17:04:32,829 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-17 17:04:32,830 - octoprint.plugins.psucontrol - ERROR - 'sense'
2021-04-17 17:04:32,830 - octoprint.plugins.psucontrol - DEBUG - Result: 0
2021-04-17 17:04:32,830 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: 0```

kantlivelong, 2021-04-17:
What kernel version are you on?
`uname -a`

whinis, 2021-04-17:
`Linux raspberrypi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux`

whinis, 2021-04-17:
Also note, I get an error in python as well that you saw 
```main:1: RuntimeWarning: This channel is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.```

But I can still read the pin

kantlivelong, 2021-04-17:
Kernel seems fine. Do you have GPIO12 being read/written to anywhere else?

whinis, 2021-04-17:
Not that I know of, also not really sure how to check. This particular RPI is hard wired into my printer at this point

kantlivelong, 2021-04-17:
Hmm. I can't see any reason for it. You can try https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO to see if that works. Just be sure to restart OctoPrint after making changes.

kantlivelong, 2021-04-21:
Hey @whinis 

Curious if this still occurs with 1.0.4?

whinis, 2021-04-21:
Just checked, it is still occurring with 1.0.4

kantlivelong, 2021-04-21:
> 
> 
> Just checked, it is still occurring with 1.0.4

PSUControl-RPiGPIO works though correct?

whinis, 2021-04-21:
Let me check, my freetime has been a bit limited the past few days

kantlivelong, 2021-04-21:
No worries. 👍

whinis, 2021-04-21:
Gonna be honest, I just installed it and cannot figure out how to use it.
I assume you set sensing to plugin but the dropdown for plugin is empty

I also attempted going to the RPi.GPIO plugin page but its blank.
No obvious errors in the log related to the plugin however

kantlivelong, 2021-04-21:
Sounds like it didn't register with PSUControl which might be due to a similar error.

This is on a Raspberry Pi correct?

whinis, 2021-04-21:
Correct, specifically a 3b+

kantlivelong, 2021-04-21:
When you have time can enable DEBUG for both `psucontrol` and `psucontrol_rpigpio`

Instructions: https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#how-to-provide-logs

Also if you can provide a process list (`ps auxww` via SSH).

whinis, 2021-04-21:
So I wish I could tell you what broke but I enabled debugging and restarted the server and it just started working.
And yes the plugin sensing works correctly.

whinis, 2021-04-21:
https://pastebin.com/iAu8Eenk
Proccess list

kantlivelong, 2021-04-22:
Only thing that really stands out is
```
pi        1393  0.0  1.8 159036 17908 ?        Sl   Apr17   0:01 zenity --info --text=Your Raspbian system has been upgraded to the latest version.  To ensure compatibility with the new version, some configuration files have been overwritten - if you had customised your system, some of your changes may have been lost.  Your original configuration files have been backed up and put in the directory /home/pi/oldconffiles
```

Not that it itself would be the cause but perhaps the device was between updates and needed a reboot?

Other than that I'd just like to ask if you could test both PSUControl w/ GPIO and PSUControl-RPiGPIO.

whinis, 2021-04-26:
So the device has been rebooted and that still shows up. May be showing on the X server which I cannot exactly see.
With PSUControl-RPiGPIO the sensing works without issue, even had a few prints performed to insure its working.

PSUControl w/ GPIO always has that error in log but I have no idea what device or resource is busy.

kantlivelong, 2021-04-26:
90% sure it's referring to the GPIO character device.

The latest changes in the devel branch bring in better exception logging which should help tell us if you're willing to give it a go.

https://github.com/kantlivelong/OctoPrint-PSUControl/archive/74e7a2a69a9a920bd43229eabfcbe19f5fc987f8.zip

whinis, 2021-04-26:
```2021-04-26 18:24:00,140 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-04-26 18:24:00,187 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-04-26 18:24:00,188 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-04-26 18:24:00,188 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-04-26 18:24:12,638 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-04-26 18:24:15,404 - octoprint.util.pip - INFO - Using "/home/octoprint/python_venvs/octoprint1/bin/python3 -m pip" as command to invoke pip
2021-04-26 18:24:18,149 - octoprint.util.pip - INFO - Version of pip is 20.0.2
2021-04-26 18:24:18,151 - octoprint.util.pip - INFO - pip installs to /home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-04-26 18:24:18,152 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-04-26 18:24:18,285 - octoprint.plugins.psucontrol - DEBUG - GPIODevice: /dev/gpiochip0
2021-04-26 18:24:18,286 - octoprint.plugins.psucontrol - DEBUG - switchingMethod: GPIO
2021-04-26 18:24:18,286 - octoprint.plugins.psucontrol - DEBUG - onoffGPIOPin: 16
2021-04-26 18:24:18,287 - octoprint.plugins.psucontrol - DEBUG - invertonoffGPIOPin: False
2021-04-26 18:24:18,288 - octoprint.plugins.psucontrol - DEBUG - onGCodeCommand: M80
2021-04-26 18:24:18,288 - octoprint.plugins.psucontrol - DEBUG - offGCodeCommand: M81
2021-04-26 18:24:18,289 - octoprint.plugins.psucontrol - DEBUG - onSysCommand: 
2021-04-26 18:24:18,290 - octoprint.plugins.psucontrol - DEBUG - offSysCommand: 
2021-04-26 18:24:18,297 - octoprint.plugins.psucontrol - DEBUG - switchingPlugin: None
2021-04-26 18:24:18,298 - octoprint.plugins.psucontrol - DEBUG - enablePseudoOnOff: True
2021-04-26 18:24:18,301 - octoprint.plugins.psucontrol - DEBUG - pseudoOnGCodeCommand: M80
2021-04-26 18:24:18,303 - octoprint.plugins.psucontrol - DEBUG - pseudoOffGCodeCommand: M81
2021-04-26 18:24:18,304 - octoprint.plugins.psucontrol - DEBUG - postOnDelay: 5.0
2021-04-26 18:24:18,304 - octoprint.plugins.psucontrol - DEBUG - connectOnPowerOn: False
2021-04-26 18:24:18,305 - octoprint.plugins.psucontrol - DEBUG - disconnectOnPowerOff: False
2021-04-26 18:24:18,305 - octoprint.plugins.psucontrol - DEBUG - sensingMethod: GPIO
2021-04-26 18:24:18,306 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPin: 12
2021-04-26 18:24:18,307 - octoprint.plugins.psucontrol - DEBUG - sensePollingInterval: 5
2021-04-26 18:24:18,311 - octoprint.plugins.psucontrol - DEBUG - invertsenseGPIOPin: False
2021-04-26 18:24:18,313 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPinPUD: 
2021-04-26 18:24:18,316 - octoprint.plugins.psucontrol - DEBUG - senseSystemCommand: 
2021-04-26 18:24:18,319 - octoprint.plugins.psucontrol - DEBUG - sensingPlugin: psucontrol_rpigpio
2021-04-26 18:24:18,322 - octoprint.plugins.psucontrol - DEBUG - autoOn: True
2021-04-26 18:24:18,325 - octoprint.plugins.psucontrol - DEBUG - autoOnTriggerGCodeCommands: G0,G1,G2,G3,G10,G11,G28,G29,G32,M104,M106,M109,M140,M190
2021-04-26 18:24:18,328 - octoprint.plugins.psucontrol - DEBUG - enablePowerOffWarningDialog: True
2021-04-26 18:24:18,329 - octoprint.plugins.psucontrol - DEBUG - powerOffWhenIdle: True
2021-04-26 18:24:18,330 - octoprint.plugins.psucontrol - DEBUG - idleTimeout: 10
2021-04-26 18:24:18,330 - octoprint.plugins.psucontrol - DEBUG - idleIgnoreCommands: M105
2021-04-26 18:24:18,331 - octoprint.plugins.psucontrol - DEBUG - idleTimeoutWaitTemp: 50
2021-04-26 18:24:18,332 - octoprint.plugins.psucontrol - DEBUG - turnOnWhenApiUploadPrint: False
2021-04-26 18:24:18,332 - octoprint.plugins.psucontrol - DEBUG - turnOffWhenError: False
2021-04-26 18:24:18,334 - octoprint.plugins.psucontrol_rpigpio - DEBUG - GPIOMode: BCM
2021-04-26 18:24:18,335 - octoprint.plugins.psucontrol_rpigpio - DEBUG - onoffGPIOPin: 0
2021-04-26 18:24:18,336 - octoprint.plugins.psucontrol_rpigpio - DEBUG - invertonoffGPIOPin: False
2021-04-26 18:24:18,338 - octoprint.plugins.psucontrol_rpigpio - DEBUG - senseGPIOPin: 12
2021-04-26 18:24:18,341 - octoprint.plugins.psucontrol_rpigpio - DEBUG - invertsenseGPIOPin: False
2021-04-26 18:24:18,343 - octoprint.plugins.psucontrol_rpigpio - DEBUG - senseGPIOPinPUD: 
2021-04-26 18:24:18,345 - octoprint.plugins.psucontrol_rpigpio - INFO - Running RPi.GPIO version 0.7.0
2021-04-26 18:24:18,348 - octoprint.plugins.psucontrol_rpigpio - INFO - Configuring sensing GPIO for pin 12
2021-04-26 18:24:18,413 - octoprint.plugins.pi_support - WARNING - Fetching the current throttle state via "/usr/bin/vcgencmd get_throttled" doesn't work
2021-04-26 18:24:20,457 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2021-04-26 18:24:20,601 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2021-04-26 18:24:21,397 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._http._tcp.local.' for _http._tcp.local.
2021-04-26 18:24:22,291 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._octoprint._tcp.local.' for _octoprint._tcp.local.
2021-04-26 18:24:22,332 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on raspberrypi for SSDP
[... cut ...]
2021-04-26 18:25:43,021 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-26 18:25:43,021 - octoprint.plugins.psucontrol - ERROR - Exception while reading GPIO line
Traceback (most recent call last):
  File "/home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages/octoprint_psucontrol/__init__.py", line 236, in _check_psu_state
    r = self._configuredGPIOPins['sense'].read()
KeyError: 'sense'
2021-04-26 18:25:43,021 - octoprint.plugins.psucontrol - DEBUG - Result: 0
2021-04-26 18:25:43,021 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: 0```

kantlivelong, 2021-04-26:
Yep. Alright so lets see what this returns:
```
fuser -m -u -v /dev/gpiochip0
ps auxww
```

whinis, 2021-04-26:
```pi@raspberrypi:~ $ fuser -m -u -v /dev/gpiochip0
                     USER        PID ACCESS COMMAND
/dev/gpiochip0:      root     kernel mount (root)/dev
                     pi         1188 f.... (pi)systemd
                     pi         1216 f.... (pi)lxsession
                     pi         1234 F.... (pi)dbus-daemon
                     pi         1330 f.... (pi)gvfsd
                     pi         1337 F.... (pi)gvfsd-fuse
                     pi         1350 f.... (pi)openbox
                     pi         1352 f.... (pi)lxpolkit
                     pi         1355 f.... (pi)lxpanel
                     pi         1356 f.... (pi)pcmanfm
                     pi         1373 f.... (pi)wicd-client
                     pi         1378 f.... (pi)agent
                     pi         1383 f.... (pi)light-locker
                     pi         1392 f.... (pi)sh
                     pi         1393 f.... (pi)zenity
                     pi         1395 f.... (pi)clipit
                     pi         1403 F.... (pi)pulseaudio
                     pi         1416 F.... (pi)dconf-service
                     pi         1428 F.... (pi)menu-cached
                     pi         1464 f.... (pi)gvfs-udisks2-vo
                     pi         1474 f.... (pi)gvfs-gphoto2-vo
                     pi         1480 f.... (pi)gvfs-goa-volume
                     pi         1487 f.... (pi)gvfs-mtp-volume
                     pi         1495 f.... (pi)gvfs-afc-volume
                     pi         1504 f.... (pi)gvfsd-trash
                     pi        15827 F.... (pi)screen
pi@raspberrypi:~ $ ps auxww
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  1.5  0.6  34980  5984 ?        Ss   Apr17 206:39 /sbin/init
root         2  0.0  0.0      0     0 ?        S    Apr17   0:02 [kthreadd]
root         3  0.0  0.0      0     0 ?        I<   Apr17   0:00 [rcu_gp]
root         4  0.0  0.0      0     0 ?        I<   Apr17   0:00 [rcu_par_gp]
root         6  0.0  0.0      0     0 ?        I<   Apr17   0:00 [kworker/0:0H-mmc_complete]
root         8  0.0  0.0      0     0 ?        I<   Apr17   0:00 [mm_percpu_wq]
root         9  0.0  0.0      0     0 ?        S    Apr17   0:00 [rcu_tasks_rude_]
root        10  0.0  0.0      0     0 ?        S    Apr17   0:00 [rcu_tasks_trace]
root        11  0.2  0.0      0     0 ?        S    Apr17  30:18 [ksoftirqd/0]
root        12  0.2  0.0      0     0 ?        I    Apr17  34:45 [rcu_sched]
root        13  0.0  0.0      0     0 ?        S    Apr17   0:01 [migration/0]
root        14  0.0  0.0      0     0 ?        S    Apr17   0:00 [cpuhp/0]
root        15  0.0  0.0      0     0 ?        S    Apr17   0:00 [cpuhp/1]
root        16  0.0  0.0      0     0 ?        S    Apr17   0:00 [migration/1]
root        17  0.1  0.0      0     0 ?        S    Apr17  13:27 [ksoftirqd/1]
root        20  0.0  0.0      0     0 ?        S    Apr17   0:00 [cpuhp/2]
root        21  0.0  0.0      0     0 ?        S    Apr17   0:00 [migration/2]
root        22  0.1  0.0      0     0 ?        S    Apr17  13:31 [ksoftirqd/2]
root        25  0.0  0.0      0     0 ?        S    Apr17   0:00 [cpuhp/3]
root        26  0.0  0.0      0     0 ?        S    Apr17   0:00 [migration/3]
root        27  0.1  0.0      0     0 ?        S    Apr17  13:18 [ksoftirqd/3]
root        30  0.0  0.0      0     0 ?        S    Apr17   0:00 [kdevtmpfs]
root        31  0.0  0.0      0     0 ?        I<   Apr17   0:00 [netns]
root        35  0.0  0.0      0     0 ?        S    Apr17   0:00 [kauditd]
root        36  0.0  0.0      0     0 ?        S    Apr17   0:01 [khungtaskd]
root        37  0.0  0.0      0     0 ?        S    Apr17   0:00 [oom_reaper]
root        38  0.0  0.0      0     0 ?        I<   Apr17   0:00 [writeback]
root        39  0.0  0.0      0     0 ?        S    Apr17   2:49 [kcompactd0]
root        57  0.0  0.0      0     0 ?        I<   Apr17   0:00 [kblockd]
root        58  0.0  0.0      0     0 ?        I<   Apr17   0:00 [blkcg_punt_bio]
root        59  0.0  0.0      0     0 ?        S    Apr17   0:00 [watchdogd]
root        61  0.0  0.0      0     0 ?        I<   Apr17   0:00 [kworker/3:1H-kblockd]
root        62  0.0  0.0      0     0 ?        I<   Apr17   0:00 [rpciod]
root        64  0.0  0.0      0     0 ?        I<   Apr17   0:00 [xprtiod]
root        65  0.0  0.0      0     0 ?        S    Apr17   1:02 [kswapd0]
root        66  0.0  0.0      0     0 ?        I<   Apr17   0:00 [nfsiod]
root        67  0.0  0.0      0     0 ?        I<   Apr17   0:00 [iscsi_eh]
root        68  0.0  0.0      0     0 ?        I<   Apr17   0:00 [iscsi_destroy]
root        69  0.0  0.0      0     0 ?        I<   Apr17   0:00 [dwc_otg]
root        70  0.0  0.0      0     0 ?        I<   Apr17   0:00 [DWC Notificatio]
[... 101 lines cut ...]
root      9710  0.1  0.0      0     0 ?        I    18:28   0:00 [kworker/2:2-events]
root     12058  0.0  0.0      0     0 ?        I    18:31   0:00 [kworker/1:1-events]
root     12426  0.1  0.0      0     0 ?        I    18:32   0:00 [kworker/3:1-events]
root     13263  0.0  0.0      0     0 ?        I    18:33   0:00 [kworker/2:0-mm_percpu_wq]
lightdm  13874  0.0  0.6  14716  6052 ?        Ss   16:21   0:00 /lib/systemd/systemd --user
lightdm  13875  0.0  0.3  37028  3772 ?        S    16:21   0:00 (sd-pam)
root     13950  1.7  0.0      0     0 ?        I<   17:50   0:56 [kworker/u9:3-xprtiod]
[... cut ...]
root     18589  0.1  0.0      0     0 ?        I    17:57   0:04 [kworker/0:1-events]
pi       18737  0.0  0.2  10136  2692 pts/0    R+   18:42   0:00 ps auxww
root     20223  0.1  0.0      0     0 ?        I    17:59   0:02 [kworker/u8:3-rpciod]
root     28164  0.0  0.0      0     0 ?        I    18:10   0:01 [kworker/u8:2-rpciod]
root     28165  0.1  0.0      0     0 ?        I    18:10   0:02 [kworker/u8:6-rpciod]
root     32447  0.1  0.0      0     0 ?        I    18:15   0:01 [kworker/u8:0-events_unbound]
```

kantlivelong, 2021-04-26:
One more
```
sudo lsof | grep '/dev/gpiochip0'
```

whinis, 2021-04-26:
``` sudo lsof | grep '/dev/gpiochip0'
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
python3    5900                  octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6098 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6099 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6102 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6103 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6193 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6194 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6212 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6214 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6228 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6320 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6321 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6322 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6347 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
python3    5900  6390 python3    octoprint   21r      CHR      254,0        0t0         83 /dev/gpiochip0
```

kantlivelong, 2021-04-26:
I lied,

```
groups pi
ls -la /dev/gpiochip0
```

If it's not permissions at this point then I've got no idea what's going on but if that was the case it should be throwing a permission denied error.

whinis, 2021-04-26:
```pi : pi adm tty dialout cdrom sudo audio www-data video plugdev games users input netdev spi i2c gpio
pi@raspberrypi:~ $ ls -la /dev/gpiochip0
crw-rw---- 1 root gpio 254, 0 Apr 17 14:17 /dev/gpiochip0```

But if you notice I am running octoprint under octoprint user so I took the liberty of showing that one as well 

```pi@raspberrypi:~ $ groups octoprint
octoprint : octoprint tty dialout gpio
```

whinis, 2021-04-26:
I added some debug code, not sure if you broke something or I broke something but here is the keys of configured GPIO
```2021-04-26 19:13:29,778 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2021-04-26 19:13:29,778 - octoprint.plugins.psucontrol - DEBUG - dict_keys(['switch'])
2021-04-26 19:13:29,779 - octoprint.plugins.psucontrol - ERROR - Exception while reading GPIO line```

kantlivelong, 2021-04-26:
Habit :)

`2021-04-26 19:13:29,779 - octoprint.plugins.psucontrol - ERROR - Exception while reading GPIO line` is expected since setup threw the exception.

Everything looks right as far as I can tell. The only other thing I can suggest is rebooting the device.

whinis, 2021-04-26:
If I had to guess based on https://github.com/rm-hull/OPi.GPIO/issues/3 its that the sense pin and control pin are on the same GPIO device. So after you setup the control it tried to export the same device again and cannot do that. Maybe you should check if they both use the same device and if they do then reuse the device setup for control.

kantlivelong, 2021-04-26:
The same GPIO device can definitely be used. It's possible that the pin wasn't cleaned up before and now can't be configured. Have you rebooted?

whinis, 2021-04-26:
I did, same error fresh after reboot

kantlivelong, 2021-04-26:
Wait...
`2021-04-26 18:24:18,348 - octoprint.plugins.psucontrol_rpigpio - INFO - Configuring sensing GPIO for pin 12`

Disable PSUControl-RPi.GPIO or set the sense pin to 0 in it.

whinis, 2021-04-26:
No change after disabling it

kantlivelong, 2021-04-26:
Logs again?

whinis, 2021-04-26:
```
2021-04-26 19:40:42,908 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-04-26 19:40:43,023 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-04-26 19:40:43,024 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-04-26 19:40:43,025 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-04-26 19:41:10,419 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-04-26 19:41:18,327 - octoprint.util.pip - INFO - Using "/home/octoprint/python_venvs/octoprint1/bin/python3 -m pip" as command to invoke pip
2021-04-26 19:41:25,049 - octoprint.util.pip - INFO - Version of pip is 20.0.2
2021-04-26 19:41:25,061 - octoprint.util.pip - INFO - pip installs to /home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-04-26 19:41:25,062 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-04-26 19:41:25,623 - octoprint.plugins.psucontrol - DEBUG - GPIODevice: /dev/gpiochip0
2021-04-26 19:41:25,624 - octoprint.plugins.psucontrol - DEBUG - switchingMethod: GPIO
2021-04-26 19:41:25,626 - octoprint.plugins.psucontrol - DEBUG - onoffGPIOPin: 16
2021-04-26 19:41:25,627 - octoprint.plugins.psucontrol - DEBUG - invertonoffGPIOPin: False
2021-04-26 19:41:25,629 - octoprint.plugins.psucontrol - DEBUG - onGCodeCommand: M80
2021-04-26 19:41:25,637 - octoprint.plugins.psucontrol - DEBUG - offGCodeCommand: M81
2021-04-26 19:41:25,642 - octoprint.plugins.psucontrol - DEBUG - onSysCommand: 
2021-04-26 19:41:25,643 - octoprint.plugins.psucontrol - DEBUG - offSysCommand: 
2021-04-26 19:41:25,645 - octoprint.plugins.psucontrol - DEBUG - switchingPlugin: None
2021-04-26 19:41:25,656 - octoprint.plugins.psucontrol - DEBUG - enablePseudoOnOff: True
2021-04-26 19:41:25,660 - octoprint.plugins.psucontrol - DEBUG - pseudoOnGCodeCommand: M80
2021-04-26 19:41:25,667 - octoprint.plugins.psucontrol - DEBUG - pseudoOffGCodeCommand: M81
2021-04-26 19:41:25,668 - octoprint.plugins.psucontrol - DEBUG - postOnDelay: 5.0
2021-04-26 19:41:25,672 - octoprint.plugins.psucontrol - DEBUG - connectOnPowerOn: False
2021-04-26 19:41:25,677 - octoprint.plugins.psucontrol - DEBUG - disconnectOnPowerOff: False
2021-04-26 19:41:25,688 - octoprint.plugins.psucontrol - DEBUG - sensingMethod: GPIO
2021-04-26 19:41:25,695 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPin: 12
2021-04-26 19:41:25,699 - octoprint.plugins.psucontrol - DEBUG - sensePollingInterval: 5
2021-04-26 19:41:25,708 - octoprint.plugins.psucontrol - DEBUG - invertsenseGPIOPin: False
2021-04-26 19:41:25,713 - octoprint.plugins.psucontrol - DEBUG - senseGPIOPinPUD: 
2021-04-26 19:41:25,726 - octoprint.plugins.psucontrol - DEBUG - senseSystemCommand: 
2021-04-26 19:41:25,728 - octoprint.plugins.psucontrol - DEBUG - sensingPlugin: psucontrol_rpigpio
2021-04-26 19:41:25,733 - octoprint.plugins.psucontrol - DEBUG - autoOn: True
2021-04-26 19:41:25,735 - octoprint.plugins.psucontrol - DEBUG - autoOnTriggerGCodeCommands: G0,G1,G2,G3,G10,G11,G28,G29,G32,M104,M106,M109,M140,M190
2021-04-26 19:41:25,739 - octoprint.plugins.psucontrol - DEBUG - enablePowerOffWarningDialog: True
2021-04-26 19:41:25,740 - octoprint.plugins.psucontrol - DEBUG - powerOffWhenIdle: True
2021-04-26 19:41:25,741 - octoprint.plugins.psucontrol - DEBUG - idleTimeout: 10
2021-04-26 19:41:25,743 - octoprint.plugins.psucontrol - DEBUG - idleIgnoreCommands: M105
2021-04-26 19:41:25,747 - octoprint.plugins.psucontrol - DEBUG - idleTimeoutWaitTemp: 50
2021-04-26 19:41:25,755 - octoprint.plugins.psucontrol - DEBUG - turnOnWhenApiUploadPrint: False
2021-04-26 19:41:25,761 - octoprint.plugins.psucontrol - DEBUG - turnOffWhenError: False
2021-04-26 19:41:25,783 - octoprint.plugins.psucontrol_rpigpio - DEBUG - GPIOMode: BCM
2021-04-26 19:41:25,787 - octoprint.plugins.psucontrol_rpigpio - DEBUG - onoffGPIOPin: 0
2021-04-26 19:41:25,794 - octoprint.plugins.psucontrol_rpigpio - DEBUG - invertonoffGPIOPin: False
2021-04-26 19:41:25,801 - octoprint.plugins.psucontrol_rpigpio - DEBUG - senseGPIOPin: 0
2021-04-26 19:41:25,806 - octoprint.plugins.psucontrol_rpigpio - DEBUG - invertsenseGPIOPin: False
2021-04-26 19:41:25,812 - octoprint.plugins.psucontrol_rpigpio - DEBUG - senseGPIOPinPUD: 
2021-04-26 19:41:25,820 - octoprint.plugins.psucontrol_rpigpio - INFO - Running RPi.GPIO version 0.7.0
2021-04-26 19:41:26,165 - octoprint.plugins.pi_support - WARNING - Fetching the current throttle state via "/usr/bin/vcgencmd get_throttled" doesn't work
2021-04-26 19:41:32,562 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2021-04-26 19:41:32,971 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2021-04-26 19:41:33,966 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._http._tcp.local.' for _http._tcp.local.
2021-04-26 19:41:34,982 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on raspberrypi._octoprint._tcp.local.' for _octoprint._tcp.local.
2021-04-26 19:41:35,002 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on raspberrypi for SSDP
[... cut ...]
2021-04-26 19:44:03,448 - octoprint.plugins.psucontrol - ERROR - Exception while reading GPIO line
Traceback (most recent call last):
  File "/home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages/octoprint_psucontrol/__init__.py", line 236, in _check_psu_state
    r = self._configuredGPIOPins['sense'].read()
KeyError: 'sense'
2021-04-26 19:44:03,449 - octoprint.plugins.psucontrol - DEBUG - Result: 0
2021-04-26 19:44:03,449 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: 0
2021-04-26 19:44:04,071 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.30) 26.96ms
```

kantlivelong, 2021-04-26:
:shrug: 

Reboot one more time just in case but can't think of anything else.

whinis, 2021-04-27:
So I looked up 
https://stackoverflow.com/questions/62900089/device-or-resource-busy-while-configuring-gpio-as-a-char-device

Which said to use gpioinfo which is from libgpiod
```
gpiochip0 - 54 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed       unused   input  active-high
        line   4:      unnamed       unused   input  active-high
        line   5:      unnamed       unused   input  active-high
        line   6:      unnamed       unused   input  active-high
        line   7:      unnamed       unused   input  active-high
        line   8:      unnamed   "spi0 CS0"  output   active-low [used]
        line   9:      unnamed       unused   input  active-high
        line  10:      unnamed       unused   input  active-high
        line  11:      unnamed       unused   input  active-high
        line  12:      unnamed   "spi0 CS1"  output   active-low [used]
        line  13:      unnamed       unused   input  active-high
        line  14:      unnamed       unused   input  active-high
        line  15:      unnamed       unused   input  active-high
        line  16:      unnamed  "periphery"  output  active-high [used]
        line  17:      unnamed       unused   input  active-high
        line  18:      unnamed      "sysfs"  output  active-high [used]
        line  19:      unnamed       unused   input  active-high
        line  20:      unnamed       unused   input  active-high
        line  21:      unnamed       unused   input  active-high
        line  22:      unnamed       unused   input  active-high
        line  23:      unnamed       unused   input  active-high
        line  24:      unnamed       unused   input  active-high
        line  25:      unnamed       unused   input  active-high
        line  26:      unnamed      "sysfs"  output  active-high [used]
        line  27:      unnamed       unused   input  active-high
        line  28:      unnamed       unused   input  active-high
        line  29:      unnamed       unused   input  active-high
        line  30:      unnamed       unused   input  active-high
        line  31:      unnamed       unused   input  active-high
        line  32:      unnamed       unused   input  active-high
        line  33:      unnamed       unused   input  active-high
        line  34:      unnamed       unused   input  active-high
        line  35:      unnamed       unused   input  active-high
        line  36:      unnamed       unused   input  active-high
        line  37:      unnamed       unused   input  active-high
        line  38:      unnamed       unused   input  active-high
        line  39:      unnamed       unused   input  active-high
        line  40:      unnamed       unused   input  active-high
        line  41:      unnamed       unused   input  active-high
        line  42:      unnamed       unused   input  active-high
        line  43:      unnamed       unused   input  active-high
        line  44:      unnamed       unused   input  active-high
        line  45:      unnamed       unused   input  active-high
        line  46:      unnamed       unused   input  active-high
        line  47:      unnamed       unused  output  active-high
        line  48:      unnamed       unused   input  active-high
        line  49:      unnamed       unused   input  active-high
        line  50:      unnamed       unused   input  active-high
        line  51:      unnamed       unused   input  active-high
        line  52:      unnamed       unused   input  active-high
        line  53:      unnamed       unused   input  active-high
```

kantlivelong, 2021-04-27:
Ahh

So based on that what does this yield:
```
cat /proc/cmdline
cat /boot/config.txt
```

whinis, 2021-04-27:
It seems this was added by another device I was testing prior to plugging this into my printer
```
 cat /proc/cmdline
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  selinux=0 dwc_otg.lpm_enable=0 rootwait rw nfsroot=192.168.1.55:/mnt/MainStorage/nfsboot/pi/,v3 ip=dhcp root=/dev/nfs elevator=deadline
```
```
pi@raspberrypi:~/libgpiod-1.1.1 $ cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# Lines added by matrixlabs_edit_settings.py.
# Commented definitions of the settings might be above.
dtoverlay=spi0-cs,cs1_pin=12
dtparam=spi=on
enable_uart=1
start_x=1
pi@raspberrypi:~/libgpiod-1.1.1 $ cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# Lines added by matrixlabs_edit_settings.py.
# Commented definitions of the settings might be above.
dtoverlay=spi0-cs,cs1_pin=12
dtparam=spi=on
enable_uart=1
start_x=1
```

kantlivelong, 2021-04-27:
`dtoverlay=spi0-cs,cs1_pin=12`

yepp.... comment that out and reboot

whinis, 2021-04-27:
Waiting on rebooting, another job is currently running

whinis, 2021-04-27:
Rebooting fixed the issue now

closed 2021-04-27 by whinis

kantlivelong, 2021-04-27:
Awesome. Thanks for working it out with me. Will add something in the wiki about this.