kantlivelong/OctoPrint-PSUControl #189: Non functional after installing 1.0.1

Markeno76, 2021-04-17:
#### What were you doing?

1. ...Install 1.0.1
2. ...
3. ...

#### What did you expect to happen?
It to work still after the update.

#### What happened instead?
It did not work after the update. Lightning bolt won't change on clicking.  I tried installing PSU Control RPGPIO and no change.  Sensing is set to sense GPIO Pin 40.  Changing it from that, the bolt will toggle, but won't trigger then GPIO  40 like it is supposed to.

#### Version of OctoPrint-PSUControl
It is not listed in the lower left but it quit on installing 1.0.1 update

#### Operating System running OctoPrint
OctoPrint 1.5.3 Python 2.7.16 OctoPi 0.17.0

#### Printer model & used firmware incl. version
Not applicable as PSU Control does not connect to the printer.  

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG
not sure how to get this

#### Wiring diagram
Pi4b GPIO21 (Pin 40) to IOT Relay unit.

kantlivelong, 2021-04-17:
https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

Set debug
restart octoprint
reproduce
send logs

htrabbit, 2021-04-17:
I have same issue , after update PSU no longer works. 
lightening bolt does not toggle 
M80 Returns OK but does not turn on

[octoprint.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6330021/octoprint.zip)

kantlivelong, 2021-04-17:
> I have same issue , after update PSU no longer works.

@htrabbit  Open a new ticket and provide all required info. Be sure to restart octoprint after debug logs are enabled.

Markeno76, 2021-04-18:
[octoprint.log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6330428/octoprint.log)
This log is from my second Octoprint setup on my other printer.  I eventually got it working on there.  I had no previously used PSU Control on that Octopi, so it was a fresh setup of it.  For some reason while changing the settings it crashed and I could not get back into the web interface.  It was still connecting to the wifi, but the web and ssh couldn't connect although I was able to connect up a display and keyboard and it was running and had wifi connection.  I rebooted it twice but no change.  I had to reimage the sd card, restored a backup and went from there. After that it seemed an odd bit of rebooting as well as changing to BCM for the numbering in the GPIO plugin.  When set to BOARD and Pin 40 it would not work.  I also use the Enclosure Plugin, and for it as well as PSU Control I have always use 40.

Doing the same changes did not work on my first printer and Octopi setup.  It had been running a print until recently, and did try doing the same changes on it, but that did not fix the issue there.  I will see about enabling debug logs on it tomorrow if I get a chance and upload it.

moarkaboom, 2021-04-18:
Same issue for me after the Update. I will try to provice details later.

tutanra, 2021-04-18:
Same issue, also if psucontrol plugin is enabled **wired ethernet** dissapear from system.

kantlivelong, 2021-04-18:
@Markeno76 
If you haven't already be sure to check any GPIO related settings and ensure pin numbering is using BCM and not BOARD.

Markeno76, 2021-04-18:
After powering off the Pi for the night, the one I uploaded the log for, that I had gotten working is no longer working.  No changes made that I am aware of.  I had it do a shutdown from the web interface and went to bed, after powering it back on his morning it no longer works.  It is also again not letting the Enclosure plugin access that GPIO pin.  Last night both could access it, the lightning bolt was showing status correctly.
I have attached the octoprint log again, same octopi same settings.

[octoprint (1).log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6331169/octoprint.1.log)

kantlivelong, 2021-04-18:
@Markeno76 looks like you installed PSUControl-RPiGPIO. Remove that and check the settings that I mentioned or open a ticket on https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO. You shouldn't really need PSUControl-RPiGPIO anyway.

Markeno76, 2021-04-18:
I have removed PSUControl-RPiGPIO, switched it back and set the pin number from the BOARD 40 designation to 21 the BCM number for both Control and Sensing.  The lightning bolt button can now turn on the printer.  It does not go Green, and it does not turn the printer off when pressed again.  I also use BCM Pin 21 for the Sense function, in floating.   Sounds like #192?
[octoprint (3).log](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6331200/octoprint.3.log)

kantlivelong, 2021-04-18:
Ah did not see you were using the same pin for both. Not really supposed to do that as an output can't really be an input at the same time.

#182

Markeno76, 2021-04-19:
I just wanted to follow up, I did get PSUControl back to fully working.  Both Switching and Sensing.  
Looking over it there were three primary issues.  One being the removal of BOARD numbering for the GPIO in favor of only having BCM Numbering for the GPIO.  It seems referenced that maybe the upgrade should have changed the pin numbers stored in it?  It did not do that for me.  The next being the inability to now use the same GPIO Pin for both Switching and Sensing functions.  There was a third issue Sensing not working without having at least Kernel 5.5.  There was the other issue it seemed some changes wouldn't take effect unless the Pi was rebooted.

There were other complications, confusion on the new PSUControl-RPiGPIO addon plugin thinking it was needed when it wasn't apparently.  A locked up and then crashed Octoprint, then a failed SD card on the original Octoprint.  Then a small wiring change.

I had to start a print on the Octopi install that I had been using PSUControl on, so I went on to add it to my second printer.  I needed to do that as I had gotten the IOT Relay unit for it the other day.  This unfortunately became troublesome in it locked up while changing settings in the PSUControl-RPiGPIO.  I had to reimage the SD Card for some reason.  I was mistakenly using the RPi.GPIO addon plugin, I did get that working, but after a shutdown and then powering it up it wasn't working again.  It was then removed, and settings back to BCM numbering but the Sensing wasn't working due to not being able to use the same pin as the Switching function any longer.  I am still temporarily using "Internal" Sensing on that Octoprint.  I will go back to it and wire the Switching pin to another pin for Sensing later on for it.

I then went back to my primary printer and trying to get it working on there as the print was over.  I tried to run an update on Raspbian via SSH, it errored out saying about a Readonly filesystem part way through the update.  This killed the system.  I pulled the MicroSD Card and tried to reimage it but it would not format, again saying read only.  It appears due to the card failure.  I imaged a new MicroSD Card with the latest OctoPi image.  While having it open I wired the Switching pin to another GPIO to use for Sensing (Per #182 ).  The Sensing still would not work.  This turned out to be that the OctoPi 18.0 image has kernel 5.4, and in order for it to work you need kernel 5.5 or higher (Per #193 ).  I did a full-upgrade that took it to 5.10.  I now have Switching and Sensing working on the GPIO with the two pins wired together (Per #182 ).

kantlivelong, 2021-04-19:
Thanks for the summary.

So in the short term I'll see if I can reimplement the "bugged" sensing/switching on same pin thing into PSUControl-RPi.GPIO. That should help with #182 and #193

closed 2021-04-19 by kantlivelong