kantlivelong/OctoPrint-PSUControl #190: PSU no longer works after update

htrabbit, 2021-04-17:
#### What were you doing?

1. ...turn on Pi 4 
2. ...on the octopint webpage select the lightening bolt to turn on printer
3. ...printer does not turn on (worked before update)
4. ... i turned on psuplugin debug but no log file is generated

#### What did you expect to happen?

#### What happened instead?

#### Version of OctoPrint-PSUControl 1.0.1

#### Operating System running OctoPrint
octopi on linux

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.psucontrol set to DEBUG

#### Wiring diagram
i am using pin 40 to trigger a ssr
[image]
[image]
[image]
[image]
[PSUlogs.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/6330143/PSUlogs.zip)

kantlivelong, 2021-04-17:
40? Do you mean to use GPIO21?

gallowayk, 2021-04-17:
Same thing... PSU control just refuses to work for me, though I was forced to use the RPi GPIO "sub-plugin". Even when configuring it the same way, the plugin doesn't function correctly or at all.

kantlivelong, 2021-04-17:
restart OctoPrint atter making changes to PSUControl-RPi.GPIO. will need debug logs with the correct pin number for this plugin to continue

htrabbit, 2021-04-17:
pin 40 is GPIO21 thats correct
should i change it to 21???
it has always been set as 40

htrabbit, 2021-04-17:
should i set PSUControl to zero's and use only the PSUControl-RPi.GPIO set to 21?

htrabbit, 2021-04-17:
2021-04-18 00:50:08,875 - octoprint.plugins.psucontrol_rpigpio - ERROR - You must setup() the GPIO channel first
2021-04-18 00:50:08,875 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False

htrabbit, 2021-04-18:
OK i have it working, It appears that PSUcontrol-rpigpio is not necessary for me at least
1. uninstall psucontrol_rpigpio
2. 
[image]

kantlivelong, 2021-04-18:
That's it. Closing

closed 2021-04-18 by kantlivelong

htrabbit, 2021-04-18:
A BIG  thank you 
this was fast response.

kantlivelong, 2021-04-18:
All good